Jyers/Marlin #1161: E stepper motor makes weird sound

etwasmitbaum, 2021-07-17:
### Description

My E stepper motor started to make some weird noises after disabling stealthChop. I was just wondering if those are normal or if my Configurations are incorrect. Demo Video with configuration files are below.

### Steps to Reproduce

1. Using my Configurations (files below)
2. Starting to print

**Expected behavior:** [What you expect to happen]
No weird sounds from my stepper motor.

**Actual behavior:** [What actually happens]
Weird sounds from my stepper motor.

#### Additional Information

Configuration files: [config.zip](https://github.com/Jyers/Marlin/files/6834656/config.zip)
Demo Video: 

https://user-images.githubusercontent.com/47484288/126036404-2720f153-d0cf-4a01-9a39-bd3f60c7e25b.mp4

Jyers, 2021-07-17:
This is normal. **_Stealth_**-Chop, as the name implies, is meant to make you motor quieter. This is why it sounds different when it is disabled and put into speedcycle mode

closed 2021-07-17 by Jyers

etwasmitbaum, 2021-07-17:
Thanks, just wanted to make sure everything is alright.
Sorry for bothering.