Jyers/Marlin #1162: Heating failure when temps are changed after starting a preheat

MohnJadden, 2021-07-17:
### Description

If I start a preheat, then change the bed and hotend temps, my printer will throw a heating failure error and require a restart.  This happens maybe 20-30% of the time I run through the steps to reproduce it.  It happened on 1.3.3 and persists into 1.3.4.  I'm using an Ender 3 v2 with a 4.2.2 board.  

I posted a few times in the support channel in the Discord and wasn't able to get anything to address the issue.  I'm not the only person having it - a few folks have replied to comments and posts, I'll direct them here to add their experiences.  

### Steps to Reproduce

1) After a print finishes, go to Prepare -> Preheat -> PLA and start it

2) Go to Control -> bed temp and set to a lower temp than it is at present - e.g. the bed is at 40, the preheat is putting it to 50, but I set it to 35

3) Go to Control -> hotend temp and set it to a higher temp than it is at present - e.g. it's at 110, the preheat is putting it to 200, but I set it to 210

4) Wait about 15-30 seconds

**Expected behavior:** The temperatures change to whatever I have set them to

**Actual behavior:** The printer display will say E Heating even if the bed is at temp, then display "heating failure: printer requires restart", and beep loudly

#### Additional Information

I'm using the unmodified 1.3.4 release for 4.2.2 boards downloaded from https://github.com/Jyers/Marlin/releases/download/v1.3.4/E3V2-ManualMesh-5x5-v4.2.2-v1.3.4.bin - no changes made by me.

Cal-LF, 2021-07-17:
If you are changing the temps during a preheat, it could very well trigger a heating failure. The FW gets confused as to what is happening. Suggest you do not adjust temps while in preheat mode and select the temp you want to start with.

Jyers, 2021-07-21:
Don't know why you opened a second issue about this, but I've tuned the thermal settings and this should be fixed in the next release.

closed 2021-07-21 by Jyers

AndyShap2020, 2022-09-03:
I have this exact same issue on your newest release (9/1/22). I have found that if you let the hotend cool to lower than 100 degrees C, you won't have any issues.  I'd love to see this fixed!  P.S. Are you still developing this product?  I don't see anything since November 2021.