Jyers/Marlin #1163: hi i need a 300 degres hotend temp on your super 10x10 hs for ender 3 v2 is possible i have try but i guess i cant by my self

jokingradio, 2021-07-18:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Jyers, 2021-07-18:
[Configs.zip](https://github.com/Jyers/Marlin/files/6835602/Configs.zip)

Here are some configs for that setup, if you need more help I'd recommend starting a discussion rather than an issue.

closed 2021-07-18 by Jyers

jokingradio, 2021-07-18:
big thanks i go try this