Jyers/Marlin #1174: Z-offset problem in Tune menu

joonghochoe, 2021-07-20:
### Description

If I change the Z-offset(babystepping) via the tune menu while printing, the Z-offset will be applied twice when changing the layer.

### Steps to Reproduce

1. Start print simple 3d model (layer height : 0.2mm)
2. After printing starts, set Z-offset to 1.0 in the tune menu(be careful as the bed can be gouged)
3. Z axis rises by 1.0mm(I set a lot to see the difference.)
4. When the layers change, the Z axis rises again by 1.0mm.
5. When the next layer changes, the Z axis rises by 0.2mm.

If the z offset is set to negative, the z-axis will fall twice(If you do it at wrong position, the bed will be gouged).

**Expected behavior:** [What you expect to happen]
It would be nice if the layer change works normally after setting the z offset value.

**Actual behavior:** [What actually happens]
After setting the Z offset, the layer changes too much.
0.2 -> 0.4 -> 0.6 -> 0.8 -> Setting Z-Offset to 1.0(Z axis rises 1.0mm) -> Again Z axis rises 1.0 -> 3.2(3.0?)

#### Additional Information
This is the same for both version 1.3.4 and master version(no bltouch, no probe). But the Creality stock version works fine.

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Jyers, 2021-07-21:
Interesting, I'm pretty sure all z offset modifications from the tune menu are done through baby-stepping, which doesn't actually change where the printer thinks the head is... so it shouldn't mess with any movement after the fact. I'll see if I can recreate this.

tititopher68-dev, 2021-07-22:
@joonghochoe @Jyers 
This test can be explained by the choice of configuration files (which does not seem to be ManualMesh ...) and the spirit of the change of the Z-Offset in the Tune menu.  Let me explain, there is no BLTouch, nor Mesh so this is not exactly the definition of Z-Offset with BLTouch or with a Mesh, but here rather a simple adjustment of the layer height.  In theory and in this situation using the babystep as defined in the code in the Tune menu, the correction artificially affects the actual position of the Z axis by modifying the Probe Z Offset.  Because in the Tune menu, this case without probe and without Mesh was not foreseen.  The command used is M290 +/- 0.01 which affects Probe Z Offset.  In a ManualMesh config the hotend is the probe, but here in the Default configuration this should not be the case.  It is therefore necessary to condition it and add before, for the Default configuration type, an M290 P0 command so as not to affect the Probe Z Offset but just the layer height correction.
 So this makes sense without this M290 P0:
 0.2 -> 0.4 -> 0.6 -> 0.8 -> Setting Z-Offset to 1.0 (Z axis rises 1.0mm) -> Again Z axis rises 1.0 -> 3.2 (3.0?) ...
 Here at the fourth layer, we apply an adjustment of 1 mm so we go immediately during printing of this layer from 0.8 to 1.8mm ... Then change the layer and we go to the layer height 0.2 + 1 or 3mm  ... why 3.2 here, I think when adjusting the offset it must have happened at the same time as the diaper change hence the extra 0.2mm ...

Soltysrysik, 2021-07-27:
I have Same problem with version 1.3.3 manual mesh leveling. Version 1.3.4 still the problem ?

Jyers, 2021-07-27:
@tititopher68-dev Very interesting. You're definitely right, now that I think of it the new Z Offset is definitely applied on the next Z move... I think the best fix may be to prevent the actual Marlin Z offset value from being synced with the local Z offset variable I use. That way it'll store it until after the print when it can be synced safely.

tititopher68-dev, 2021-07-27:
@Jyers 
To clarify and so that there is no misunderstanding, the problem (if you want) only exists if the default configuration is used, i.e. without bed Leveling, without BLTouch and  without Mesh.  The problem (having checked it to be sure) does not exist with a BLTouch or Noprobe UBL or ManualMesh ...

Jyers, 2021-07-27:
@tititopher68-dev Are you positive? I'll check myself when I get home, but the way I see it if you change the Z offset and then execute a z move, the new offset is going to be applied and the Z move will be offset by that difference. This should happen no matter the config.

tititopher68-dev, 2021-07-27:
@Jyers 
In theory, only in manualmesh, the "Z-Offset" is very very close to 0. The problem also comes from not heating the Bed to printing temperature with this configuration to make the Z-Offset ...
 Hence by conditioning with a "#if HAS_LEVELING", we use the M290 Z +/- 0.01 and otherwise we make an M290 P0 Z +/- 0.01

Jyers, 2021-07-27:
@tititopher68-dev Interesting, I'll try it when I get home. Btw don't forget `P0` won't work because BABYSTEP_ZPROBE_OFFSET is disabled. You'd need to put the conditional around where it sets the position in the screen code

tititopher68-dev, 2021-07-27:
@Jyers 
Ah the messages intersect ...
 In fact in Marlin, we must distinguish between Z babystepping and Z Probe babystepping ... currently whatever the config, we apply the Z Probe babystepping (M290 Z +/- 0.01) which is correct if we use a Mesh (bilinear or  UBL), a Probe.  But in the most basic configuration, no, because no mesh is used, no Probe ... then Z Babystepping must be used via M290 P0 Z +/- 0.01 ...

Jyers, 2021-07-27:
@tititopher68-dev Z Probe babystepping as you call it, is never used because `BABYSTEP_ZPROBE_OFFSET` is disabled. So every `M290` command behaves as if it has a `P0` attribute by default. The screen firmware is what actually sets the new Z offset, which is automatically synced with either the home offset or the probe offset............... WAIT, I've just had a realization! We were both kind of right. The reason it is only happening without a probe is because the home offsets get re-applied on the next move, while the probe offsets get applied on the next home! That'd do it! So I still think the best approach is to have it wait until after the print to sync the Z offset, either that or try to stop the home offset from being applied after it gets changed.

tititopher68-dev, 2021-07-27:
@Jyers 
You will think it can be stupid, but deep down I don't think so.
 The problem is therefore with the Noprobe versions (ManualMesh, Noprobe UBL) and the basic default version.
 For Noprobe:
 Shouldn't we just remove the Z-offset menu for these versions in Prepare and Tune?  And do a Bed Z. It would then be necessary to apply to each change the correction +/- 0.01 to all the points of the mesh to only act on the correction if we have an activated mesh.
 For default, we remove the Z-Offset options ...

Jyers, 2021-07-27:
@tititopher68-dev I would disagree. I think the  Z offset is a valuable tool for all configs. I see you're reasoning for the 'Bed Z' concept, but that's pretty much what the Z Offset does. Having it not sync during printing is as simple as one conditional, while trying to create a while new offset system based on applying the offset to the entire mesh would be a bit of a task. 

You're welcome to try your implementation and see how it goes, I'd be interested to hear how it works out! For my part I think I'm going to try my solution and see if it fixes the issue.

tititopher68-dev, 2021-07-27:
@Jyers
 I don't know if this will be the right solution, but I think I've pinpointed another issue impacting this issue.  In ManualMesh only, HAS_BED_PROBE is true ... and this impacts the Z-Offset ... modifies probe_z_offset ...
 therefore generates the bug.
 Maybe this should be conditioned to Probe_manually.  Then load babystep.h
 #if! HAS_BED_PROBE && ENABLED (BABYSTEPPING)
   #define JUST_BABYSTEP 1
 #endif
 #if EITHER (BABYSTEP_ZPROBE_OFFSET, JUST_BABYSTEP)
   #include "../../../feature/babystep.h"
 #endif
 and use one of the add_mm functions for the z-axis, which would do the job.
 #if EITHER (BABYSTEP_ZPROBE_OFFSET, JUST_BABYSTEP)
         if (BABYSTEP_ALLOWED ()) babystep.add_mm (Z_AXIS, +/- 0.01 or new value - last value);
       #endif
 I don't know if I am clear ... and not helped by the translation

Jyers, 2021-07-27:
I did a quick check and that should not be the case...

`#if ANY(HAS_Z_SERVO_PROBE, FIX_MOUNTED_PROBE, NOZZLE_AS_PROBE, TOUCH_MI_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, SOLENOID_PROBE, SENSORLESS_PROBING, RACK_AND_PINION_PROBE)`
    `#define HAS_BED_PROBE 1`
`#endif`

And just to be sure I also check for `NOZZLE_AS_PROBE`. It's only defined in the config files as

`/**`
 `* Use the nozzle as the probe, as with a conductive`
 `* nozzle system or a piezo-electric smart effector.`
 `*/`
`//#define NOZZLE_AS_PROBE`

tititopher68-dev, 2021-07-27:
@Jyers 
You're right ... I'm the one bugging ... but I think that would fix the up and down microsteps in Tune with the Default, Manual Mesh and Noprobe UBL configurations.
 Because this adds the +/- 0.01 1 only once to the Z position without affecting the corrections of the mesh and without affecting the Z offset probe ...
 What do you think ?

Jyers, 2021-07-27:
@tititopher68-dev 
You're right that that will do exactly what you have described (although I think you could get the same result by simply putting and `#if` around the bit of the display code that changes the offset), but the issue is that I want it to change the Z offset 😅 
The way I see it, if someone has to change the offset to get their print perfect, they'd want to keep using that offset. So that's why I was thinking of just desyncing the value till the end of the print so that the offset still gets applied, just not until after the print so it doesn't mess anything up. Hopefully that makes sense and isn't getting completely butchered in translation.

tititopher68-dev, 2021-07-28:
@Jyers 
The problem would it not come from there:
 set_home_offset (Z_AXIS, -zoffsetvalue);
 that it would be necessary to condition, by applying it only if one does not print ... because currently and according to the data of @joonghochoe, this explains that the first M290 Z ... modifies the Z position for the first time and then this  order a second time!  And nothing after ...
 I now think this is the solution ...

Jyers, 2021-07-28:
@tititopher68-dev 
Yes, that is what I meant by 'de-syncing' the value, just not setting it while printing.

Sorry I couldn't tell if you were agreeing with me there or trying to offer me my own solution ;-)

tititopher68-dev, 2021-07-28:
@Jyers 
In fact, I wanted to make sure I understood what you had in mind.  We both agree and it's cool ... ;-)

Note: I will soon send you various improvements (discord?), Such as extensions for dwin.cpp, aesthetic fixes in certain color configurations, etc ... and for your information, I had tested your commit during  of the selection of values, what I proposed to you, works with all the fonts, your choice still poses problems with the other fonts on the large values ​​...

Jyers, 2021-07-28:
@tititopher68-dev 
Sounds good, if you send me the fonts I can probably tune my system to better accommodate them. At the end of the day, what we really need is a standardized font width... Having different widths floating around is only going to create more issues in the long run. However we can continue this discussion over discord.

Jyers, 2021-07-30:
Alright this should be good now 😄

closed 2021-07-30 by Jyers