Jyers/Marlin #1179: Display firmware update screen colour & icon pack

Gokuh1980, 2021-07-20:
Hi, i have a question, after flash de SD card i put de DWIN_SET in the card but when i see the boot it never goes to orange, simply start up and let me use the display, why is this?
How do i change the icon pack? i did a lot a reserch but couldn´t find anything.

Thanks in advance!

adam3654, 2021-07-20:
Are you putting the SD card with the "DWIN_SET" folder into the SD card slot located on the back of the board for the screen?

Gokuh1980, 2021-07-20:
Yes, of course, i forgot to mention. SORRY!

do you have any ideas?

Thanks in advance!

adam3654, 2021-07-20:
Just to verify, the steps you are taking is as follows:

1. Format SD card 4k / fat32. If the SD card is over 8gb, remove the root partition, and make a new 8gb partition leaving the rest of the card unallocated. 

2. Copy one of the firmware set folders from: https://github.com/Jyers/Marlin/tree/Ender_3_V2_JyersUI/Display%20firmware%2FFirmware%20Sets   and paste into SD card

3. Rename folder on SD card to "DWIN_SET"

4. Power off printer 

5. remove ribbon cable from back plate of LCD screen, unscrew and remove back plate to expose micro SD slot

6. Insert micro SD card into slot on back of LCD board and reinsert the ribbon cable

7. Power on printer, screen should start blue then turn orange after a few seconds

8. Power off printer, remove the ribbon cable and SD card from LCD screen, replace back of LCD screen, replace ribbon cable, and return screen to mount.

9. Power on printer.

adam3654, 2021-07-20:
If you get that working, you might want to change the display font while you're at it. Grab the release zip from here:
https://github.com/Quint097/DWIN_Font_Tools/releases/tag/v1.1

Copy one of the font files into the DWIN_SET folder prior to flashing the LCD.

Gokuh1980, 2021-07-21:
Thank you for your answer, i followed your procedure, but retested and no, the same, i´ve tested 2gb SD card, and the original 8Gb card, and nothing, i don´t know, it could be a malfunction of the sd slot?

adam3654, 2021-07-21:
It's always just booting to the main screen? Maybe the SD slot is messed up. One more thought, you're inserting the card with the contacts facing the board, right? I'm pretty sure it won't fit in upside down, but thought I'd throw that out there.

Cal-LF, 2021-07-21:
If the screen .bin file will not update, it is likely the SD card causing this. I have had a card that will work in the mainboard but not in the screen. Keep trying with a different card and it should work. Format as adam3654 outlined above.

Snoozyl, 2021-07-23:
> Just to verify, the steps you are taking is as follows:
>
> 1. Format SD card 4k / fat32. If the SD card is over 8gb, remove the root partition, and make a new 8gb partition leaving the rest of the [...]

I have also followed these same steps. I do however only have a 32gb SD card so will be trying an 8gb card when I get ahold of one. My concern though is that I have already updated the mainboard before the screen. One guide I found online said the official guide states that the screen should be done first. Would that be a problem? If not ill try other cards.

Thanks

adam3654, 2021-07-23:
I've loaded the screen firmware after flashing the board. I don't really think the order matters.

Snoozyl, 2021-07-23:
Thanks for the update. Ill see what happens with other cards. Hopefully that resolves the issue.

Also just as a clarification, The screen immediately goes from black to orange if that is any help with diagnosing the problem.

adam3654, 2021-07-23:
I've had that happen to me, but it was when I was messing with the font file and had issues. The board doesn't like something and is not flashing the contents of the folder. What it is that the board doesn't like, I haven't a clue. I just kept trying different things until it worked.

Cal-LF, 2021-07-23:
@Snoozyl There will be a delay to the orange screen. If not, then it is not completing the update.
Try a new full format of the card to 4k. Then copy only the DWIN folder to the root directory.

Snoozyl, 2021-07-24:
> @Snoozyl There will be a delay to the orange screen. If not, then it is not completing the update.
> Try a new full format of the card to 4k. Then copy only the DWIN folder to the root directory.

I did try formatting and copying to the root. Ill post an update as soon as I get to try another SD card

Snoozyl, 2021-07-25:
> > @Snoozyl There will be a delay to the orange screen. If not, then it is not completing the update.
> > Try a new full format of the card to 4k. Then copy only the DWIN folder to the root directory.
> 
> I did try formatting and copying to the root. Ill post an update as soon as I get to try another SD card

Okay so I got ahold of the old 8gb card and the flash was a success. Seems the screen doesn't like a card over 8gb but the printer its self is fine with a 32gb.

Thanks for all the help!

Cal-LF, 2021-07-25:
@Snoozyl Excellent! The screen is finicky for sure.