Jyers/Marlin #1186: Leaving the "Load filament" screen causes firmware to crash.

Dr-Ari-Gami, 2021-07-21:
Whenever I leave the load filament screen before the z axis finishes going down, the whole firmware crashes, providing a blank screen with a cursor. 

It seems like the "Please wait until done screen" is either missing or ends early. (I don't have my 3d printer in front of me to confirm. 😳)

GordinhoPl, 2021-07-21:
I have the same, need to be patient and wait until it is going down, and then leave menu to start print...

Jyers, 2021-07-21:
Interesting, I'll look into this.

JAZ-013, 2021-08-10:
Confirming I have the same issue.  I hit this almost every time I do a filament change.  Seriously, this is the ONLY issue I have with this firmware, so thanks for having a look into it.  :D

Dr-Ari-Gami, 2021-08-16:
Correct me if I'm wrong, but this issue was sort of fixed in v1.3.5. Instead of the printer screen showing a blank cursor, it just takes the user back to the "Change Filament" screen. While this is not ideal, it is better than the firmware crashing.

Dr-Ari-Gami, 2021-08-16:
Nevermind, my experience was a fluke. It still crashes.