Jyers/Marlin #1198: Modifed Y home value not appled to auto bed leveling.

AlphaDE, 2021-07-22:
### Description

I am using an Ender 3 V2 with rails on all axis (Swiss Micro on the X axis). This moves the hotend and BLtouch about 16mm forward. When running automatic bed leveling, BLtouch in the 5x5 setup hits the clamp for the glas bed resulting in wrong values. I tried to set modified home values for the Y axis, but they only work for printing itself, not for auto bed leveling.

### Steps to Reproduce

1. Use an Ender 3 V2 with rails on the X axis, moving the hotend/bltouch in positive Y direction.
2. Set a proper Y home modified value to compensate this (-16).
2. Do auto bed leveling with BLtouch in 5x5
3. The probe hits the glas bed clamp on the third (middle) test, that value is about 1.5 mm higher due to the clamp
4. The Y axis home correction is not used to shift the position of BLtouch.

**Expected behavior:** Use the modified Y home value and apply it to the Y coordinate when doing auto bed leveling.

**Actual behavior:** The Y home modifier is not used.

#### Additional Information

The current version of Jyers firmware for Ender 3 V2 (E3V2-BLTouch-5x5-HS-v4.2.7-v1.3.4) is used.

Jyers, 2021-07-22:
Just checking, you did save your home offsets and home all axis before leveling right?

AlphaDE, 2021-07-22:
Thanks for the quick response.

Yes, I switched the printer off in the meantime, the modified Y home persists. I temporarily solved the problem by changing the probe Y value so the probe just touched beside the clamp.

If you need any testing, let me know.

tititopher68-dev, 2021-07-22:
@AlphaDE 
The parameter you are quoting are not the correct ones in your case.

You must start by manually moving the head to your desired position 0,0 (Prepare menu -> Move -> (activate Live move) and move the X and Y axes).
You write down the X and Y values.
You edit the configuration.h file by looking for #define X_MIN_POS, you define here the opposite value of that found for X (for example if X = 20mm, then you will have to define -20). ditto for Y, by setting the value Y_MIN_POS. (for example if Y = 30mm, then it will be necessary to define -30). Finally, you will have to manually find the MAX values ​​for X and Y and also modify #define X_BED_SIZE and #define Y_BED_SIZE with the values ​​found.
Then you compile it all.
When you do your HOME ALL, you will be in X, Y at the 0,0 position of the BED, but on the screen you will see (according to the example given) -20.0 and -30.0. It's normal. If you ask Move X or Move Y to go to position 0, the head will go to position 0 of the Bed.
And the bed leveling will be done correctly.  You can reset the correct value for Probe Y Offset.

The only thing that will then have to be done in the slicer is to redefine the size of the Bed with the new firmware values.

AlphaDE, 2021-07-23:
@tititopher68-dev 

Thanks for the detailed explanation. I use Jyers pre-compiled firmware for my Ender 3 V2 with BLtouch and filament sensor. 

I used the "Home offset" in the configuration menu for the Y offset (M206 Y-16)

This is applied to printing but why isn't it applied to auto bed leveling? In other words, the M206 command has the effect you describe to achieve by changing configuration.h (similar to the BLtouch offset which this firmware also allows to configure dynamically from the control screen). And the changed M206 value persists and is present at startup but seems not to be used by auto bed leveling.

I have a setup to compile the firmware for my other Ender 3 but not for the extension @Jyers did for the graphical LCD screen. I see whether I can get it compiled but would prefer to rely on the precompiled version.

tititopher68-dev, 2021-07-23:
@AlphaDE 
Command M206 has no effect on Bed Leveling.  The Bed Leveling uses the parameters of Bed, X and Y_BED_SIZE, X and Y_MAX_POS, X and Y_MIN_POS.
 Also the procedure which I propose (and which I use personally, also described in a very good video of the YouTube channel "Teaching Tech") will answer entirely to your request ... but it will be necessary to compile your firmware with the  modified configuration files ...

AlphaDE, 2021-07-23:
@tititopher68-dev Thanks for additional details, so I get the firmware from here with the specific Ender 3 V2 configuration files from the separate repository.