Jyers/Marlin #1208: Compiling for SKR Mini E3 V2.0?

Freethefire, 2021-07-24:
### Description
Hi, 
I'm trying to compile your firmware for the Skr Mini E3 v2.0 board, with a CR10 display. 
But I keep getting these errors and I'm not versed enough in programming to really understand what they mean.
Or is it pointless? I was hoping to keep some of your menu additions, but maybe they need the DWIN display?
[Configuration.zip](https://github.com/Jyers/Marlin/files/6872004/Configuration.zip)

`Compiling .pio\build\STM32F103RC_btt_maple\src\src\feature\ammeter.cpp.o
In file included from Marlin\src\module/stepper/../../feature/../lcd/marlinui.h:28:0,
                 from Marlin\src\module/stepper/../../feature/tmc_util.h:25,
                 from Marlin\src\module/stepper/trinamic.h:35,
                 from Marlin\src\module/stepper/indirection.h:44,
                 from Marlin\src\module/stepper.h:47,
                 from Marlin\src\MarlinCore.cpp:55:
Compiling .pio\build\STM32F103RC_btt_maple\src\src\feature\babystep.cpp.o
Marlin\src\module/stepper/../../feature/../lcd/marlinui.h: In static member function 'static bool MarlinUI::button_pressed()':
Marlin\src\module/stepper/../../feature/../lcd/buttons.h:59:19: error: 'keypad_buttons' was not declared in this scope
   #define RRK(B) (keypad_buttons & (B))
                   ^
Marlin\src\module/stepper/../../feature/../lcd/buttons.h:64:28: note: in expansion of macro 'RRK'
     #define BUTTON_CLICK() RRK(EN_KEYPAD_MIDDLE)
                            ^~~
Marlin\src\module/stepper/../../feature/../lcd/marlinui.h:622:50: note: in expansion of macro 'BUTTON_CLICK'
Compiling .pio\build\STM32F103RC_btt_maple\src\src\feature\bedlevel\bedlevel.cpp.o
     static inline bool button_pressed() { return BUTTON_CLICK() || TERN(TOUCH_SCREEN, touch_pressed(), false); }
                                                  ^~~~~~~~~~~~
Compiling .pio\build\STM32F103RC_btt_maple\src\src\feature\bedlevel\ubl\ubl.cpp.o
Marlin\src\module/stepper/../../feature/../lcd/buttons.h:59:19: note: suggested alternative: 'update_buttons'
   #define RRK(B) (keypad_buttons & (B))
                   ^
Marlin\src\module/stepper/../../feature/../lcd/buttons.h:64:28: note: in expansion of macro 'RRK'
     #define BUTTON_CLICK() RRK(EN_KEYPAD_MIDDLE)
                            ^~~
Marlin\src\module/stepper/../../feature/../lcd/marlinui.h:622:50: note: in expansion of macro 'BUTTON_CLICK'
     static inline bool button_pressed() { return BUTTON_CLICK() || TERN(TOUCH_SCREEN, touch_pressed(), false); }
                                                  ^~~~~~~~~~~~
Marlin\src\module/stepper/../../feature/../lcd/buttons.h:55:33: error: 'BTN_OFFSET' was not declared in this scope
   #define EN_KEYPAD_MIDDLE  _BV(BTN_OFFSET + BLEN_KEYPAD_MIDDLE)
                                 ^
Marlin\src\module/stepper/../../feature/../lcd/buttons.h:59:37: note: in definition of macro 'RRK'
   #define RRK(B) (keypad_buttons & (B))
                                     ^
Marlin\src\module/stepper/../../feature/../lcd/buttons.h:55:29: note: in expansion of macro '_BV'
   #define EN_KEYPAD_MIDDLE  _BV(BTN_OFFSET + BLEN_KEYPAD_MIDDLE)
                             ^~~
Marlin\src\module/stepper/../../feature/../lcd/buttons.h:64:32: note: in expansion of macro 'EN_KEYPAD_MIDDLE'
     #define BUTTON_CLICK() RRK(EN_KEYPAD_MIDDLE)
                                ^~~~~~~~~~~~~~~~
Marlin\src\module/stepper/../../feature/../lcd/marlinui.h:622:50: note: in expansion of macro 'BUTTON_CLICK'
     static inline bool button_pressed() { return BUTTON_CLICK() || TERN(TOUCH_SCREEN, touch_pressed(), false); }
                                                  ^~~~~~~~~~~~
Marlin\src\module/stepper/../../feature/../lcd/buttons.h:55:33: note: suggested alternative: 'STR_OFF'
   #define EN_KEYPAD_MIDDLE  _BV(BTN_OFFSET + BLEN_KEYPAD_MIDDLE)
                                 ^
Marlin\src\module/stepper/../../feature/../lcd/buttons.h:59:37: note: in definition of macro 'RRK'
   #define RRK(B) (keypad_buttons & (B))
                                     ^
Marlin\src\module/stepper/../../feature/../lcd/buttons.h:55:29: note: in expansion of macro '_BV'
   #define EN_KEYPAD_MIDDLE  _BV(BTN_OFFSET + BLEN_KEYPAD_MIDDLE)
                             ^~~
Marlin\src\module/stepper/../../feature/../lcd/buttons.h:64:32: note: in expansion of macro 'EN_KEYPAD_MIDDLE'
     #define BUTTON_CLICK() RRK(EN_KEYPAD_MIDDLE)
                                ^~~~~~~~~~~~~~~~
Marlin\src\module/stepper/../../feature/../lcd/marlinui.h:622:50: note: in expansion of macro 'BUTTON_CLICK'
     static inline bool button_pressed() { return BUTTON_CLICK() || TERN(TOUCH_SCREEN, touch_pressed(), false); }
                                                  ^~~~~~~~~~~~
Marlin\src\module/stepper/../../feature/../lcd/buttons.h:55:46: error: 'BLEN_KEYPAD_MIDDLE' was not declared in this scope
   #define EN_KEYPAD_MIDDLE  _BV(BTN_OFFSET + BLEN_KEYPAD_MIDDLE)
                                              ^
Marlin\src\module/stepper/../../feature/../lcd/buttons.h:59:37: note: in definition of macro 'RRK'
[... 88 lines cut ...]
   #define EN_KEYPAD_MIDDLE  _BV(BTN_OFFSET + BLEN_KEYPAD_MIDDLE)
                             ^~~
Marlin\src\feature\../module/stepper/../../feature/../lcd/buttons.h:64:32: note: in expansion of macro 'EN_KEYPAD_MIDDLE'
     #define BUTTON_CLICK() RRK(EN_KEYPAD_MIDDLE)
                                ^~~~~~~~~~~~~~~~
Marlin\src\feature\../module/stepper/../../feature/../lcd/marlinui.h:622:50: note: in expansion of macro 'BUTTON_CLICK'
     static inline bool button_pressed() { return BUTTON_CLICK() || TERN(TOUCH_SCREEN, touch_pressed(), false); }
                                                  ^~~~~~~~~~~~
Marlin\src\feature\../module/stepper/../../feature/../lcd/buttons.h:55:46: note: suggested alternative: 'EN_KEYPAD_MIDDLE'
   #define EN_KEYPAD_MIDDLE  _BV(BTN_OFFSET + BLEN_KEYPAD_MIDDLE)
                                              ^
Marlin\src\feature\../module/stepper/../../feature/../lcd/buttons.h:59:37: note: in definition of macro 'RRK'
   #define RRK(B) (keypad_buttons & (B))
                                     ^
Marlin\src\feature\../module/stepper/../../feature/../lcd/buttons.h:55:29: note: in expansion of macro '_BV'
   #define EN_KEYPAD_MIDDLE  _BV(BTN_OFFSET + BLEN_KEYPAD_MIDDLE)
                             ^~~
Marlin\src\feature\../module/stepper/../../feature/../lcd/buttons.h:64:32: note: in expansion of macro 'EN_KEYPAD_MIDDLE'
     #define BUTTON_CLICK() RRK(EN_KEYPAD_MIDDLE)
                                ^~~~~~~~~~~~~~~~
Marlin\src\feature\../module/stepper/../../feature/../lcd/marlinui.h:622:50: note: in expansion of macro 'BUTTON_CLICK'
     static inline bool button_pressed() { return BUTTON_CLICK() || TERN(TOUCH_SCREEN, touch_pressed(), false); }
                                                  ^~~~~~~~~~~~
*** [.pio\build\STM32F103RC_btt_maple\src\src\feature\babystep.cpp.o] Error 1
=========================================================================================================== [FAILED] Took 125.82 seconds ===========================================================================================================`

tome9111991, 2021-07-24:
what default_envs you use?

Freethefire, 2021-07-24:
> what default_envs you use?

STM32F103RC_btt_maple

Thanks for replying. 
I started out compiling with STM32F103RC_btt but didn't have any luck with that at all.

Mountain3DPrints, 2021-07-24:
Hey man I was hoping to get your insight. I have a cr10 along with 3 E3v2's
and a 5pro.

All my E3's are running your firmware. I'm trying to frankenprinter my cr10
in to a E3-mega. I've got the skr mini v3 plus the 3.0 display. Full size
E3 touch like on the original. My issue is every time I try to compile fw,
I get some nasty crash. Plus I've heard of similar efforts but the person
complaining of terrible noise

tome9111991, 2021-07-24:
I do not believe that you will get menu additions for Cr10 with the firmwares
only for DWIN

Freethefire, 2021-07-24:
> I do not believe that you will get menu additions for Cr10 with the firmwares
> only for DWIN

Ok, was hoping :)
Thanks again!

Mountain3DPrints, 2021-07-24:
The plan is not to to have mine running as a cr10. The only cr10 parts I'm
including in the build is the actual machine itself. Everything else
(board, screen etc.) Is setup exactly as the E3v2 is.

Mountain3DPrints, 2021-07-24:
Edit. To have mine running as E3v2

closed 2021-07-25 by Freethefire