Jyers/Marlin #1210: SKR MINI wont compile

Mountain3DPrints, 2021-07-24:
Hello all,
I have been trying to compile this fw for the skr mini + tft35.
I go to select the envs for the STM32F103RC or any other variation at this point and my respons remains the same. 
I cam only paraphrase as I am away from my machine and pc atm. 
The message always says "error: please specify upload port for use global"
there is also an issue "stm32f103rc (or any other section) is not valid please make the proper selection" and leaves me with a mile long list to chose from including all the boards I  have tried inserting for the envs. 

Please, any insight would be interstellar.

TIA

tome9111991, 2021-07-24:
i belive "error: please specify upload port for use global" is just a warrning look into you .pio folder.
there should be the .bin file

Mountain3DPrints, 2021-07-24:
No, that's as soon as I'm done making my changes/arrangements and I click
complie button that immediately comes up and cancels the command

Cal-LF, 2021-07-24:
Jyres FW supports the V422 and V427 mainboard and the E3V2 screen. Wondering why are you trying to make it work with another  hardware set? This would be better posted in the discussions rather than a firmware issue.

Mountain3DPrints, 2021-07-24:
I wasn't aware. I see dozens of people in this area trying to get skr to
work. Don't worry sir. I will be taking my issues and all my firmware needs
elsewhere. Thanks for the help. There are plenty of comps out there in the
wild. The three machines I have running 4.2.x are just fine. This is an out
of the ordinary situation because of the circumstances of my 4th machine.
Take care

Mountain3DPrints, 2021-07-24:
Also since you are wondering I have an old cr10 platform I am using the
footprint and has been entirely covered to resemble a E3v2. They only ran a
8bit with very limited support or room for changes. Hence the conversion.
In under $100 I effectively have a large "E3v2" just need to get the fw to
compile.

Cal-LF, 2021-07-24:
@Mountain3DPrints I think my question is a valid one to get more info to help answer your question.
Not seeing dozens of people as you state but this post would fit better in the discussions area.

Mountain3DPrints, 2021-07-24:
Ok well I wasn't intentionally being intrusive? in this area. I have never
personally interacted with forums, as I've without fail been able to find
what I need just in basic search. It's fine. Thank you for your time. I'll
be on my way. I only interjected at all because the Jyres fw is far
superior to everything else I've ever ran on all 5 of my machines.
"Vanilla" Marlin 2.0.x is great, but Jyres is pretty much a whole different
fw in its own as you know.

Good day sir

Cal-LF, 2021-07-24:
@Mountain3DPrints Come back to the discussion area with any update on your project. I'm sure others would like to here what you have going.

RipperDrone, 2021-07-25:
Afaik @Jyers is helping right now to get the Users framework integrated back into the Marlin main branch. Big kudos to @Jyers to support the root branch rather than risking a 'my own branch, anything else not invented here' attitude that has doomed many open source projects to fragmentation and finally failure. Thank you so much for your supportive spirit, @Jyers 😍.

@Mountain3DPrints - this means if you wait some more for the DWIN PRs to be merged into Marlin bugfix branch, you will have most of the marvelous Users functionality in your firmware AND be able to compile it:

- Download or clone latest Marlin bugfix branch
- copy configuration files into the Marlin folder which suit your hardware
- do manual modifications for SKR as nicely put together in Wiki style by @blazewicz here: https://github.com/MarlinFirmware/Configurations/issues/535
- compile with STM32F103RC_btt environment (if you have genuine STM32 chip on board, in case of GD chip you might want to use STM32F103RE_btt env)

Should not throw errors anymore...

Jyers, 2021-07-25:
@RipperDrone Thanks! Doing what I can 😄 

@Mountain3DPrints Actually thanks to the OR, I think I now understand what exactly is preventing the SKR boards from compiling! There has been a lot of regression from the main marlin repo that's been sneaking past my merges. If I fix that, it should solve the issues you and others are having!