Jyers/Marlin #1218: Does "Leveling active" checkbox need to be ON?

shtefcs, 2021-07-25:
### Discussed in https://github.com/Jyers/Marlin/discussions/1212

<div type='discussions-op-text'>

<sup>Originally posted by **shtefcs** July 24, 2021</sup>
Hello, great work on new UI/UX and upgrades.

I have a question about Manual mesh leveling. I created a mesh, but I am not sure if its applied correctly. I know it applies by default before the print start, but I also applied in Cura slicer with the `M420 S1` command.

Does "Leveling active" checkbox need to be ON in order to apply the Mesh leveling? And what is the feature for?

And another question, if the mesh is applied, and while we are printing, shouldn't the Z coordinate live update/show position at the bottom of screen same as X and Y is updated while hotend is moving?

I guess it should, but the Z coordinate is static for every layer. And what I understood it should compensate (move up or down) because we applied mesh leveling.

Looking forward to hearing your answers.

PEace,
</div>

Jyers, 2021-07-25:
Hi! All great questions

The leveling active checkbox is only for manually enabling leveling if you forgot to put M420 in your start gcode. Outside of this it isn't important for anything, if you have M420 S1 you're all good.

The z coordinate shown on screen is programmatically 'above' for lack of a better term, the offset applied by the mesh. So although the printer says it's at one height, that height is being offset as it moves around the bed. If you want to make sure your mesh is being applied I recommend watching the z coupler very closely as your printhead sweeps across the bed.

closed 2021-07-25 by Jyers

shtefcs, 2021-07-25:
Thanks for the clear answers brother!

>The leveling active checkbox is only for manually enabling leveling if you forgot to put M420 in your start gcode. Outside of this it isn't important for anything, if you have M420 S1 you're all good.

Thats what it came to my mind, but was not sure.

> The z coordinate shown on screen is programmatically 'above' for lack of a better term, the offset applied by the mesh. So although the printer says it's at one height, that height is being offset [...]

What do you think about putting the option in Leveling menu, that we can turn ON live Z positioning while the mesh is applied? Just a suggestion from my point of view on UX. Since X and Y are real-time updating with their positioning, I think Z should as well. At least as an optional thing.

Anyway, great work on UX so far.

DasSmeagle, 2021-07-26:
@shtefcs think again ;)

The display show "Z" above the bed. so it HAS to be zero alle the time, while in reality the head is moved. But the goal of the movement is to keep the head at same height relative to the bed! It corresponds to the height sent via "move" commands in GCode! I think, it is correct, you don't see the mesh in Z-value on display. 

Would make sense to show it in addition to Z and Offset, because Z + Offset + Mesh value together give the "real" height position. But I don't think it would help much, only you could see very quickly, if your mesh is enabled and working as expected. But you cannot do much about that value. So a "Mesh active"-Info would be as good. Or a look at the spindle turning while moving on x/y plane.