Jyers/Marlin #1224: Creating custom icon set with .bat file

dandevri, 2021-07-26:
### Description

Trying to create a 'monotone' icon set to go along with my black and white colour scheme of my Ender 3 V2. There seems to be a problem/bug with the `.bat`.

### Steps to Reproduce

* I have a folder full of `.jpeg` files same file names as the original icon set
* Renamed the folder to _monotone-Icons_ as stated in the readme.md
* Copied the Create-ICO.bat into that folder and run it
* The terminal flashes briefly but no `.ICO` file is added to the folder

tititopher68-dev, 2021-07-26:
@dandevri 
Is Python3 properly installed ?

dandevri, 2021-07-27:
@tititopher68-dev Ah, python installation is still a prerequisite even if you're using the `.bat` files? I actually got it working on my Mac by running the 'original' python scripts from @b-pub so I didn't use the `.bat` files in the end.

I'm not a regular Windows user so I didn't really know if the problem was with the .bat specifically files. But if they work with Python installed this issue can probably be closed.

closed 2021-07-27 by Jyers