Jyers/Marlin #1225: Different Z-offset compared to stock FW and inconsistent layer thickness across the bed with BLTouch

DataGhost, 2021-07-26:
Beginner to 3D printing here, so please forgive me if I've made any obvious mistakes. This issue is possibly related to issues #916 and #569.

I have an Ender3 V2 4.2.2 with BLTouch. The machine is mostly stock, except for a printed mount for a laser which offsets the BLTouch a bit. I have (I think) correctly measured the new offsets. I had the stock firmware (Ender-3 V2_BLTouch_Marlin-2.0.1 - V1.1.1.bin) and have printed probably around 50 items with no problem. I'm using G28 and G29 in my start gcode after heating up and I'm using Cura to slice 0.12mm layers. I only have PLA filament for now. I have dumped my Stock FW settings with M503, cleared EEPROM in JyersUI and then executed the stock dump to transfer my original settings, after which I selected to save them in the UI. These settings are what my machine was shipped with, I haven't changed any of them myself except the PLA temps.

Stock FW gives me F amount of friction doing the paper test on a leveled and heated machine at -2.22mm Z offset. First layer becomes a nice continuous sheet. JyersUI gives me the same F amount of friction at the same settings. However, first layers on JyersUI are consistently and noticeably worse (subsequent layers are fine), looking under-extruded and the extruded lines are not forming a smooth sheet but have gaps in them. 3DBenchy also gets a terrible first layer compared to stock FW. Decreasing Z offset on JyersUI to -2.25mm makes it impossible for me to properly do the paper test (nozzle dents the paper), however those prints turn out much like they did in stock FW on -2.22mm, both my tests and 3DBenchy. I can only assume this discrepancy shouldn't exist, or am I missing something? Switching back to stock FW with the same settings (-2.22mm) gives me good prints again. The JyersUI ones have consistently thicker first layers by up to 0.03mm (0.10 - 0.16mm, avg 0.14-ish) when printing a test in the center, as opposed to the Stock FW first layers (0.09mm - 0.14mm, avg 0.12-ish).

Just today I made another test print which is a big single-layer X over the whole print area, this gives me a very thin layer (0.06mm) at the top left corner and a rather thick one (0.16mm) in the bottom right. I haven't verified this specific on stock FW yet but I thought the mesh is supposed to compensate for this regardless. I do see Z turning slightly when I move the head from X0 to X220, so it must be doing something.

Purely based on good-looking prints from stock FW I feel that this can't be a bed misalignment or other mechanical issue and has to be FW-related. Are there any differences I should be aware of when calibrating my machine, or settings that shouldn't be transferred?

I have typed out a really long version with photos and graphs, but I decided it was too long to just chuck into my first ticket here. I've kept it in case I'm not making some stupid mistake and more info is required.

Probe test: deviation 0.001293

Mesh in stock:
```
      0      1      2
 0 -0.009 -0.032 +0.015
 1 +0.122 +0.022 +0.016
 2 +0.137 +0.032 -0.001
```

Mesh in JyersUI:
```
      0      1      2
 0 -0.026 -0.052 +0.010
 1 +0.098 +0.008 -0.007
 2 +0.082 -0.011 -0.052
```

Settings:
```
  G21    ; Units in mm (mm)
; Filament settings: Disabled
  M200 S0 D1.75
; Steps per unit:
  M92 X80.00 Y80.00 Z400.00 E93.00
; Maximum feedrates (units/s):
  M203 X500.00 Y500.00 Z10.00 E50.00
; Maximum Acceleration (units/s2):
  M201 X500.00 Y500.00 Z100.00 E5000.00
; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
  M204 P500.00 R1000.00 T500.00
; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
  M205 B20000.00 S0.00 T0.00 X8.00 Y8.00 Z0.40 E5.00
; Home offset:
  M206 X0.00 Y0.00 Z0.00
; Auto Bed Leveling:
  M420 S0 Z0.00
  G29 W I0 J0 Z-0.02650
  G29 W I1 J0 Z-0.05150
  G29 W I2 J0 Z0.01050
  G29 W I0 J1 Z0.09800
  G29 W I1 J1 Z0.00850
  G29 W I2 J1 Z-0.00700
  G29 W I0 J2 Z0.08250
  G29 W I1 J2 Z-0.01050
  G29 W I2 J2 Z-0.05200
; Material heatup parameters:
  M145 S0 H180.00 B45.00 F255
  M145 S1 H200.00 B60.00 F255
  M145 S2 H240.00 B70.00 F255
  M145 S3 H240.00 B70.00 F255
; PID settings:
  M301 P28.72 I2.62 D78.81
  M304 P462.10 I85.47 D624.59
; LCD Brightness:
  M251 B255
; Power-Loss Recovery:
  M413 S1
; Z-Probe Offset (mm):
  M851 X-43.20 Y-10.30 Z-2.25
; Filament load/unload lengths:
  M603 L0.00 U0.00
; Filament runout sensor:
  M412 S0 D0.00
```

Start G-code:
```
G92 E0 ; Reset Extruder

; Heat stuff while calibrating for the print
M140 S{material_bed_temperature} ; Start heating the bed
M105
G28 ; Home all axes
M190 S{material_bed_temperature} ; Wait for the bed to reach temperature

M104 S{material_print_temperature} ; Start heating nozzle
M105
G29 ; Auto bed-level (BL-Touch) after bed has heated up
G1 X0.1 Z15 F5000.0 ; Head left and up, for clearing later
M109 S{material_print_temperature} ; Wait for nozzle to heat up

; Enqueue three beeps to notify we're starting
M300 S880 P20
M300 S0 P100
M300 S988 P20
M300 S0 P100
M300 S1047 P20
M300 S0 P100

; Clear nozzle
M117 Cleaning nozzle
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish

M117 Printing...
```

tititopher68-dev, 2021-07-26:
@DataGhost 
It is not a firmware issue since you execute in your startGcode the command G29, which is also the one in the firmware.
 The stock firmware does not use the same version of Marlin as the Jyers firmware at all.  Since then, there have been a lot of changes, especially in the auto bed leveling.
Be careful not to activate in configuration_adv.h the babystep with M851 ...
 Repeat a test with a Bed Leveling carried out from the firmware with your printing temperatures (important factor) and I advise you to do the Z-Offset also like this.  In your Gcode, after the G28 make an M420 S1 ...
 Also make your mesh in 5x5, because the beds at Creality are often well deformed on the edges and the center.
 For your information, my first layers are always perfect whether in UBL or Bilinear ...

DataGhost, 2021-07-27:
I see I failed to include the exact firmware I'm using. I used the prebuilt binaries from here, E3V2-BLTouch-3x3-HS-v4.2.2-v1.3.4.bin first, and I suspected HS throwing off some measurements so I switched to E3V2-BLTouch-3x3-v4.2.2-v1.3.4.bin with the same results. These should be the closest to the Creality FW, right? That's why I was under the impression that doing G29 at the beginning of every print would just give me an as recent as possible mesh using the same method as the menu does.

I have done all bed leveling with heated bed, and Z-offset with heated bed and nozzle. Still, determining Z-offset shouldn't be affected by my start gcode at all, because I haven't run a print yet when I'm doing that. I thought that M420 S1 wasn't needed when the leveling checkbox is enabled in the menu, but I'll give that a go tomorrow and see if that improves anything.

Jyers, 2021-07-27:
Very interesting. As @tititopher68-dev, if there is any differences in the leveling code, it is from the main Marlin repo, not specifically this fork, as the Creality firmware is based of an older version of Marlin.

From what I can tell you're mesh seems fairly similar from what you had with stock firmware, just a tad lower, which could explain the need for a slightly different offset.

Perhaps try giving the 5x5 leveling a try and see if you get better results, there is a chance the mesh could be over compensating at the edges due to the low mesh reslolution.

DataGhost, 2021-07-27:
I guess this could only happen if the BLTouch trigger point is slightly different in both firmwares, right? That would effectively change the Z-offset. I'm still printing a bunch of square tests now with various settings on the machine, with and without G29 in the start gcode, and I'll do another run on stock to compare, so I'll update when those results are in. Still all on 3x3 because I wanted to change as little as possible going from firmware to firmware to try and exclude any errors on my part, but I'll run 5x5 after that as well.

phenomeus, 2021-07-27:
I'm also experiencing this behaviors since 1.3.4 only (somehow). I use HS 10x10 on my 4.2.2. e3v2 with BLtouch. 

my testprint (4 corner squares and one in the center) is off. everything looks good during manual leveling, leveling by hand etc. the mesh looks a little bit different than with the firmwares before. I always preheat my bed before doing any leveling or ubl. 

right now its the center: its not sticky as it should be, or as the corners are. right now I'm lost what to do.

Axionkt, 2021-07-28:
Feel free to disregard this if this is not helpful in the current conversation as I haven't spent time to check if the issue comes from me but I'm almost positive that the mesh isn't applied on my printer with no changes on my part in config when it used to work before.

My last compile was on commit f478a3d355 and I generated a new mesh.
I kept the same starting gcode but modified the slot as per 96e83129ed I get an inconsistent 1st layer and the Z coupler doesn't seem to be moving even when printing lines going from ~230,230 to ~0,0 when I am sure I could see it moving before. Since my bed is slightly lower in the center because of the clips, the print doesn't stick in the center but it is too close in the corners.

It could be because I was able to get a much flatter mesh with the addition of assisted tramming so I am not sure as using the console to load the mesh and display it it's definitely not flat.

I'm not really looking for help as I have not spent time on this problem yet but if you suspect there is actually a problem I thought I could maybe help give a time frame as to when this started to occur, which was before the latest pull from the Marlin repo.

phenomeus, 2021-07-28:
for me this is exactly the behaviour and it started with 1.3.4 only! prior that, UBL was mostly perfect. on slot 0, one mesh, set and forget. this is exactly what I have:  "...the print doesn't stick in the center but it is too close in the corners."

something changed.

DataGhost, 2021-07-28:
Well I may have been sloppy in remembering and doing my first tests. Flashing back to stock I tried my 9-square printing test and some squares were also thinner than others, most notably in the far left corner. So it seems the issue is present there as well, leaving only the z-offset in my initial report as a difference between stock. I can probably live with a slightly different trigger point for both firmwares if there have been changes/improvements in that area. I saw Z movement when moving X in the menu in both 3x3 and stock FW. I used the startgcode from my initial post for these tests.

I have tried the 5x5HS firmware which only marginally improved the issue (but that could be run-to-run variance). Still here the top-left square is much thinner than the others, so I didn't see that as any improvement and did not do any other tests on that firmware. I used the startgcode from my initial post for this test.

I then tried UBL15x15HS and that gives me the same thin top-left square. Issuing move X on the display does not result in any Z movement most of the time, with the mesh saved, leveling active and various settings of fade (0, 10, 250). Only sometimes does Z move (or seem to move) but moving X back to the previous position immediately after then does not result in any Z movement anymore. I tried homing a few times in between to see if that changed anything but to no avail. I have also saved mesh and settings a couple of times. For the tests with UBL, I changed the G29 in my original startgcode to
```
G29 L1
G29 A
G29 J2
```

The attached 15x15 mesh shows that the top-left corner is higher than the rest of the platform, which is mostly very close to 0.0. That would suggest the mesh isn't being used despite the settings on the machine and in my gcode. One of the runs was particularly bad where it printed the top center square fine, and then proceeded to print an insanely thin top left, causing the extruder to skip because the nozzle was blocked by the glass plate. I don't think this should ever happen with any leveling, right? Especially not 15x15 UBL.

I'm kind of at a loss as to what else I can try. What fade settings should I use for maximum ABL/UBL effect (never stop applying the mesh), 250 I presume? Is there some gcode already I can run to visually verify UBL is active and working? Else I can try to hack up some of my own later, I'm guessing heatup, G28, G29 L1, G29 A, G29 J2 and then some G0/G1 commands to move X and Y around at Z0.2 without E?

DataGhost, 2021-07-29:
I've written some gcode (attached) that just moves the head around. It's pretty clear that at Z0.2, the Z axis is moving quite a lot while X and Y are changing, so that's good, the mesh is being used. There is a small gap between the nozzle and the bed, as expected. However, at least in the top left corner (mesh point 0,14) the nozzle seems to touch the bed or come very close to touching it. That's consistent with my squares test prints being very thin there and sometimes even getting a skipping extruder there. Seems like the mesh isn't being applied as much as it should be, at least in that area. It's not exclusive to that point, it's happening in the region up to there already where the space between the bed and nozzle gradually gets closer until fully in that corner. This is most easily visible when it's at X5 moving Y only, or when moving the bottom right to top left part of the cross, starting to get close about where the mesh starts turning green. So again, seems like the "strength" of the mesh application is insufficient. Moving on to Z10 and Z50 there is no movement in Z anymore, which seems consistent with fade=10.

I decided to re-measure the mesh. The first time I measured the mesh (two days ago) I had the serial console open. I noticed the following output hours after measuring:
```
Probing mesh point 157/225.
echo:busy: processing
X:196.40 Y:130.60 Z:12.25 E:0.00 Count X:15712 Y:10448 Z:4900
echo:Bed Leveling ON
echo:Fade Height 10.00
Done.
```
and didn't remember what I did to cause that, whether I turned off my PC in the meantime. Just to make sure, I re-measured it today and kept an eye on it. It again stopped after 156 points on screen and 157 points in the console. "Probed all reachable points" but I've seen it being close-ish to the edges of that top-left corner around point 150 or so (didn't pay too much attention unfortunately) though it looked like there was still room for at least one row of measuring points more, and I can't really believe that 68 measuring points are unreachable. Going into manual mesh tuning, mesh point 14,14 is at 220x220 and 13,13 is at 205x205. Moving to 245x230 brings my probe (offset -43.2, -10.3) at approx 201x219 (rounded down). So the furthest reachable point is 12,13. Given the position of my probe, 0,0 is always reachable. That makes the only the top row unreachable as well as the two right-most columns, for a total of 30+15-2 = 43 points, not 68. So somehow, somewhere, it's excluding some additional 15 points. I'm assuming none of these missing points coincide with the cross I'm making in my attached test gcode, so the distance between the bed and nozzle should stay the same throughout most of that movement. I guess I should manually tune those missing points, or make a thinner laser mount so my probe can probe the top row too (it's only 0.3mm off). What bothers me though, is that I don't know how the missing points have been filled in and there is no feedback (in e.g. the mesh viewer) that some points weren't measured, and which ones they are. Outlining those points in the mesh viewer should make it clear that they couldn't be measured and are actually calculated. Just to clarify, I repeated the movement test after the re-measuring and it showed the same results.

Still though, reachable mesh point 0,13 is in the middle of the top-left square, and the thinnest part of said square is actually between points 0,13 and 0,12, so that doesn't seem right. I can understand the part between 0,14 and 0,13 to be wrong, but not the measured parts. Unless of course one or both of 0,12 and 0,13 weren't measured either.

Is there a way for me to view which points were measured and which were calculated, so I know which ones I need to manually tune? Or could this be added as a feature? I'd say outlining them in the mesh viewer is a good start. Also during leveling the message could be changed to ```Probing mesh point 157/225 (Xx Yy).``` and a list of skipped/unreachable points could be printed after (or before) leveling.

Is it possible to measure a single mesh point using the BLTouch in an existing mesh using some gcode? So I don't have to go through the full 157 points when I don't even know if it's going to measure the point I want it to or if that's one of the skipped ones. Preferably automatic, but if I can have it BLTouch at manually chosen coordinates, that's fine, too.

And is there a way for me to print "real" coordinates? Like when I move to Xx Yy Zz with an active mesh it actually moves to Xx Yy Zw because of the mesh application (so w = z + m with m being the offset calculated by the mesh). My intuition says m should be exactly the value of the mesh point when x and y are exactly on it, but I suspect it to have a different value for some reason. Either that, or the initial measurement is wrong somehow.

[ablubltest.gcode.txt](https://github.com/Jyers/Marlin/files/6900677/ablubltest.gcode.txt)

tititopher68-dev, 2021-07-29:
@DataGhost 
Except that you forgot the probe_marging value in your reasoning, hence your 15 missing points ...

tititopher68-dev, 2021-07-29:
@DataGhost 
By modifying my Probe_marging, my Inset value and my x_max_pos, I reach 201 points out of 225 ...

DataGhost, 2021-07-29:
15 doesn't completely make sense though. I can understand 13 or 14 because the top row and rightmost two columns have been excluded already, so 15 points doesn't make a full row or column. Or it's off by one.

I do see the probing margin in the config, not sure what it does exactly. Looking at the 10x10 config (I don't see the 15x15 one, and I'm still using the binary releases, I haven't compiled them myself yet) it seems to be set to 20. It's got X_BED_SIZE 230 and X_MAX_POS at X_BED_SIZE+15, which is consistent with me being able to move X to 245. Probing point 0,0 is at X10 Y10 and they go on at 15 increments. If the PROBING_MARGIN of 20 actually excludes the 0 row and column, then I'd find it strange to brand this firmware as 15x15, because it's only ever going to be 14x14. Also the 0 row and column are reachable perfectly fine by the probe so I can only assume it's going to have 10 for PROBING_MARGIN in that config. I don't see why any other points than the ones physically unreachable should be unreachable.

201 out of 225 also makes no sense, it's not a square or even a rectangle. 14x14 = 196, 13x15 = 195, 14x15 = 210. In my case, it should (be able to) measure 12x14 = 168 points, not 157. I'm just saying that from a mathematical point of view, I have not looked at the code yet and I can't think of any reason to have a non-rectangular grid by excluding sides partially. So I'm pretty interested to see which points it then excludes.

tititopher68-dev, 2021-07-29:
@DataGhost 
There is a misunderstanding of the operating principle of the UBL.  Your reasoning is incorrect.
 First of all, it is not the firmware that calculates the reachable points or not.  It's all in the Marlin base.  Also if you feel that there is a problem, I invite you to post your issue (which is not ...) on the GitHub Marlin page ...
 In UBL mode, Probing_margin and Mesh_inset are in configuration.h
 At home for example, my Probing_Margin is at 5 and my Mesh_Inset at 10.
 My X_MAX_POS is at 249 (230 + 19).  My probe_x and y offset is -40, -7 ...
 The probing_margin is a margin which reduces the surface of the plate in X and Y by 10mm or 5mm per side.

 In UBL, the bed is effectively divided into X x Y points and the area is measured in a spiral and not in a Cartesian way (this is also important) from the center to the outside.

 Initially, your Bed size will be reduced by 2x the value of Probe_marging so if 230x230 -> 220x220 (with my values).
 Then the margin for the Mesh will be applied, i.e. the Mesh_inset, which further reduces the size of the measured Bed (from the values ​​of the Bed Size), for me 210x210 ... with a Mesh_inset of 10.
 In the end, given that the size of the 210x210 mesh will be retained, then that my inset is greater than my Probe_marging ... so the trick is already to have a Probe_marging <Mesh_inset to increase the number of reachable points.

 And finally the mesh is calculated centered on this last value.

 So already on the left, a minimum band (with my values) of 15mm will not be measured and therefore considered unreachable ... and on the right it will also be 15mm for me.
 The greater your probe offset, the less will be the possibility of measuring the right part of the Bed, because in the end, it is not the position of the probe that counts but the nozzle.  Even if the probe can go all the way to the edge of the bed, the nozzle will not.
 Also, the lower your Offset probe, the more reachable points will be.
 Then if your X axis is not perpendicular to the Z axis (measure with a square the position of the X axis on the left and right side of the profile, very often you will have an offset of 0.5 to almost 1mm unless you  install a dual Z or if you geometrically calibrate your axes, there are very good videos on Youtube for the Ender3V2 when editing like the Yonet Tournois channel), you will have high values ​​on the left on the result of your Mesh and that will also play  on your first few layers.

DataGhost, 2021-07-29:
I made two mistakes in my posts, sorry. I calculated 43 points to be unreachable while the firmware refused to probe 68 of them, but the difference between those two numbers is not 15 as I said, it's 25. The second mistake was later on where I said 12x14=168 points should be reachable where that should have been 13x14=182 points, bringing it back in line with 225-43=182.

I don't fully agree with your explanation, at least not on the row 0 and column 0. Please correct me on any points that I'm wrong, though. I'm still just learning and figuring out printing, this project and its code. I have learned a lot about my printer and this firmware from this ticket alone already.

I found Marlin-specific G-code documentation that was a bit better than an apparently outdated generic documentation page which I used before. I also had a quick look into the code to determine what was going on. I ended up using G29 P1 T V4 to get a visual log of the leveling process. The end result is posted below. One issue in the code is that it claims to be probing point 157, but only decides not to do that after printing that message (off by one). So in reality it has only measured 156 points, in a 12x13 grid. It would have been easier for me to see this had I not made the mistake of thinking 15 points were missing instead of 25, because that would be much closer to rectangular. This off-by-one (156 vs 157) is a bug, though, not a very severe/important one but a bug nonetheless. But it's one in the core so I'll probably have to report that with Marlin then. In any case, this made it easier to understand why this thin printing happened. The top-left square in my test print is 5x195 .. 25x215. The measured mesh point closest to that is (1,13) at 25x205, so just touching the right side of the square, while the rest of the square is in a completely unmeasured part in probably the most distorted part of my bed. An unlucky coming-together of circumstances.

The order in which points are measured is in a way that vaguely resembles a spiral, yes, and that is done from the current X and Y, not necessarily the center (I started from X0 Y0 once and it started probing from the corner outward). The reason for that I've come to understand is enabling quick checks on small centered prints when starting the probing from the center. That does not seem important at all in this case, I just want an extensive as possible measurement. Anyway, that does not change the location or reachability of any of them. The points are still laid out in a square grid, so any measured grid should resemble a rectangle, not any other shape. The amount of measured points therefore must be a multiple of two integers in the range 0..15 inclusive. So I really don't understand how you are getting a 200- or 201-point grid. The only possible multiple other than 225 is 210 which would be shown as 211 in the console. Is that what you meant? Otherwise I'm really interested to see what it looks like for you. If I make my laser holder 0.3mm thinner I should theoretically be able to get 210 measured points as well.

Back to the grid I measured, it's missing the 0 row and 0 column, both of which are fully reachable by both the probe and the nozzle. For some reason it's been excluded but I cannot check why without having the UBL-BLTouch-15x15-HS config, which is not present in the repository. The 0 row and 0 column should never be excluded with the default probe mounting location, and not with my slightly offset location, either. Looking at the values that are filled in by G29 P3 and the resulting print, it looks like my bed is pretty warped at (0,12), (0,13) and (0,14), and possibly in some other places too, so the calculated values for them are wrong. That would make it important for me to have this column measured, and probably at as many points as possible, so I'd like to stay on 15x15, but I can't easily compile it myself because the configuration files for that specific build aren't in the repository.

If I understand your explanation correctly, even the 10x10 config is in the repo wrong, since it has MESH_INSET 10 and PROBING_MARGIN 20, so that would also exclude the outer rows and columns and be at most 8x8 in reality if I'm correct? I would consider that a bug in JyersUI, at least in the configuration of the releases. If the 15x15 release ships with a similar configuration that effectively makes it a 13x13 grid at most, if all points were reachable by both the probe and the nozzle. So then both are wrong probably. I just don't know which constants were changed in the configuration compared to 10x10 (also which constants enable HS), so I'd really appreciate it if that config was committed to the repository as well, so I can base my custom config off of it and build 1.3.4 again with different constants. I can then adjust my probe slightly and adjust the margin (probably) to measure everything except column 14 to get these 210 points.

With regards to the X axis not being perpendicular, I feel like this should not matter (much). At any given X x1 the angle of the axis with respect to Z would be a1, and at x2 would be a2, with probably a1 != a2. But the height h1 of the nozzle at x1 to the bed will be measured at (well, close to) x1, and h2 at x2. It's likely that, with a perfectly flat and level bed, a sagging X axis has h2<h1 for x2>x1, but this will be measured by the probe and reflected in the measured mesh so the machine will/should be able to compensate for that wi [... cut ...]
 1 |   .     +0.016  +0.010  +0.007  -0.027  -0.026  -0.016  -0.016  -0.021  -0.012  +0.021  +0.022  +0.043    .       .    
   |
 0 |   .       .       .       .       .       .       .       .       .       .       .       .       .       .       .    
        0       1       2       3       4       5       6       7       8       9      10      11      12      13      14      
    ( 10, 10)                                                                                                      (220, 10)
```

Jyers, 2021-07-30:
I'm going to be honest, I didn't read all of that... Yall are writing books in here 😅 

But I did skim it enough to cover the basic issues in a concise way here:

UBL is not designed to probe every point, that's not the principal of it. But just because it doesn't probe a 15x15 grid, doesn't make it incorrect to label the it a 15x15 firmware. The grid size IS 15x15, that is the resolution of the grid, that is what you see in the mesh viewer, and therefore that is the grid size for which the firmware is labeled. 

The UBL mesh is supposed to cover the whole bed. Why? Because that way (if the user so wants to) they can manually measure the points the probe can not reach. Just because some "unreachable" points can't be measured doesn't make your entire mesh invalid. Now why do I say "unreachable" This is because the 20 mm probing margin is a safe estimate, many people use binder clips and other such decently sized implements to hold their bed down, meaning that in order for the probe no to hit them the probe margin is set to a reasonable value. The mesh inset, is set to cover most of the bed. Up to 10 mm from the edge to be exact. This allows for a decent large mesh that can be manually calibrated around the edges if the user chooses but still provides a large number of probe points.

Now for what I can do, here are my thoughts: I like the idea of adding an option to the leveling menu to not calculate unreachable points and measure them yourself if you would like. It may be overkill for most cases, but I think it would be a nice option. I also definitely think I could add an outline to mesh points that have been calculated in the mesh viewer, it shouldn't be too difficult and could help some users. I'll definitely put there on the todo list and look into them in the future.

Lastly the reason there are no 15x15, 5x5, etc firmware examples is because it's one line difference:
`#define MAX_GRID_POINTS_X 10` -> `#define MAX_GRID_POINTS_X 15`
I know not everyone knows configs, but a simple search for "marlin config change grid size" would get you that answer in seconds.

Oh dear, I've made a bit of a book myself... Hopefully it's an informative book at least 😄

DataGhost, 2021-07-30:
Mkay, that's probably something I could have found if I dived a bit deeper into the working of UBL and the exact meaning of everything. I was just going off my own assumptions as to what everything is and is supposed to do, and extrapolated the 3x3 measuring 9 and 5x5 measuring 25 points to mean that 15x15 should measure 225 (as it also said in the progress), and at least not skip any that look reachable to me. So that's mostly my fault for not researching enough, I do think part of what you said here could/should be included in a few lines on the wiki or on the releases page, to get these expectations straight. It may also be worth making things like the margin configurable in an advanced config menu (probably low prio) so recompilation and reflashing isn't required when changing the type of bed clips used, for example. I'm not sure if people do this often, it'll make using prebuilt binaries more flexible though. The copious amount of hardcoded constants could make this difficult, unfortunately.

As my bed clips are completely stock, the default margin is way too much but your arguments for using it in binary distributions seem reasonable. For the config I was indeed hoping it'd be that simple, they're just pretty huge with a lot of constants and derived constants that directly or indirectly influence each other, and a large amount of fine-tuning options so I feared it would be a more elaborate process. The easy way (the programmer way) is just to copy what's there and make minimal adjustments :P but as you say there's just one difference I'll give it a crack tomorrow with the 10x10, grid points to 15 and margin to 5 (should be good for my bed). I'll have a search for which constant(s) change(s) HS then too and see how that improves my prints.

Good to see that at least something came out of this and ended up on a todo list :) the outline in the mesh viewer is probably fairly easy, as the firmware already knows which points it's going to skip, so that can be a simple calculation requiring no storage. The most elegant (and probably overkill) way would be to indicate for each point whether it was probed, manually set, interpolated or unset but I don't know how well-equipped this hardware is in terms of storage. That'll add 2 bits per mesh point (best-case), so about 60 bytes for 15x15.

And yes, thanks both for explaining a lot.

tititopher68-dev, 2021-07-30:
I will also see if the Probe_marging and Mesh_inset parameters could be configurable from the interface.  Another Peruvian developer (taking other people's code without sharing his own by allowing himself to modify the Marlin license ... for that! A shame) made the size of the bed and the number of leveling points editable from the interface  .  I do not adhere to this approach because on the one hand, it heavily affects the Marlin base code, which is not in Marlin's mind and complicated maintenance, on the other hand, we do not all change  the days the size of the bed or the number of leveling points.  With a BLTouch and in bilinear mode, a 3x3, better a 5x5 for a Bed from 220 to 235mm, this is more than enough.  And in UBL, a 10x10 is more than enough.  Beyond that with beds of more than 300mm, a 7x7 gets along and a 15x15 in UBL too ... and frankly, installing VSCode and 2 plugins + Python is not a totally insurmountable difficulty to compile your own configuration.
 (I am ready to help those who have difficulty with this, as I already do with my own build).

badbod, 2021-08-10:
For UBL 
All probe points exist within the mesh, mesh size is based on bed size - mesh inset.

Probe margin, you are reduced the probe reachable area of the mesh. The mesh is bed size regardless of margin, you have just told the the probe not to go there.

mesh inset, you reduce the mesh size, all probe points remain reachable within the mesh.

So,  if you have 10x10 mesh you have 100 probe points
Bed is, say, 300x300 then mesh is 300x300 area of 100 probe points
if you set mesh_inset=20 you now have a mesh of 280x280 of 100 probe points.

Ideally you want probe margin 0 or 
      //#define PROBING_MARGIN 0

Use mesh_inset  to keep the probe away from the edges and have the maximum reachable probe points. 

With unmodified Ender 3 V2 you have a problem with some ~30mm of the right hand side of the bed as unreachable.  

You will probably want something like
  configuration.conf
//#define PROBING_MARGIN 0
#define MESH_INSET 5

  adv_configuration.conf
#define PROBING_MARGIN_RIGHT 30

UBL will guess the missing points for you but you cant avoid that unless you modify the hardware so you can reach X=280.

7x7 is plenty in UBL for such a small bed.

If you really don't want to play or compile, stick with Bilinear. 3x3 or 5x5  (whatever can reach all the probe points) 

Just remember - if it cant be probed it has to be guessed so unlikely to be particularly accurate or repeatable.

JayBee-git, 2021-09-23:
I can confirm the btouch auto leveling on 1.3.5 is inconsistent. It does work perfectly on 1.2.2 and official firmware, but with the latest one on first layer with either mesh or G29 at each print, some areas are too thin and other too thick. Either the auto leveling isn't working correctly or not activating properly.
The Z offset is also different than the official one, but same as 1.2.2 so I guess that's normal.