Jyers/Marlin #1228: Z offset keeps changing

Mountain3DPrints, 2021-07-27:
Hey guys. 

I've got three 3 v2's. 1 5pro and 10 v3.

All of the 3s on Jyres. Same fw. I literally did them all at the same time. 

My 2nd machine is my problem child. I'm constantly messing with it. It is always something
 My main issue right now is every single time I run a print after setting offset, via in menu or forcing it through in pronterface, is changing every run. This problem has persisted pretty much from the beginning on this machine. I'm to mt eyeballs. 

TIA

anilgunduzz, 2021-07-27:
Hi, I think I have a similiar problem and I noticed my z-axis switch is not working properly. Z-axis is stopping without hearing the click noise. I ordered Creality CR-Touch recently. I'll test again without the Z-axis switch when the probe arrives.

TordyCapz, 2021-07-27:
I have a similar problem.
When I create the mesh with UBL, while I am raising or lowering the Z axis, it stops working (No click sound).
I have to go back to a previous point and then go back to where I was to check that I am at the correct offset.

tititopher68-dev, 2021-07-28:
@TordyCapz @anilgunduzz @ Mountain3DPrints,
 Please be much more precise in "your issue", in particular give your configurations (firmware versions, precompiled or not, type of probe, etc ...).
 @ Mountain3DPrints, no firmware issue, why?  Again, you say you have the same problem using Pronterface (which sends only Gcodes ! and which has nothing to do near or far with the firmware).  Once again, before implying the firmware in question, please start by checking your hardware and the physical calibration of your printer ...

Mountain3DPrints, 2021-07-28:
Ok. I wouldn't be reaching out if I hadn't done all the necessary physical
checks and as much troubleshooting as I could.  I guess I just assumed that
it was implied? I am running Jyres 1.3.4 or close to otherwise why would I
be in this thread. As far as pronterface goes I will set the offset pushed
through proter and then saved to eeprom. Do a print fine. Load up another
one and it's like I never did anything.

Mountain3DPrints, 2021-07-28:
I was setting up my frankenprinter ( cr10 v3 running configs and hardware
for an Ender 3 v2) and found 2 limit switched behaving like this. They also
through an extra 3 pin switch in the box?
 Anyways, I was doing my final assembly checklist and found Z crashing in
to the end of the travel. I got around to the otherwise after killing
power, moved the bed back and came to find this switch that would click if
I depressed with my fanger, but would not depress under tram.

What did I do?

If you take your switch you will notice the little fella has its own fanger
coming off that contacts the bed. A guy could pull the fanger out just ever
slow slight as to not cause irreparable damage and bricking the component.
This will effectively give you a better contact point and should work.

As far as you ordering touch sensor,

   I highly recommend you either slice both your touch sensor and you
factor switch in parralel or better yet find a dupont 2-1 connector. This
is the best method as you don't have to cut any wires and it's still plug
and play.

I recommend doing this because if that touch sensor fails at all you still
have the mechanical switch to catch the impending crash. The best thing to
do is just set the mechanical switch a tiny little bit (1.5-3 mil) below
where it was at so you don't have to crank the z offset for the touch
sensor.

The touch sensor is great. When it works. They are notorious for misses and
just crashing tools in general.

Redundancy Redundancy Redundancy as my good friend and fellow youtubeer
@CaptainJoe ways says.

If you are savy you can go in to the build and uncomment the z axis endstop
also.
They would both work and that way you and your bed are safe!

Happy Printing!

TordyCapz, 2021-07-28:
> @TordyCapz @anilgunduzz @ Mountain3DPrints,
> Please be much more precise in "your issue", in particular give your configurations (firmware versions, precompiled or not, type of probe, etc ...).

Oh, sorry...
Firmware version: 1.3.4, precompiled, UBL - NoProbe - 5x5 - v4.2.2
Printer: Stock, with no modifications, no probe, etc...
I tested, and I was doing tests, and this also happened to me with the ManualMesh firmware. And printing I have no problems with the Z axis. Only when I want to make an adjustment in the mesh, or create a new one.

Soltysrysik, 2021-08-01:
new firmaware 1.3.5. fixed that

Grenalis, 2021-08-19:
I had that issue and ordered switches.  But I thought with a BL OR CR touch you unplug Z stop switch.

Krakatomi, 2021-09-15:
I think that we are suffering this bugs: [#21833](https://github.com/MarlinFirmware/Marlin/issues/21833) [#22322](https://github.com/MarlinFirmware/Marlin/issues/22322)
As the last edit from[ credomane](https://github.com/MarlinFirmware/Marlin/issues/21833#issuecomment-913965078) say, it's apparently fixed in the [bugfix-2.0.x](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) Marlin branch.