Jyers/Marlin #1251: 1.3.5 - Manual Mesh Leveling with Probe Centre is right side

Tw0bit, 2021-07-31:
### Description

I am using 1.3.5 3x3 HS for V4.2.2 board. In the Manual Bed Level (not the icon 4 Levelling/Mesh area, but rather under icon 2), with the PROBE enabled the CENTRE is at X=230mm (Right Side). With PROBE disabled, it goes to CENTRE as expected

### Steps to Reproduce

1. Go into Icon 2,
2. Enter the Manual Bed Level
3. Without the PROBE, go to Centre - CORRECT LOCATION.
4. With PROBE, its to the right

**Expected behavior:**
I would expect the PROBE to be CENTRE ON X, i.e. 230/2 + OFFSET

**Actual behavior:** [What actually happens]

With PROBE on, the CENTRE is all the way right.

vernonbc, 2021-07-31:
I am having the same result when "centre" is selected after performing a Manual Bed Level with probe activated.
I am using 1.3.5 5x5 HS for 4.2.7 board.

tititopher68-dev, 2021-07-31:
@vernonbc @Tw0bit 
You are right !

To fix this issue -> just this line to modify : 

sprintf_P(cmd, PSTR("G0 F4000\nG0 Z10\nG0 X%s Y%s"), dtostrf(X_MAX_POS/2.0f - probe.offset.x, 1, 3, str_1), dtostrf(Y_MAX_POS/2.0f - probe.offset.y, 1, 3, str_2));

by : 

 sprintf_P(cmd, PSTR("G0 F4000\nG0 Z10\nG0 X%s Y%s"), dtostrf((X_BED_SIZE + X_MIN_POS)/2.0f - probe.offset.x, 1, 3, str_1), dtostrf((Y_BED_SIZE + Y_MIN_POS)/2.0f - probe.offset.y, 1, 3, str_2));

Jyers, 2021-08-02:
Oops, thought I had removed all my uses of MAX_POS 😅 

I'll get that fixed