Jyers/Marlin #1253: Leveling "active" is not restored when repowering the printer

AlphaDE, 2021-08-01:
### Description

With Jyers 1.3.5, BLtouch activation and mesh are not restored when powering on the printer (Ender 3, 4.2.7 mainboard, Ender 3 V2 display, so from firmware, it's an Ender 3 V2, latest E3V2-BLTouch-3x3-HS-v4.2.7-v1.3.5 just reflashed to reproduce).

### Steps to Reproduce

1. Flash firmware, revert to default settings.
2. Reconfigure Z offset to previous value and save
2. Create New Mesh (Auto Bed Leveling)
3. Confirm to store and view new Mesh (Leveling is now ticked)
4. Store settings manuall from Control menu
4. Power off printer
5. Power on printer
6. No initialization of BLtouch (keeps blinking red, no movement), leveling is not enabled, Mesh though can be viewed and is the one stored before.

**Expected behavior:** 

After repowering, set leveling to active and initialize BLtouch.

**Actual behavior:** 

Leveling is not activated after repowering and BLtouch not initialized.

tititopher68-dev, 2021-08-01:
@AlphaDE 
No issue. Just the first time to do a store settings...

AlphaDE, 2021-08-01:
@tititopher68-dev I didn't get that.... the auto leveling stores the settings, I do it again manually to ensure everything is saved, but BLtouch status is not restored and BLtouch not initialized at startup. So what to change to fix it?

tititopher68-dev, 2021-08-01:
No, the Bed Leveling status is deactivated because your BLTouch is in alarm (keep blinking red).  This is why it does not initialize.  But this is not due to the firmware.  It's a hardware problem.
 Did you use the 5-pin port dedicated to BLTouch, if so, be aware that some motherboards have a 5v power supply problem for this port.  This is certainly what is happening.  There is a hardware fix for this which consists of soldering a wire with a 10Kohms resistance between the 5v of the Runout sensor connector and the 5v of the dedicated BLTouch port.

AlphaDE, 2021-08-01:
I use both the dedicated ABL port and the runout port (Bltouch and Ezout). 

I check when I am back home. I have a spare motherboard (4.2.7) which I ordered recently to have a backup. I switch the boards to see whether that helps as my E3V2 does not show this issue.

But why does the BLTouch works fine afterwards?

Could there be a connection with the Ezout?

tititopher68-dev, 2021-08-01:
@AlphaDE 
Like what with more precise information, we see more clearly ... OK, you must not use 2 leveling probes at the same time, hence your problem at initialization ... Please nnplug your ABL and everything should be fine in order ;-)

AlphaDE, 2021-08-02:
@tititopher68-dev Just to clearify, I am not using two Z probes, I am using one Z probes (BLtouch) and one filament sensor each using it's dedicated port on the 4.2.7 mainboard.

tititopher68-dev, 2021-08-02:
@AlphaDE 
Your issue is certainly your ABL. It's an electromagnetic coil type probe and requires much more energy than a simple filament detector (single contactor) ... I understand correctly, but your ABL blocks, by its 5v supply, the initialization of the BLTouch  .  Do not forget that the runout port and the shared BLTouch port the same 5v source ... Also unplug it and you should no longer have any issue with the BLTouch at initialization.

Jyers, 2021-08-02:
Just to throw in one extra note here, leveling will never be activated on reboot, marlin's behavior is to disable it by default. This is why you need to enable it in your start gcode. The other issue is definitely a hardware issue

AlphaDE, 2021-08-09:
@tititopher68-dev 
During the weekend, I made some checks:
- I disconnected the filament runout sensor (which I called EZout, perhaps erronously and which is connected to the right port with the correct wiring and does work when connected). No change
- I replaced the 4.2.7 motherboard with a new one. No change.
- I checked the connection of the Bltouch. No change.

It remains strange.

tititopher68-dev, 2021-08-10:
@AlphaDE 
Strange !
Perhaps the power supply .... Check if the connectors are tight.

AlphaDE, 2021-08-13:
The connectors seemed to be ok, but I found something else. In another issue here, grounding was pointed out as an issue. As I equipped my ender with a dual z axis solution, the power supply was moved to the side to give room for the stepper motor. By that, the screws connecting the frame bare metal to the power supply housing were not used anymore and the power supply was mounted into plastic holders without direct connection to the frame. I added a grounding cable from the power supply housing to the frame on the back and for the first attempts, the initialisation now works.

ShivamJoker, 2021-09-12:
I am also having issue with restoring the print on power loss. It starts printing in the mid air 😐