Jyers/Marlin #1258: extra text on the screen

NpFaith, 2021-08-01:
### Description

Extra text on the screen, you can see on the image.

### Steps to Reproduce

1. Printing from Ultimaker Cura 4.10 and send it to Octoprint in Raspberry
2. Printing from Octoprint with Dashboard Plug-in
3. After some layers it appears in the screen

**Expected behavior:** [What you expect to happen]

Last version didn't have this problem.

**Actual behavior:** [What actually happens]

[image]

#### Additional Information
I'm using the E3V2-ManualMesh-5x5-v4.2.2-v1.3.5.bin without any change.

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tititopher68-dev, 2021-08-01:
@NpFaith 
No firmware issue but certainly plugin issue.

NpFaith, 2021-08-02:
I will test more and I will report after, thanks for your time and keep the
good work. I really like this firmware.

Jyers, 2021-08-02:
Let's consolidate this to #1263

Almost definitely the same kind of bug going on

closed 2021-08-02 by Jyers

NpFaith, 2021-08-02:
I'm printing now on the 3x3 manual mesh and I didn't have problems yet. I
think 5x5 have the bug but I will print more and give you feedback. Thanks.