Jyers/Marlin #1259: BLTouch, UBL, Start G-Code, Z Stop Switch

Shortwayz, 2021-08-01:
Hi! Newbie on the 3D printer here.

I am a bit confused with all the people seemingly doing different set ups concerning the items in my title. So just a few straight questions...

I installed an Antclab BLTouch and installed the "UBL-BLTouch- 10x10" firmware:

1. Do I need to adjust the Start G-code as well?

2. Do I need to remove the Z-Stop Switch?

vascofazza, 2021-08-01:
Adjusting the G-code is desirable and advised in order to automatically enable the automatic leveling and mesh before each print.
Yes, you need to remove the Z-stop switch and calibrate the Z-offset before starting any print. (just the first time)

Shortwayz, 2021-08-01:
Ah so it does not use the saved mesh if you do not tell it to do so in the G-code?

tititopher68-dev, 2021-08-01:
@Shortwayz 
Exactly. In your StartGcode, after G28 ;
Either add :
M420 S1 ;
Or : 
G29 L0 ; load the mesh in slot 0
G29 A ; Active Leveling
G29 J2 ; autotilit the mesh (see Discussion #1223...)

Shortwayz, 2021-08-01:
> Discussion

Thanks very much.

Shortwayz, 2021-08-01:
Thank you both!

Jyers, 2021-08-02:
Looks like you got this sorted :)

closed 2021-08-02 by Jyers

Shortwayz, 2021-08-05:
> @Shortwayz 
> 
> Exactly. In your StartGcode, after G28 ;
> 
> Either add :
> 
> M420 S1 ;
> 
> Or : 
> 
> G29 L0 ; load the mesh in slot 0
> 
> G29 A ; Active Leveling
> 
> G29 J2 ; autotilit the mesh (see Discussion #1223...)
> 
> 

Hi, One more question ...

What exactly is the difference between using the M420 S1 and the G29 L0 code?

Is there a reason to use the one or the other?

Jyers, 2021-08-05:
G29 L0 and such are the preferred method for UBL. This is because it makes sure you load the right mesh (UBL allows for multiple meshes) and also auto-tilts the mesh before printing, which makes sure you mesh matches the current tilt of your bed.

Shortwayz, 2021-08-06:
I was under the impression that G29 was only used if probing before each print!?! But that's not the case then?
I can just recall Slot 1 mesh, which was previously stored, with G29 L1, then activate UBL with G29 A and finally check Tilt with G29 J2 ...
and it will start using the stored mesh for that print?

Jyers, 2021-08-06:
G29 in a bilinear config will probe the bed, this is not the case with UBL. We use those three UBL commands just as you described to load and tilt the mesh before each print.