Jyers/Marlin #1263: glitch on the screen

AkidDinie, 2021-08-02:
### Description

### Steps to Reproduce

1.slice as usual using a prusaslicer and export g code to Micro sd card
2. printing using sd card
3. after a few layers this glitch appears on the screen

**Expected behavior:** [What you expect to happen]
print as usual with no problems
**Actual behavior:** [What actually happens]
[image]

#### Additional Information
The previous version did not have this problem (1.3.4)

Now I'm using the E3V2-ManualMesh-5x5-v4.2.2-v1.3.5.bin

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tititopher68-dev, 2021-08-02:
@AkidDinie 
Sorry but no firmware issue. -> certainly LCD issue. Please check connexion and flash again your LCD display.

Mineplayerminer, 2021-08-02:
I had missing buttons and texts, after restarting and cleaning the pins on the LCD, it fixed my problem.

Jyers, 2021-08-02:
I still want to believe this is not a firmware issue, but I've had quite a few reports of this lately so I'm going to leave this open so we can try to find the best (probably physical) way to fix this.

Bezur-Ro, 2021-08-02:
Same problem happens for me usually about 30-40 min in

AkidDinie, 2021-08-03:
>@tititopher68-dev
>I've tried this method..and it seems to help to solve the problem .. thanks

AkidDinie, 2021-08-03:
I've tried this method ias (tititopher68-dev) said and it seems to help to solve the problem .. thanks

tititopher68-dev, 2021-08-03:
@Jyers
Never had this issue in my own build
But I added this in configuration.h to stabilize communication between LCD Display and Motherboard
#define LCD_BAUDRATE 115200

tititopher68-dev, 2021-08-03:
@Mineplayerminer  @AkidDinie @Bezur-Ro 
Add just 
#define LCD_BAUDRATE 115200
In configuration.h
To fix the baudrate between LCD display and Motherboard

Mineplayerminer, 2021-08-03:
> @Mineplayerminer  @AkidDinie @Bezur-Ro 
> Add just 
> #define LCD_BAUDRATE 115200
> In configuration.h
> To fix the baudrate between LCD display and Motherboard

I just cleaned the LCD display connector pins and it solved the problem. No needs to change the baud rate for me.

NpFaith, 2021-08-03:
I'm printing now on the 3x3 manual mesh and I didn't have problems yet. I think 5x5 have the bug but I will print more and give you feedback. Thanks.

tititopher68-dev, 2021-08-03:
@NpFaith 
It's not a real firmware issue !!  Because there is no difference between a 3x3 and a 5x5 apart from the number of leveling points!
 It is a hardware problem of transmission speed between the screen and the Motherboard.
 The LCD Display is randomly unstable at 250,000 bauds, a new value set by Marlin ...
 Also go back to my previous post and add to configuration.h:
 #define LCD_BAUDRATE 115200 and check, clean and remove as much as possible the screen cable on the motherboard side.  Too much heat causes random transmission errors ... (and I'm not going to start a physics class to justify this known phenomenon ...)

NpFaith, 2021-08-03:
I was just reporting what is happen to me, I put the E3V2-ManualMesh-5x5-v4.2.2-v1.3.5.bin compiled firmware and I had this screen issue in the first and second print, then I decide to test with E3V2-ManualMesh-3x3-v4.2.2-v1.3.5.bin and I´m in the 3 print now and so far so good. But in case I have the problem I will make that change in the in configuration.h. Thanks for the help.

Jyers, 2021-08-03:
@tititopher68-dev, that value is already defined in the firmware in the dwin.cpp file:

`bool DWIN_Handshake(void) {`
`  #ifndef LCD_BAUDRATE`
`    #define LCD_BAUDRATE 115200`
`  #endif`
`  LCD_SERIAL.begin(LCD_BAUDRATE);`

Most likely this is just a connector issue as @Mineplayerminer reported. If the issue appears or anyone else finds that cleaning the connector helps, feel free to update us here.

Mineplayerminer, 2021-08-03:
My friend has also an Ender 3 V2 and he had a bad soldier on the LCD connector so he got a new screen for free as a replacement as the first LCD screen was faulty. So kind from Creality.

zippy-man, 2021-08-10:
I had zero issues with the screen and the first print after flashing 1.3.5 the screen became garbled. It happens from time to time and usually is shows white like the pictures above or the temp display has artifacts. 

I also noticed that the display will sometimes become unresponsive for a few seconds. It happens more when the UBL probing is active. It is almost easy to reproduce when the probe is dropping and retracting just spin the knob and you can see it does not respond by moving the highlight curser next to names. 

Maybe an LCD issue but I am leaning towards no. Multiple reports of display issues after flashing 1.3.5. what are the odds that many LCD's start having issues all at the same time?

esrpradeep, 2021-08-10:
After a long hour of printing time, I am also facing the garbled message on my LCD screen. The cable connections are not the problem, and it happens after a long time.
 
I'm using 3V2-ManualMesh-5x5-v4.2.7-v1.3.5.bin, and tried flashing the original version of DWIN_SET.

Is there a method to fix this issue?

[image]

Jyers, 2021-08-10:
I'm guessing this may have been an issue from upstream. I've had several reports now of the issue going away with the latest source. That is including a merge after the 1.3.5 release. So hopefully that will turn out to be the issue here

Jyers, 2021-08-10:
Please try https://github.com/Jyers/Marlin/releases/tag/v1.3.5.b and report back

pgastwirth, 2021-08-11:
I installed 1. 3.5.b and have a small tear near bed temperature. I was a few versions behind but finally installed my bltouch so I installed the latest that was available 3x3 mesh 
[image]

NpFaith, 2021-08-11:
pgastwirth what firmware file did you choose?

pgastwirth, 2021-08-11:
E3V2-BLTouch-3x3-v4.2.7-v1.3.5.b.bin

tititopher68-dev, 2021-08-12:
@pgastwirth 
Using octoprint ?

pgastwirth, 2021-08-12:
Yes

tititopher68-dev, 2021-08-13:
@pgastwirth 
Ok if you know how to complile the firmware, in configuration.h, modify : 
#define BAUDRATE 115200
by
#define BAUDRATE 250000
and
you can also uncomment #define BAUD_RATE_GCODE

With that, in a terminal, you can do : M575 P1 B250000 -> baudrate at 250000 or M575 P1 B115200 ->baudrate at 115200.

Don't forget to modify the speed connection in octoprint...

At home, I also added in configuration.h (just under #define BAUDRATE 250000):
#define LCD_BAUDRATE 115200
My connection is fixed at 250,000 baud and I never had this issue ...

tititopher68-dev, 2021-08-19:
@Jyers
 I have a feeling that this issue has the same cause as that of the M48 (https://github.com/Jyers/Marlin/issues/1321#issuecomment-901716866) ... and related to HAL STM32 ...  serial communication problem.

IrishBrewer, 2021-08-20:
I am having the same glitches at the top and bottom of my scree similar to Bezur-Ro's image after having recently upgraded to the 4.2.2 v1.3.5 firmware with a 5x5 manual leveling grid.  I have something printing now on two V2s with this firmware so I will try the v1.3.5b firmware when the printer is free.  This seems like too much of a trend to blame on dirty connections, especially since my printers are fairly new.  Another way to verify would be to go back to the V1.3.4 to see if it goes away.   I will add that my prints did continue to complete normally so it is just a sort of annoyance.  Also, if I go to the tune menu and then return to the print screen, the top of the screen rectifies itself (at least for a while) and the bottom remains glitched.

tititopher68-dev, 2021-08-20:
@zippy-man @pgastwirth @IrishBrewer @Jyers 
You can look at https://github.com/Jyers/Marlin/issues/1321#issuecomment-902971371

IrishBrewer, 2021-08-21:
My prints on both printers finished so I updated to 1.3.5.b on both printers, releveled and started new prints and I'm over an hour into them and so far so good - no glitching.

tititopher68-dev, 2021-08-21:
The merge has just been done, all that remains is to compile and verify that everything is functional.

closed 2021-08-22 by AkidDinie

Ali229, 2021-08-31:
I'm still seeing this issue, started happening after the updates on the 21st. Reinserted the cable, still seeing the issue. Restarting the printer takes away the problem temporarily.

IrishBrewer, 2021-08-31:
Yes, I edited my post above to indicate that it is still happening to me as well on the latest Jyers firmware (on both of my V2 printers).  And yes, rebooting sets things back to normal for a bit but it seems to glitch a couple hours into the print (if the print takes that long).  I was thinking about deprecating my firmware to 1.3.4 as I'm pretty sure it would go away.

PapsDizzy, 2021-09-04:
today at a print of 39 hours i get the glich after 16 hours. so i must wait of finish the print to see what happend.

Ali229, 2021-09-09:
So what is the solution for this I see both changing it to #define BAUDRATE 115200 and #define BAUDRATE 250000. Which one is working, anyone had success?