Jyers/Marlin #1265: Changing Filament Midprint

031113, 2021-08-02:
I have a problem changing the filament while printing. when pressing the command to replace the filament hotend moves to the upper left corner. i want the hotend to move to the bottom left corner (home) like ender 3 pro. then how to do? can you modify it?

Cal-LF, 2021-08-03:
You can change these setting by compiling your own firmware. There are not menu settings for making these changes. They are set when compiled.

031113, 2021-08-03:
> You can change these setting by compiling your own firmware. There are not menu settings for making these changes. They are set when compiled.

Can you help me? I don't know how to recompile it?

Cal-LF, 2021-08-03:
@031113 I think I have connected with you on FB. Follow through with installing VS Code and I will guide you further.

closed 2021-08-03 by Jyers