Jyers/Marlin #1266: Linear advance

Soltysrysik, 2021-08-03:
Why is Linear Advance not working for me?

Is this option disabled in this firmaware?

I used firmaware E3V2-ManualMesh-3x3-v4.2.2-v1.3.5.bin
Cura and plugin Linear Advance

I see changelog in firmaware Jyers v1.3.3 released this on 23 May

"Added Linear Advance Support to Source"

Cal-LF, 2021-08-03:
@Soltysrysik Linear advance is being tried on Jyres personal configuration only.

Jyers, 2021-08-03:
@Soltysrysik "Added Linear Advance Support to Source" Means that I added the ability to enable linear advance in the config files. This means you still need to compile your own firmware to enable linear advance. It also requires a hardware mod to the mainboard in order to work on creality boards. See [here](https://www.reddit.com/r/ender3v2/comments/n3clew/ender_3v2_422427_board_tmc_uart_mods/) for more info on the mod

closed 2021-08-03 by Jyers