Jyers/Marlin #1267: ender3 v2 with jyers marlin 1.3.5 on a 4.2.7 board ignoring Z offset value set when printing

ACaust, 2021-08-04:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ACaust, 2021-08-04:
[](url)https://github.com/Jyers/Marlin/releases/download/v1.3.5/E3V2-BLTouch-3x3-v4.2.7-v1.3.5.bin

tititopher68-dev, 2021-08-05:
@ACaust 
I'm sorry but the Z-Offset is well applied.  (I hope you have saved your Z-Offset with Save in the Z-Offset menu). The old versions v1.3.3 or 1.3.4 use the same code on the Z-Offset application when launching the Print and it's OK.
 Either there was an error during the pre-compilation of the firmware, but I very much doubt it .... Or your startGcode is not correct ....
 However, a golden rule, after each firmware flash, you must do a Reset to Default then Store Settings and finally a Restore Settings ...

ACaust, 2021-08-05:
I will try that I got the start g code set up right I have 2 of the same machine running the same firmware and equipment. The other works just fine.
But I did not do the reset to default  then Store Settings and finally a Restore Settings. 
could it be possible A bad BL touch?

ACaust, 2021-08-07:
problem solved turns out the lead screw for my Z axis was just ever so slightly loose so it would spin without moving in the correct steps thank you for your help anyways @tititopher68-dev works good now other then my bed needs replaced it doesnt adhere very well with the bed that got all scratched up

closed 2021-08-09 by Jyers