Jyers/Marlin #1279: Source problems on 4.2.7 and ABL. Axels not moving

oricardomiranda, 2021-08-07:
Hello

I have an Ender 3 V2 with a 4.2.7 board.

I have an ABL só I've been trying to use the open source folder but without any luck.

The axels just won't move.

I've been successful with other firmwares before.

I think it might be my board, because in the other firmwares I get to choose my board. Is this source I can't seem to find it. Can someone help me?

I'm attaching a zip file with my configs

matthewdolman, 2021-08-09:
You didn't include the config files in the zip.

in your Configuration.h have you set this:

`#define MOTHERBOARD BOARD_CREALITY_V427`

I had the same problem and realised the board was wrong.

oricardomiranda, 2021-08-09:
Thank you so much. You are a life saver. 

Had some difficulties with the upload.

closed 2021-08-09 by Jyers