Jyers/Marlin #1282: Can't compile firmware anymore

Le0xFF, 2021-08-08:
### Description

I can't compile the firmware anymore.
I'm still able to compile 1.3.4 version, but with the latest commits, I'm not able to compile it anymore.
When I try to run `python buildroot/share/vscode/auto_build.py build`, the error `Invalid board` comes up.

### Steps to Reproduce

1. `git clone https://github.com/Jyers/Marlin.git && cd Marlin`
2. `python buildroot/share/vscode/auto_build.py build`

#### Additional Information

[image]

Jyers, 2021-08-09:
I was unable to recreate this behavior. Perhaps try deleting the `.pio` folder and trying again

Le0xFF, 2021-08-10:
The `.pio` folder is not created until one compilation goes well, so it was never created for me.
By the way I solved the issue compiling it directly from VSCode. No error at all from there!

closed 2021-08-10 by Le0xFF