Jyers/Marlin #1294: Screen firmware

dnettoRaw, 2021-08-09:
Hello
I have the screen DGUS2 V3.5 ( https://www.creality3dofficial.com/collections/accessories/products/upgraded-touch-panel-screen-display-for-ender-3-pro-ender-3-v2?sca_ref=401182.rYODNiwm8M )
 and i can't use your firmware i realized i can only flash firmware that has T5L
 i loved the gotcha layout

tititopher68-dev, 2021-08-09:
@dnettoRaw 
It's completely normal. This firmware is build specifically for the combination of a Creality 4.2.2/4.2.7 board with the stock display that comes with the Ender 3v2.

Jyers, 2021-08-09:
@tititopher68-dev is correct. The creality touch screens are very fundamentally different than the stock V2 displays. While I may make an implementation for them in the future, there is no such support now.

closed 2021-08-09 by Jyers