Jyers/Marlin #1295: Hotend live temperature garbled display

sidsingh78, 2021-08-09:
### Description

NpFaith, 2021-08-10:
https://github.com/Jyers/Marlin/issues/1263

tititopher68-dev, 2021-08-10:
@sidsingh78 
Exactly, the same issue #1263...Check connector and if necessary, reflash the display firmware

esrpradeep, 2021-08-10:
<reposting from the thread "glitch on the screen #1263", since this one is 'open'>

After a long hour of printing time, I am also facing the garbled message on my LCD screen. The cable connections are not the problem, and it happens after a long time.

I'm using 3V2-ManualMesh-5x5-v4.2.7-v1.3.5.bin, and tried flashing the original version of DWIN_SET.

Is there a method to fix this issue?

[image]

Jyers, 2021-08-10:
Please try https://github.com/Jyers/Marlin/releases/tag/v1.3.5.b and report back to #1263

closed 2021-08-10 by Jyers