Jyers/Marlin #1299: Any ABL work until 5th point

peterADM, 2021-08-11:
### Description

Current firmware: E3V2-BLTouch-5x5-HS-v4.2.7-v1.3.5.b.bin
I try to level with the leveling tool and when the bl touch finishes the 5th point. The leveling "completes" even thought it needs 20 more points. the mesh only displays the front 5 points leveled. Ive re flashed the firmware same issue. I have printed before the issue happened. 

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]
level all 25 points and give me accurate mesh

**Actual behavior:** [What actually happens]
levels first 5 then promts me with a "safe to EPRON" thus only giving me the front 5 points accurately leved
#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

bharatpareek, 2021-08-11:
Some users has issue with the Bltouch port. the trigger signal or ground is not attached properly. Connect the two pin to the z switch stop & flash the firmware below.

Check the 5th point in FAQ section here: https://smith3d.com/ender-3-v2-bltouch-firmware-installation-guide-by-smith3d-com/

OberfragGER, 2021-08-11:
Oh. Thtas pretty odd. I have the same issue but with the E3V2-UBL-BLTouch-10x10-HS-v4.2.2-v1.3.5.b.bin. I stops on the 65th probe. I wasnt sure if i had that on 1.3.4 too, so i switched back to E3V2-UBL-BLTouch-10x10-HS-v4.2.2-v1.3.4.bin.

Same result there :(

That pretty odd to flash another firmware...in my option.

tititopher68-dev, 2021-08-11:
@OberfragGER
 Sorry but in your case there is not an issue.  This has been said, written many times ....
 In UBL mode, due to the position of the probe relative to the nozzle, not all the points of the Bed are reachable and therefore measurable.  This is also why, once the mesh is created, you get a message saying: All reachable points are probed!
 The other points are then interfered with.  The advantage of the UBL is to allow, for these interfering points, to make the manual adjustment according to the zone to be printed and not to have to systematically redo the Bed Leveling using Autotilt.

 In addition, I invite you to take a little tour on the Internet to learn what UBL mode is or take a quick tour of the Wiki ...

 @peterADM
 This is most certainly a hardware problem with connection of the BLTouch or a malfunction of the BLTouch (internal position of the palapation PIN, it is adjustable, see demagnetization of the PIN...).

OberfragGER, 2021-08-11:
> @OberfragGER
> Sorry but in your case there is not an issue. This has been said, written many times ....
> In UBL mode, due to the position of the probe relative to the nozzle, not all the points of the [...]

Thanks a lot for you explanation Doesnt realize „the problem“ before so it sounded familiar.

I will do so and look deeper into the wiki.

peterADM, 2021-08-15:
Hey! sorry for being gone for 4 days i have been busy. my bl touch is a 3.1, so there is only the 5 pin connector. I have flashed the 5x5hs and the 5x5, same thing with the 3x3 and same issue. it stops after it needs to go to the next row of points. (ie. the 5th probe point on the 5x5 and the 3rd probe point for the 3x3). and i tried the firmware you sent form the smith 3d website.

My main board is a creality 4.2.7 btw.
bl touch 3.1 with a 5 pin connector.

tititopher68-dev, 2021-08-15:
@peterADM 
After every flash, please do a Reset to Default...
Hardware issue, certainly BLTouch or the cable of the BLTouch, maybe cut when the head comes completely to the right of the bed ...

jockorange, 2021-08-18:
@tititopher68-dev 
I have the same issue.
I had an ender 3v2 with the auto leveller with firmware  E3V2-BLTouch-3x3-v4.2.2-v1.3.4
Works great
Upgraded my machine with the ender extender kit.
Someone had precompiled firmware and i installed it and it did 4 points and then gave up and asked me to save to eprom
If i go back to the old firmware i was using the bed levelling works

tititopher68-dev, 2021-08-18:
@jockorange 
I am just looking for list of Marlin's issues. And effectively, these are issue with homing and levelling since the last bugfix (!?)
@Jyers 
I think it would be desirable to go back (build 1.3.4) keeping the latest commits.  Too much instability on basic functions (non-constant homing, incorrect UBL autotilt, Bed Leveling: indeed, having flashed your last build, I also have in 5x5 BLT at the 5th position a leveling stop !!)

jockorange, 2021-08-19:
@tititopher68-dev 
After messing about for a while I managed to get it working 
I re levelled the bed manually then the auto levelling worked, I ran auto levelling 4 times and it worked

tititopher68-dev, 2021-08-20:
@OberfragGER @bharatpareek @jockorange @peterADM 
You can look at : https://github.com/Jyers/Marlin/issues/1321#issuecomment-902971371

tititopher68-dev, 2021-08-21:
The merge has just been done, all that remains is to compile and verify that everything is functional.

jlmurray4279, 2021-09-29:
I figured it out. So I installed some new silicone springs and then only tightened the bed a little bit. This cause one side of the bed to be out of level with the gantry so much that the bltouch (CL touch) was not sensing the right side. Then when I tightened the screws and leveled the bed the probe sensed the right side and continued to do the leveling. So for the beginners trying to figure it out I would try leveling the bed and make sure to tighten the screws a deceit amount. If that doesn't work then try to level the X gantry. I hoped this helps