Jyers/Marlin #1301: Heaters turned off before last G-code command completes

camz, 2021-08-12:
### Description

I noticed this while printing 200mm long V-slot covers (https://www.thingiverse.com/thing:3890851) using v1.3.5.
The last part of the print is a straight line for the full 200mm length of the V-slot cover.  The display showed the print complete screen (with the continue button) as soon as the last line of the print started.  The display showed both the nozzle and bed temperatures starting to drop while the last 200mm line was still printing.

### Steps to Reproduce

1. Download files for V-slot covers from thingiverse (https://www.thingiverse.com/thing:3890851)
2. Modify the OpenSCAD file changing the length to 200mm (line 2 of OpenSCAD file)
3. Compile (F6) and export as STL
4. Import into Cura, slice for 0.2 layer height, use the default orientation in Cura for the V-slot cover
5. Print using the gcode file from Cura
6. Observe the display on the last layer (should consist of two 200mm lines on on each side)

The last G-code command of a 200mm line only takes 4 seconds to print @ 50mm/sec, but that is long enough for the nozzle to start cooling (especially with the fan at 100%).   Luckily it didn't cool enough to prevent the filament from extruding.
This probably isn't noticeable if the last G-code command involves a shorter distance than the 200mm for the V-slot cover.

Printer was using the pre-compiled `E3V2-Default-v4.2.2-v1.3.5.bin` from the Releases page.

**Expected behavior:** [What you expect to happen]
Nozzle and bed heaters should remain on until *after* the last G-code command completes.
Display should not show "print complete" message until *after* the last G-code command completes.

**Actual behavior:** [What actually happens]
Nozzle and bed heaters appeared to turn off after *starting* the last G-code command, but before the command completed.
Display showed "print complete" message (and button) after *starting* the last G-code command.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

rmoorecpcc, 2021-08-16:
The display will show complete when the last gcode command is buffered I believe. Can you confirm via serial log that the heaters did in fact turn off (heater will show a target temp of 0 of course) with a subsequent command issued or anything? I'm not a dev here, just pointing out the display will show complete when the mainboard buffer gets the last command, which can be many seconds ahead of actual finish if the last few moves are long ones.

camz, 2021-08-16:
Not sure how to capture a serial log, definitely willing to try (with some instructions).
As I mentioned in the issue/bug report, the last layers of the V-slot covers are long straight lines, in my case I was printing 200mm long covers at 50mm/sec, so I had 4 seconds to observe that the nozzle temperature was, indeed dropping while the last part of the print was still happening.

I understand that you are saying the display shows "complete" after the last command is buffered (but not necessarily executed).
That *IS* what I observed and reported, that too is technically a bug, and since I also saw the nozzle temperature dropping during the last line of the print, it appears that other actions (nozzle and bed heaters) are also being turned off at the same time as that completed message.  As a result the two behaviors (premature display AND premature heaters turning off) are, indeed related.

gothcha, 2021-08-21:
@camz I too can confirm this behaviour, but is not a bug in the firmware. It is as discussed - Marlin processes additional commands whilst the previous moves are still in progress!

However I have found that removing the lines
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed
from the end code in Cura stops the hot end from turning off until the print is complete and the Continue pop up appears.  Note Cura automatically appears to add M140 S0 (bed temp zero) prior to the end code which still turns off the bed prematurely, sadly this cannot be stopped but could be removed with a post process script. I guess the bed turning off a few seconds early is less of an issue.

Note the current firmware turns off everything when the print has finished when the print is from the SD card.

camz, 2021-08-25:
Got it.  This isn't an issue of bug in the firmware at all, but rather an issue with the end g-code used in Cura.
With that knowledge in hand, I had a look at the configuration in Cura and found those in the End G-code settings.
I suspect then, that modifying the end G-code to include an M400 command before these would fix the issue (assuming that command is supported).

camz, 2021-08-25:
Not a bug, behavior is due to end G-code defaults in Cura slicer.

closed 2021-08-25 by camz