Jyers/Marlin #1307: Reloading bug

supergoodgamer420, 2021-08-13:
### Description

UI Seems to be reloading forever everytime i turn on the printer, the temperature sensors seems to stay at 0 degrees Celcius and i cannot use the control knob 
This has been happening to me ever since i have updated to 1.3.5/1.3.5b
If i reflash the firmware, it works again for a short time until it suddenly starts reloading again
### Steps to Reproduce

1. The only thing I did was create a mesh bed level, store the settings then restart the printer by turning it off and on

**Expected behavior:** Printer to work 

**Actual behavior:** Non-stop reloading with no ability to fix it 

#### Additional Information

Video attached
https://user-images.githubusercontent.com/88906538/129411150-b14b72ca-8861-4e53-a7dd-04eb8ce4c955.mp4

Cal-LF, 2021-08-13:
An issue like that is likely a hardware problem. Check PSU voltage and connections. Could be a recently added or changed part.
What is your hardware setup? BLTouch?

supergoodgamer420, 2021-08-13:
> @supergoodgamer420 An issue like that is likely a hardware problem. Check PSU voltage and connections. Could be a recently added or changed part.
> What is your hardware setup? BLTouch?

Nope, I downgraded to Jyer's 1.3.4 and the issue does not persist anymore. My Ender 3 V2 is stock is completely stock in terms of electrical components - no BL-Touch, No changes to the mainboard/PSU

Cal-LF, 2021-08-13:
@supergoodgamer420 This has not been reported before and there are many users. Have you reset the eprom after changing firmware? V1.3.5 is a major change and clearing the memory may fix your issue.

supergoodgamer420, 2021-08-13:
@Cal-LF How do I do that?

Cal-LF, 2021-08-13:
@supergoodgamer420 In the Control menu, choose "reset to defaults"  then "restore settings" then "store settings". With any changes you make to your settings, select "store settings" each time.

supergoodgamer420, 2021-08-13:
@Cal-LF Ok thank you, I will attempt that once my 4 hour print finishes

supergoodgamer420, 2021-08-17:
@Cal-LF It worked for a while but the bug happened again, although this time it only reloaded twice and then started working properly again.

Cal-LF, 2021-08-17:
Let's see if @jyers or other users have an idea what this could be. If it is not happening with the previous version, there may be an issue with the latest release. Although other users would most certainly have reported this.

Hologos, 2021-08-19:
> Let's see if @jyers or other users have an idea what this could be. If it is not happening with the previous version, there may be an issue with the latest release. Although other users would most certainly have reported this.
> 
> I have gotten a reboot a couple times when putting SD card and moving to the Print menu. This could be something to do with the SD card reading in the firmware. This has happened on previous releases also.

This is happening to be as well. Sometimes when I access print menu, it crashes. I also noticed that after power up, sometimes the BL Touch doesn't push down the probe and is blinking red. I have to restart the printer several times.

supergoodgamer420, 2021-08-20:
It looks like removing the SD Card while it's running will temporarily fix the crash, so the bug has something to do with how the printer reads the card.