Jyers/Marlin #1312: Setting E0_DRIVER_TYPE to TMC2208 for Linear Advances breaks firmware

nullstalgia, 2021-08-15:
Howdy, it's been a while.

Came in to grab whatever came upstream, and ran into an issue whenever I try to use my config that I melded with the newest templates.

It would compile, but the screen would stay black when flashing, even on another SD card.

My configs:

https://gist.github.com/nullstalgia/6fa19037827acb4137735c8bd8f510fa

tititopher68-dev, 2021-08-15:
@nullstalgia 
In your attached configuration.h file, you declared for E0 a TMC2208_Standalone instead of TMC2208.

nullstalgia, 2021-08-15:
@tititopher68-dev 

True, but that is the only difference that allows the firmware to be loaded.

Jyers, 2021-08-15:
@nullstalgia My config is running fine and I also use TMC2208, so I'd either point to some other weird SD issue, or try comparing your config to mine

nullstalgia, 2021-08-16:
On a fresh clone of the JyersUI branch, I got this with Jyer's Config in the folder

[image]

Then, after a compile, it changed to this

[image]

Changing it to Standalone clears all workspace errors

[image]

Jyers, 2021-08-16:
Ahhhh, that's just a VSC issue. See #932. That will fix it

nullstalgia, 2021-08-16:
@Jyers I've been able to narrow it down to something weird.

If I set my pin to say, PA13, It flashes properly, I get a TMC CONNECTION ERROR on the screen, and M122 errors out with ALL LOW.

Rolling back to my older firmware, running M122 returns me this at the bottom:

[image]

This is the guide I followed for my mod:

https://drive.google.com/file/d/15cUf10lMxW4NUHE9qVAlamXmULdrOJxz/view

And even the [mod author you linked](https://github.com/Jyers/Marlin/issues/1266) used the same method (Red wire across the MCU)

[image]

I get no serial output from the printer, and the screen is black the whole time.

TOTA42, 2021-09-24:
@nullstalgia 
The Red wire that your image goes is actually PA3, not PA13, which is correctly stated in https://drive.google.com/file/d/15cUf10lMxW4NUHE9qVAlamXmULdrOJxz/view on pages 2 and 5.  Another issue is that TMC_UART_PIN_E isn't used in the code I believe. I think it should be E0_SERIAL_TX_PIN and E0_SERIAL_RX_PIN.
If you change your 
#define TMC_UART_PIN_E PA13
to 
  #define E0_SERIAL_TX_PIN                  PA3
  #define E0_SERIAL_RX_PIN                  PA3

 I think you'll be fine.

doubleO00, 2021-11-03:
Any thoughts on why after setting E0 from TMC2208_Standalone to TMC2208 after the solder wire mod my LCD wont show, I get a blank screen, but if I change back to TMC2208_Standalone the LCD screen lights up normal.

Trying to get Lin_Advance to work on E0,  firmware compiles with both TMC2208_Standalone and  TMC2208 with no errors

When I flash the bin file that I build with TMC2208 instead of TMC2208_Standalone I lose LCD

tiagofreire-pt, 2021-11-06:
Are you sure that you have TMC2208, instead of TMC2209?

Your board seems to be an "B" one:

https://github.com/Jyers/Marlin/discussions/707#discussioncomment-701286

BradleyHight, 2022-01-17:
@Jyers I am working with your config on Marlin 2.0.1 and similar to those above I receive a black/blank screen, and no serial response upon trying to flash with "E0_DRIVER_TYPE" as "TMC 2208". Upon comparing with the raw file provided in "Configuration Files" my only changes are the slice thermistor ("#define TEMP_SENSOR_0 67"), and pulling TMC UART in on the STM32 pin PA3 ("#define TMC_UART_PIN_E PA3") per [Wong Sy Ming's guide](https://drive.google.com/file/d/15cUf10lMxW4NUHE9qVAlamXmULdrOJxz/view). I also see this configuration confirmed by [u/LookAtTheShinyShiny](https://www.reddit.com/user/LookAtDaShinyShiny/) on reddit; however upon talking with him, he is running v1.3.4 with stability which I will be trying later. Yet on this branch, simply changing back to "E0_DRIVER_TYPE" as "TMC 2208_STANDALONE" allows for complete flash and stepper control while not supporting the TMC UART interface. Achieving dimensional accuracy with the high flow rate Mosquito Magnum is critical for my use case and I would greatly appreciate your help on further troubleshooting or bug fixes to achieve linear advance on my E3 V2

BradleyHight, 2022-01-18:
Can confirm successful flash and UART response with M122 via serial with your config on v1.3.4 as recommended by u/LookAtTheShinyShiny. This issue appears to be isolated to v2.0.x. Please advise upon bugfix or if you require further testing on my end.