Jyers/Marlin #1321: M48 Issue

eblieb, 2021-08-18:
Running v1.3.5.B 
3V2-UBL-BLTouch-10x10-HS-v4.2.2-v1.3.5.b.bin and can  not run an M48. 

Symptoms are 
Run G28 - Homes fine
Run M48 

2021-08-18 14:20:22,888 - Send: N3 M48*28
2021-08-18 14:20:22,902 - Recv: M48 Z-Probe Repeatability Test
2021-08-18 14:20:23,773 - Recv:  T:25.77 /0.00 B:25.51 /0.00 @:0 B@:0
2021-08-18 14:20:24,895 - Recv: echo:busy: processing
2021-08-18 14:20:25,773 - Recv:  T:26.24 /0.00 B:25.36 /0.00 @:0 B@:0
2021-08-18 14:20:26,894 - Recv: echo:busy: processing
2021-08-18 14:20:27,773 - Recv:  T:26.14 /0.00 B:25.55 /0.00 @:0 B@:0
2021-08-18 14:20:28,894 - Recv: echo:busy: processing
2021-08-18 14:20:29,773 - Recv:  T:26.16 /0.00 B:25.51 /0.00 @:0 B@:0
2021-08-18 14:20:30,894 - Recv: echo:busy: processing
2021-08-18 14:20:33,901 - Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regul$
2021-08-18 14:20:35,852 - Recv: start

Then nothing.

tititopher68-dev, 2021-08-18:
@eblieb 
According to your comment, it's  certainly just a hardware issue

eblieb, 2021-08-18:
Nope reverted back to E3V2-UBL-BLTouch-10x10-HS-v4.2.2-v1.3.5.bin and I was able to run M48 without an issue. So it is something in the Marlin source that 1.3.5.b was updated to. I tried to submit the bug report in the Marlin git but they said they wont support a bug if it is in a fork, eventho it is in their source.

tititopher68-dev, 2021-08-18:
@eblieb 
I investigated.  No change in sources between versions 1.3.5 and 1.3.5.b for M48.  Have you tried from the interface?
 Your Terminal return indicates a communication error (Time out with Printer communication).

Tuncay-Ayhan, 2021-08-18:
I can confirm this problem also. I have communicated with @Jyers about this
via Discord. It’s an issue I also have since 1.3.5 . 1.3.4 doesn’t have
this issue.

It’s very random and I’ve tried different approaches. From GCode on SD to
Octoprint etc.

Nothing seems to make sense. When reverting back to 1.3.4 the issues is non
existent.

Tuncay-Ayhan, 2021-08-18:
> @eblieb
> I investigated. No change in sources between versions 1.3.5 and 1.3.5.b for M48. Have you tried from the interface?
> Your Terminal return indicates a communication error (Time out with Printer communication).

I can confirm this problem also. I have communicated with @Jyers about this
via Discord. It’s an issue I also have since 1.3.5 . 1.3.4 doesn’t have
this issue.

It’s very random and I’ve tried different approaches. From GCode on SD to
Octoprint etc.

Nothing seems to make sense. When reverting back to 1.3.4 the issues is non
existent.

<p>

<p>

### When using the manual leveling with probe function on the printer itself
<details>
  <summary>Click to expand!</summary>
  
  ```
  Recv: Error:Probing Failed
WARNING! Received an error from the printer's firmware, ignoring that as configured but you might want to investigate what happened here! Error: Probing Failed
Recv: //action:notification Probing Failed
Recv: Error:Probing Failed
WARNING! Received an error from the printer's firmware, ignoring that as configured but you might want to investigate what happened here! Error: Probing Failed
Recv: //action:notification Probing Failed
Recv: Error:Probing Failed
WARNING! Received an error from the printer's firmware, ignoring that as configured but you might want to investigate what happened here! Error: Probing Failed
Recv: //action:notification Probing Failed
Recv: Error:Probing Failed
WARNING! Received an error from the printer's firmware, ignoring that as configured but you might want to investigate what happened here! Error: Probing Failed
Recv: //action:notification Probing Failed
Recv: Error:Probing Failed
WARNING! Received an error from the printer's firmware, ignoring that as configured but you might want to investigate what happened here! Error: Probing Failed
  ```
</details>

<p>

<p>

### Video 1.3.5 M48 - Dying screen and no M48 execution
[Video](https://user-images.githubusercontent.com/28215053/129969610-cc8fcfd9-9423-4844-a61d-193474652228.mov)

tititopher68-dev, 2021-08-18:
@Tuncay-Ayhan 
I confirm that there are indeed a lot of problems with the marlin sources for 3 weeks (homing, bed leveling, autotilit, M32, M48 now ...)

Tuncay-Ayhan, 2021-08-18:
> @Tuncay-Ayhan
> I confirm that there are indeed a lot of problems with the marlin sources for 3 weeks (homing, bed leveling, autotilit, M32, M48 now ...)

I couldn't find anything obvious myself in upstream, so we'll have to wait and be patient till the moment someone finds the cause of it. Nothing to do about it as of now.  :-D

tititopher68-dev, 2021-08-19:
@Tuncay-Ayhan @Jyers @eblieb 
It seems to come from the HAL STM32 libraries ... there are 2 patches released on August 10th ...
 So either @Jyers merges again with the latest bugfixes, or we integrate the modifications for 35 files:
 Define HAL_STM32 for HAL / STM32 (MarlinFirmware # 22537)
 Update HAL / STM32 wrappers
 Followup to MarlinFirmware # 22537

Tuncay-Ayhan, 2021-08-19:
> @Tuncay-Ayhan @Jyers @eblieb
>
> It seems to come from the HAL STM32 libraries ... there are 2 patches released on August 10th ...
>
>  So either @Jyers merges again with the latest bugfixes, or we [...]

It's good to hear you've found the possible fixes. Now we'll have to wait for Jyers.

tititopher68-dev, 2021-08-19:
@Tuncay-Ayhan 
Tonight, I will compile a version with these fixes and we'll see ...

eblieb, 2021-08-20:
Just reverted back to 1.3.4 and all of my BLTouch issues are gone. I was constantly having issues on 1.3.5 and 1.3.5.b

tititopher68-dev, 2021-08-20:
@Tuncay-Ayhan @eblieb @Jyers 

OK it's a HAL STM32 issue. 
I integrated all the modifications for STM32 and No screen glitch, M48 OK, ABL G29 OK

I worked on a BTT skr E3 mini V2 (compilation issue) with this firmware 1.3.5b and  no issues.

eblieb, 2021-08-20:
Nice... hopefully it gets merged into Jyers firmware soon. I just ordered a CRTouch on Amazon to replace my BLTouch... The cable on the BLTouch is going bad because it doesnt use the proper connector

tititopher68-dev, 2021-08-20:
@Jyers 
I think it would be necessary to merge the latest Marlin bugfixes again to fix all this issues...

Jyers, 2021-08-21:
@tititopher68-dev Done 😄 

Sorry I've been a bit absent lately, I've had some quite tasking responsibilities this week along with a rapidly approaching move

tititopher68-dev, 2021-08-21:
@Jyers
 No problem !  Sometimes we all have other obligations ...
 Have you noticed that by modifying 4 or 5 files in the sources, it makes the firmware also compatible with BTT Skr 1.4 and E3 mini motherboards?
 In any case, good move ahead ...

 @ Tuncay-Ayhan @eblieb
 The merge has just been done, all that remains is to compile and verify that everything is functional.

Tuncay-Ayhan, 2021-08-21:
> @Jyers
>
> No problem !  Sometimes we all have other obligations ...
>
> Have you noticed that by modifying 4 or 5 files in the sources, it makes the firmware also compatible with BTT Skr 1.4 and E3 mini [...]

Thanks for the headsup!

eblieb, 2021-08-24:
Any update on when the new firmware is going to be released with the bugfixes?

eblieb, 2021-08-29:
@Jyers @tititopher68-dev any update on when a new compile may be released?

Jyers, 2021-08-30:
@eblieb Sorry for the delay. I will hopefully have a new release soon. I have another feature or two to add before I do though. I've just finished moving so I should have a bit more time now to get things sorted.