Jyers/Marlin #1322: Ender 3 V2 10x10 UBL with BL Touch

Grenalis, 2021-08-18:
When leveling the bed it only does the middle square area, around 53 out of 100 or so if I remember.  Then it says complete.  Have to use the non UBL 5x5 to get entire bed.

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tititopher68-dev, 2021-08-18:
@Grenalis 
And still the same question ... Please do a little research in the issues (open and closed) band discussions and you will find your answer to this normal behavior ....

Grenalis, 2021-08-19:
It's always refreshing to read a reply to a issue with something with aggressive disregard.  Especially when it is pointed at someone who is new to something.  Thank You titiopher68-dev.  You helped soooo much with your reply.  I would have thought asking a question on the site where the firmware was at would be of some help.  My Bad.

tititopher68-dev, 2021-08-19:
@Grenalis 
No one has said that 3D printing is easy and yes, 3D printing requires research (sometimes, even very often), as we do to find and fix bugs.  But to come back and close this file, the subject has been discussed many times as here for example (https://github.com/Jyers/Marlin/issues/1299#issuecomment-897009352) or in the Wiki ...  Take the answer as you wish.  If you feel that you politely ask to do a research and find out the info here is insulting, not aggressive.  I am afraid that you are facing many disappointments.  Did you just try to find the information here or elsewhere and also make it clear that you were a complete beginner (we've all been there and that's not the problem).  With that, I wish you a good day and that you can progress and sincerely succeed in your 3D printing projects.