Jyers/Marlin #1323: SKR Mini E3 V2.0, Ender 3v2 LCD won't compile.

Freethefire, 2021-08-19:
### Description

Hi, 
I'm trying to compile your firmware for a SKR Mini E3 V2.0 and the stock Ender 3 v2 LCD. I've got it working with the latest Marlin bugfix, but I keep getting this error when trying to compile yours:

`Compiling .pio\build\STM32F103RC_btt\src\src\lcd\e3v2\creality\creality_marlinui.cpp.o
Compiling .pio\build\STM32F103RC_btt\src\src\lcd\e3v2\creality\dwin.cpp.o
Compiling .pio\build\STM32F103RC_btt\src\src\lcd\e3v2\creality\rotary_encoder.cpp.o
Marlin\src\lcd\e3v2\creality\creality_dwin.cpp: In member function 'void CrealityDWINClass::Draw_Print_ProgressRemain()':
Marlin\src\lcd\e3v2\creality\creality_dwin.cpp:761:31: error: 'class MarlinUI' has no member named 'get_remaining_time'
  761 |   uint16_t remainingtime = ui.get_remaining_time();
      |                               ^~~~~~~~~~~~~~~~~~
Marlin\src\lcd\e3v2\creality\creality_dwin.cpp: In member function 'void CrealityDWINClass::Start_Print(bool)':
Marlin\src\lcd\e3v2\creality\creality_dwin.cpp:5644:8: error: 'class MarlinUI' has no member named 'set_progress'; did you mean '_get_progress'?
 5644 |     ui.set_progress(0);
      |        ^~~~~~~~~~~~
      |        _get_progress
Marlin\src\lcd\e3v2\creality\creality_dwin.cpp:5645:8: error: 'class MarlinUI' has no member named 'set_remaining_time'
 5645 |     ui.set_remaining_time(0);
      |        ^~~~~~~~~~~~~~~~~~
Marlin\src\lcd\e3v2\creality\creality_dwin.cpp: In member function 'void CrealityDWINClass::Stop_Print()':
 5655 |   ui.set_progress(100 * (PROGRESS_SCALE));
      |      ^~~~~~~~~~~~
      |      _get_progress
Marlin\src\lcd\e3v2\creality\creality_dwin.cpp:5656:6: error: 'class MarlinUI' has no member named 'set_remaining_time'
 5656 |   ui.set_remaining_time(0);
      |      ^~~~~~~~~~~~~~~~~~
*** [.pio\build\STM32F103RC_btt\src\src\lcd\e3v2\creality\creality_dwin.cpp.o] Error 1
=========================================================================================================== [FAILED] Took 324.37 seconds ===========================================================================================================
Environment      Status    Duration
---------------  --------  ------------
STM32F103RC_btt  FAILED    00:05:24.369
====================================================================================================== 1 failed, 0 succeeded in 00:05:24.369 ======================================================================================================
[Configuration.zip](https://github.com/Jyers/Marlin/files/7015015/Configuration.zip)
`

### Steps to Reproduce

It's kinda complicated to explain all the steps, since it involves building a new cable and so on. 
But I've used information from here to get the lcd working:
https://github.com/MarlinFirmware/Configurations/issues/535
and here:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/424

Any suggestions would be very appreciated!

Best regards Alex

tititopher68-dev, 2021-08-19:
@Freethefire
 The firmware is not intended for this motherboard but only for Creality motherboards v4.2.2 and v4.2.7 ...

Freethefire, 2021-08-19:
> @Freethefire
> The firmware is not intended for this motherboard but only for Creality motherboards v4.2.2 and v4.2.7 ...

I'm aware of that, but there is people that get it working on my main board. There are even threads about it on this GitHub.

tititopher68-dev, 2021-08-19:
@Freethefire
 I anticipated your answer by completing my previous post ...
 For the E3 mini, this was true thank with LCD modifications and with the old firmware based on ExtUI ... which is no longer the case at all now.  You might as well move your request to the Discussion section where those who have tried it may be able to help you.

Freethefire, 2021-08-19:
> @Freethefire
> I anticipated your answer by completing my previous post ...
> For the E3 mini, this was true with LCD modifications and with the old firmware based on ExtUI ... which is no longer the case at all now. You might as well move your request to the Discussion section where those who have tried it may be able to help you.

Ah ok, I didn't realise that.
Yes your right about moving to discussion. I'm not sure how to do that though. I think there's been a button before but I can't find it.

tome9111991, 2021-08-20:
You have enable

```
#if EITHER(SDSUPPORT, LCD_SET_PROGRESS_MANUALLY) && ANY(HAS_MARLINUI_U8GLIB, HAS_MARLINUI_HD44780, IS_TFTGLCD_PANEL, EXTENSIBLE_UI, DWIN_CREALITY_LCD)
  #define SHOW_REMAINING_TIME       // Display estimated time to completion
  #if ENABLED(SHOW_REMAINING_TIME)
    #define USE_M73_REMAINING_TIME  // Use remaining time from M73 command instead of estimation
    //#define ROTATE_PROGRESS_DISPLAY // Display (P)rogress, (E)lapsed, and (R)emaining time
  #endif
```
In config_adv?

Freethefire, 2021-08-20:
@tome9111991 

Oh, no I haven't. Thanks you so much for your reply!

tome9111991, 2021-08-20:
Try thid config

https://github.com/Jyers/Marlin/blob/Ender_3_V2_JyersUI/Configuration%20Files/E3V2%20Templates/Default-NoProbe/Configuration_adv.h

tititopher68-dev, 2021-08-20:
@tome9111991 @Freethefire 
These functions are not defined in Marlinui.h because LCD_SET_PROGRESS_MANUALLY is not defined in configuration_adv.h...
You must have too :
#define LCD_SET_PROGRESS_MANUALLY
It's the conditional to define ui_set_progress() and ui_set_remaining()...

Freethefire, 2021-08-20:
@tome9111991 
Thanks, that compiled. But not getting anything on screen when flashing.
@tititopher68-dev 
Thanks again!

tititopher68-dev, 2021-08-20:
@Freethefire 
Comme je l'ai dit il faut modifier certains fichiers sources :
In Marlin/src/inc/Conditionals_LCD.h, line 1087 : 
Modify : 
#define LCD_SERIAL_PORT 3 // Creality 4.x board
By : 
#if MB(BTT_SKR_MINI_E3_V1_0, BTT_SKR_MINI_E3_V1_2, BTT_SKR_MINI_E3_V2_0, BTT_SKR_E3_TURBO)
  #define LCD_SERIAL_PORT 1
#else
  #define LCD_SERIAL_PORT 3 // Creality 4.x board
#endif

In Marlin/src/lcd/e3v2/creality/creality_dwin.cpp
Modify : 
while (LCD_SERIAL.availableForWrite() < 32)
By : 
while (!LCD_SERIAL.available())

And modify in Marlin/src/e3v2/creality/dwin.cpp : 
while (!LCD_SERIAL && PENDING(millis(), serial_connect_timeout)) { /*nada*/ }
By : 
while (!LCD_SERIAL.connected() && PENDING(millis(), serial_connect_timeout)) { /*nada*/ }

tititopher68-dev, 2021-08-20:
@Freethefire 
And configuration_adv.h
#define SHOW_REMAINING_TIME
To fix ui_get_remaining...

Freethefire, 2021-08-20:
I've redone it 3 times now, and I still get:

```
Compiling .pio\build\STM32F103RC_btt\src\src\lcd\e3v2\creality\rotary_encoder.cpp.o
Marlin\src\lcd\e3v2\creality\creality_dwin.cpp: In member function 'void CrealityDWINClass::Draw_Print_ProgressRemain()':
Marlin\src\lcd\e3v2\creality\creality_dwin.cpp:761:31: error: 'class MarlinUI' has no member named 'get_remaining_time'
  761 |   uint16_t remainingtime = ui.get_remaining_time();
```

I'm guessing I'm doing something wrong, even though your instructions are very thorough.

Freethefire, 2021-08-20:
@tititopher68-dev 
Thank you so much for your patients!
Building mesh now. 

What tripped me up was that:
while (!LCD_SERIAL.connected() && PENDING(millis(), serial_connect_timeout)) { /nada/ }
was supposed to be:
while (!LCD_SERIAL.connected() && PENDING(millis(), serial_connect_timeout)) { /*nada*/ }

closed 2021-08-20 by Freethefire

tititopher68-dev, 2021-08-20:
@Freethefire 
Ah OK !
Mesh correctly created ?
Do you use a BLTouch ? If yes can you do a M48 Test (menu advanced / Probe)...this give me a response for 3 others issues...

Freethefire, 2021-08-21:
@tititopher68-dev 
Yes, mesh created. At first I used UBL, but that only gives me so many random points, so I'm going with bilinear for extrapolation.
Bltouch clone results(still some glue residue on the build plate, will clean and update):
```
Recv: Mean: 0.000250 Min: -0.003 Max: 0.002 Range: 0.005
Recv: Standard Deviation: 0.002077
Recv: 
Recv: X:158.40 Y:120.50 Z:11.40 E:0.00 Count X:12672 Y:9640 Z:4560
```