Jyers/Marlin #1327: Updated to 1.3.5 now nothing is moving

Mountain3d, 2021-08-20:
E3v2
4.2.7

Fw-1.3.5

Updated last night. Everyone looks happy. Initiated abl mesh sequence and now there is no movement whatsoever. I will upload a copy of the .h and adv file whe. I get back to the farm.

Any ideas would be greatly appreciated!

tititopher68-dev, 2021-08-20:
@Mountain3d 
Just look at https://github.com/Jyers/Marlin/issues/1321#issuecomment-902971371

You can revert to 1.3.4 for the moment...

tititopher68-dev, 2021-08-21:
@Mountain3d 
The merge has just been done, all that remains is to compile and verify that everything is functional.

ZipperZ, 2021-08-31:
> @Mountain3d
> Just look at [#1321 (comment)](https://github.com/Jyers/Marlin/issues/1321#issuecomment-902971371)
> 
> You can revert to 1.3.4 for the moment...

Went back to 1.3.4, does not help... Yesterday it helped just to reflash 1.3.5.b.

Or this might be something else? As it worked right after flashing 1.3.5.b for one print. Then homing stopped working

gururise, 2021-10-20:
Having the same exact problem on 1.3.5b (Manual Mesh Firmware).  No movement whatsoever after attemping to mesh level.

Went back to DEFAULT 1.3.5b firmware and all is working.