Jyers/Marlin #1332: v1.3.5 - v1.3.5.b Ender 3 v2- Z Axis reset

Mr-Hick, 2021-08-22:
### Description

Z Offset resets on every printer restart, 
I tried using 
E3V2-UBL-BLTouch-10x10-v4.2.2-v1.3.5.bin
E3V2-UBL-BLTouch-10x10-v4.2.2-v1.3.5.b.bin
E3V2-UBL-BLTouch-15x15-HS-v4.2.2-v1.3.5.b.bin

### Steps to Reproduce

Change Z Offset in printer UI
Restart Printer
Z Offset is gone and needs to be set again.

**Expected behavior:** [What you expect to happen]
Configuration should stay saved in EPROM

**Actual behavior:** [What actually happens]
Configuration does not stay saved in EPROM
Z Offset resets every time.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tititopher68-dev, 2021-08-22:
@Mr-Hick 
The versions you are using are no longer the ones corresponding to the source code.  Also, it will be good to compile your own firmware using configuration files.
 To be sure, after flashing your firmwares, did you do a Reset to default, then Store settings?
 Always to be sure, after having defined your Z-Offset, did you do a SAVE?
 Because with me, regardless of the version of the Build, no problem with the backup, the value is always the one that was set at each power-up.

Mr-Hick, 2021-08-22:
Reseting to default fixed it.

Thank you

closed 2021-08-22 by Mr-Hick