Jyers/Marlin #1335: Filament Runout Sensor Reversed

Dragon41673, 2021-08-22:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I am trying to add a BIQU Filament Break Sensor (non-smart model) while filament is IN the sensor, the sensor tells the printer that filament is NOT in the sensor. When the filament is removed, it tells the printer that filament is IN the sensor. Is it possible to add a check box to reverse the read on the sensor to resolve this issue?

### Bug Timeline

New to me, just discovered a couple days ago

### Expected behavior

Provide a check box to change the sensor reading to allow proper reading of the filament sensor

### Actual behavior

Filament sensor reads wrong. When filament is inserted, printer reads it as the filament has run out. When filament is NOT in the sensor, printer reads it as there is filament in the sensor.

### Steps to Reproduce

1. Filament sensor is the basic BIQU Filament Break Sensor found here...
https://www.amazon.com/dp/B08QJDP36R/ref=dp_iou_view_product?ie=UTF8&psc=1

2. Just insert filament and you will see

### Version of Marlin Firmware

V1.3.4

### Printer model

Ender 3 V2

### Electronics

Stock

### Add-ons

Creality Direct Drive Extruder & BL Touch Sensor

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

None

Cal-LF, 2021-08-22:
The firmware does not support the filament state for hi or low and has been addressed in past posts. The filament state can be change by compiling your own firmware. Another option is to join Jyers patreon automated compile service to make your custom firmware. 
Reference #1215   #1309.
If you would like to get set up to compile your own,
MS VS Code is what I have used. I don't know of any other. I followed Teaching Tech video and a couple other videos to get going with it.
Download VS Code https://code.visualstudio.com/download
Next install Platform IO extension and Auto Build Marlin extension to complete.
https://www.youtube.com/watch?v=eq_ygvHF29I 2:29 time start to 7:30 time stop, for pertinent info.

Use one of Jyers E3V2 Templates copied to the Marlin folder as a starting point.

Dragon41673, 2021-08-23:
@Cal-LF is there a filament sensor that will actually work with this printer so I don't have to go through hell to get this to work? The BTT Smart Sensor didn't work. These simple mechanical ones don't either.

What's the point of having the Filament Runout setting in the firmware if it doesn't work?

Cal-LF, 2021-08-23:
There are many owners using the filament sensor so it does work.
Is it possible the plug to the main board is not in the proper place? Or is a wire broken? Test with a multimeter for continuity of the sensor switch wires back to the board plug.

Dragon41673, 2021-08-23:
Sensors (I have 2 Ender 3 V2's) are definitely working properly, but the printer is seeing them work in reverse. I've gone over specs & wiring, I know the plug going into the 4.2.2 board was wired incorrectly so I resolved that. Sensors are wired properly and I metered them to make sure. All works...they are just in reverse

Cal-LF, 2021-08-23:
Ok, that is more clear. The sensor state should be reversed from it's current setting. LOW to HIGH in the Config_h file. Or HIGH to LOW if that is the case.

Dragon41673, 2021-08-23:
Well...I'm definitely not a coder, so how much will this end up costing me? Any idea...AND I'm guessing I'll have to pay for the update over & over as new firmware editions are released IF I choose to update?

Cal-LF, 2021-08-23:
No cost to you if you learn the compile process. Coding is not needed, only editing a text file. Configuration_h in the downloaded zip file. If you want to try, I will guide you.

By the way, we may see the filament state menu option in the future but it's not supported by Marlin at this point. There is another firmware that does have this option made by mriscoc. It does work but is outside the Marlin firmware parameters.

Dragon41673, 2021-08-23:
I wouldn't mind learning, just a massively busy life, kids, 2 jobs, etc. Got these 2 months ago to do things for myself and show my kids...still haven't completed more than a couple XYZ cubes, a few Benchy boats, and 2 small buildings they saw on Thingiverse.

I sincerely appreciate your offer...please let me know what needs to be done, I'll go watch that video while I wait for your reply. Already have the VS Code installed (tried opening the BIN file and that obviously didn't work).

Cal-LF, 2021-08-23:
The .bin file is the result of compiling with VS Code so you can't undo it.
 If you want the latest update but not yet released, you can download the the zip file from the Green Code button on the main github page. Make sure you have the Default Branch on the Ender_3_JyersUI button to the left of the Green button.

Text file for compiling steps.
[Compiling Steps.txt](https://github.com/Jyers/Marlin/files/7028551/Compiling.Steps.txt)

Dragon41673, 2021-08-23:
Starting to work through your list...noticed the 5x5 mesh is not there. Any ideas? I think 10x10 is insane, and 3x3 is too little

Cal-LF, 2021-08-23:
In the config files you can choose 3, 4, 5, 6, etc. these are the grid settings. I use 4 for a 4X4 grid.
This text file if of changes I make for my personal setup. You can pick some of these but the line numbers may be different because I use Manual Mesh.
[Jyres Compile Changes.txt](https://github.com/Jyers/Marlin/files/7028642/Jyres.Compile.Changes.txt)

Dragon41673, 2021-08-23:
Ok...I found the filament sensor line (I'm pretty sure) in the config h file and changed the first one to HIGH (was low)...does this look correct or do I need to change ALL these lines from LOW to HIGH?

`#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define FIL_RUNOUT_ENABLED_DEFAULT false // Enable the sensor on startup. Override with M412 followed by M500.
  #define NUM_RUNOUT_SENSORS   1          // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.

  #define FIL_RUNOUT_STATE     HIGH        // Pin state indicating that filament is NOT present.
  #define FIL_RUNOUT_PULLUP               // Use internal pullup for filament runout pins.
  //#define FIL_RUNOUT_PULLDOWN           // Use internal pulldown for filament runout pins.
  //#define WATCH_ALL_RUNOUT_SENSORS      // Execute runout script on any triggering sensor, not only for the active extruder.
                                          // This is automatically enabled for MIXING_EXTRUDERs.

  // Override individually if the runout sensors vary
  //#define FIL_RUNOUT1_STATE LOW
  //#define FIL_RUNOUT1_PULLUP
  //#define FIL_RUNOUT1_PULLDOWN

  //#define FIL_RUNOUT2_STATE LOW
  //#define FIL_RUNOUT2_PULLUP
  //#define FIL_RUNOUT2_PULLDOWN

  //#define FIL_RUNOUT3_STATE LOW
  //#define FIL_RUNOUT3_PULLUP
  //#define FIL_RUNOUT3_PULLDOWN

  //#define FIL_RUNOUT4_STATE LOW
  //#define FIL_RUNOUT4_PULLUP
  //#define FIL_RUNOUT4_PULLDOWN

  //#define FIL_RUNOUT5_STATE LOW
  //#define FIL_RUNOUT5_PULLUP
  //#define FIL_RUNOUT5_PULLDOWN

  //#define FIL_RUNOUT6_STATE LOW
  //#define FIL_RUNOUT6_PULLUP
  //#define FIL_RUNOUT6_PULLDOWN

  //#define FIL_RUNOUT7_STATE LOW
  //#define FIL_RUNOUT7_PULLUP
  //#define FIL_RUNOUT7_PULLDOWN

  //#define FIL_RUNOUT8_STATE LOW
  //#define FIL_RUNOUT8_PULLUP
  //#define FIL_RUNOUT8_PULLDOWN`

Cal-LF, 2021-08-23:
No, only the runout state. Scroll up to the top of this section to see it.

Dragon41673, 2021-08-23:
Ok good then I did it right.

So...I'm still a little confused on the BL Touch mesh. Where do I set it to 5x5? Maybe I'm searching the wrong term

Cal-LF, 2021-08-23:
Sorry told you wrong.  #define FIL_RUNOUT_STATE     LOW        // Pin state indicating that filament is NOT present.

Dragon41673, 2021-08-23:
Cal...I don't think you did. Originally the following line was LOW, I switched it to HIGH

`#define FIL_RUNOUT_STATE HIGH // Pin state indicating that filament is NOT present.`

For the BL Touch Mesh..is it this?

`  #define GRID_MAX_POINTS_X 3    // Don't use more than 7 points per axis, implementation limited.
`

And I would change that to 5?

Cal-LF, 2021-08-23:
Yes on the grid. Ok on the runout state to High.

Dragon41673, 2021-08-23:
Well crap...this seems a lot easier than I thought, then again you are giving me a ton of info too. I found another line for leveling...

This is line 1593
`  #define GRID_MAX_POINTS_X 3`

The other one above is line 1646

Cal-LF, 2021-08-23:
Do you have the PIO and Marlin Auto Compile extensions installed?

Dragon41673, 2021-08-23:
> Do you have the PIO and Marlin Auto Compile extensions installed?

yes

Cal-LF, 2021-08-23:
You are correct, if you are using UBL. You might want to start with the BL Touch 3X3 to get a feel for the working of it. Same changes.

Dragon41673, 2021-08-23:
Actually I just noticed that they are completely separate leveling systems. I'm not using the UBL (that I know of) so I changed only the one on line 1593 which falls under

`#if EITHER(AUTO_BED_LEVELING_LINEAR, AUTO_BED_LEVELING_BILINEAR)`

Cal-LF, 2021-08-23:
Ok. Give the compile a try. Open the Marlin Auto compile, show ABM panel and click the top build button.

Dragon41673, 2021-08-23:
Well...I hope I did this right. Everything after compile was a bit crazy, maybe because it is the first time doing it. Giving it an upload now...

Dragon41673, 2021-08-23:
Ok...it flashed, says it is a Jyers 1.3.5 in the info section. It's currently doing a mesh right now and it seems to be doing a 5x5, so I will report back in a while once I start a print and manually cause a filament break to happen.

Cal-LF, 2021-08-23:
You may find some of the features are not the same as the precompiled files. The Config files are a starting point and other setting changes may be needed. Also you can have the settings already in your firmware that you might change in the menu screens.

Cal-LF, 2021-08-23:
I am excited for you to succeed with your first firmware creation.

Dragon41673, 2021-08-23:
LOL I am too...it's heating up now

Cal-LF, 2021-08-23:
Jyers has updated a couple times since last release. You can include those by using the zip file from the Green button. The BL Touch feature has most of the updates so you might give it a try.

Dragon41673, 2021-08-23:
I did carry over the config files from the 3x3 BL Touch folder...is that what you mean?

All heated up, doing a bed level before print (forgot that was a step, could have skipped the pre-level one lol)

Cal-LF, 2021-08-23:
Yes, but those are from the V1.3.5 release. I am referring to the latest zip file from the Green button.

Dragon41673, 2021-08-23:
Ah ok...well, it works! Thank you very much sir! It's very late here, and I have to be up early in the morning, hopefully I can play with it tomorrow or Tuesday and get prints going. I REALLY appreciate your help on this! You've been pretty much the only person that has helped after asking in forums, Facebook, and support threads. THANK YOU

Cal-LF, 2021-08-23:
These are some of my designs from Tinkercad that I have recently been working on. 
https://www.thingiverse.com/vtxrider1/designs

Cal-LF, 2021-08-23:
You can close this Issue if you want.

Dragon41673, 2021-08-23:
> These are some of my designs from Tinkercad that I have recently been working on.
> https://www.thingiverse.com/vtxrider1/designs

Thank you...ill take a peek possibly tomorrow!

closed 2021-08-23 by Dragon41673

Cal-LF, 2021-08-23:
@Dragon41673 When flashing firmware, the eprom has to be cleared of any data saved from the previous install. 
Reset to defaults, restore settings, save settings to clear and save any new or changed settings. Save settings each time changes are made.
The mesh made from the BL Touch is not cleared because it is saved in it's own bunker. Manual mesh is cleared and needs to be redone.

Dragon41673, 2021-08-23:
No problem sir...thank you very much!