Jyers/Marlin #1338: Fillament Runout Sensor - False Positive at the Start of print

HermanGaviriaMarin, 2021-08-23:
### Description
Several times (not necessarily every single time) when a print job is sent and begins the filament runout sensor creates a sort of false positive reading on the runout distance, this has happened for me at the purge line (always at the purge line, no where else). I can confirm that my nozzle is not scrapping the bed (trust me I've checked several times), to me it seems that the issue is almost queued up to happen at the exact same spot (somewhat middle of purge line). Tbh its not a huge issue but its very annoying, specially because you have to wait till it unloads and loads filament. 

### Steps to Reproduce
Either flash precompiled firmware (latest version) in the releases tab or compile your own firmware (without changing anything regarding runout sensor--I usually just change max temp and thermistor sensor).

1. Load up firmware (its happened throughout several version of the firmware, probably since before version 1.3.4)
2. Send a print job either through Octoprint or MicroSD (I broke my Micro SD card port on the printer so Octorprint)
3. Hope that it doesn't say "Filament runout" as soon as it starts printing (Usually sends the message as soon as any material begins to be extruded (which does extrude and come out without any issue or clicks on the extruder)

4. Wait for material to be Unloaded and loaded again
5. Occasionally Stays Frozen on filament runout and doesn't give option to continue print (only way which sometimes works is to hit continue from octoprint, but doesn't always works)

**Expected behavior:** 
Print should begin without sending a runout sensor message since it hasn't even begun to extrude, moreover when it begins to make the purge line it extrudes it and lays it down perfectly like there isn't a jam or anything wrong

**Actual behavior:** 
Well I kinda already explained it -- I'm just filling out this template since its more organized.

#### Additional Information

[JYERS. Herman CONFIG.zip](https://github.com/Jyers/Marlin/files/7033705/JYERS.Herman.CONFIG.zip)

HermanGaviriaMarin, 2021-08-23:
> ### Description
> Several times (not necessarily every single time) when a print job is sent and begins the filament runout sensor creates a sort of false positive reading on the runout distance, this [...]

Cal-LF, 2021-08-23:
Hi Herman
Your config is not set up for the BTT sensor. That is why it is not working properly.

Try the Fil_Runout_State changed to HIGH if yours is not working.
#define FIL_RUNOUT_STATE     LOW 
This may have to be set to HIGH to work with some sensors.
 #define FILAMENT_RUNOUT_DISTANCE_MM 10 or more.
 #define FILAMENT_MOTION_SENSOR
Need to adjust the filament load and unload distance to fit my total feed length.

HermanGaviriaMarin, 2021-08-23:
Cal, 
So let me just confirm what needs to be done, I will go into the source code and go either on Config or Config_adv and there I will uncomment those line which you just stated? Will this require me to comment out "//" some lines, like the ones that were defined at first ?

Cal-LF, 2021-08-23:
Yes, the configuation_h file is where the lines shown above are located.
Control/F will open a search window to type in key words. Like fil_runout_state or motion_sensor.

HermanGaviriaMarin, 2021-08-23:
Lol I understand that part, what I'm asking refers to the previous configuration, does uncommenting #define FIL_RUNOUT_STATE LOW,#define FILAMENT_RUNOUT_DISTANCE_MM 10 or more, and #define FILAMENT_MOTION_SENSOR cause some sort of conflict with the previous lines that acted for how the run out sensor used to work. Basically will I have to undefine (comment out) previous lines in the firmware. Also thank you for the comprehensive explanations.

Cal-LF, 2021-08-23:
You may have to change the LOW to HIGH in the #define FIL_RUNOUT_STATE LOW if it does not work. Mine is set to LOW but others have had to change. This is already commented out.

#define FILAMENT_RUNOUT_DISTANCE_MM 10 or more. Yours is set at 0 so change that to 10.

Uncomment the #define FILAMENT_MOTION_SENSOR

TechFrQ, 2021-09-04:
I have a similar problem but I need to invert the sensor. I was able to find a command to do it
#define FIL_RUNOUT_INVERTING true       // set to true to invert the logic of the sensor.
Does this command only go in Configuration.h or is it required in Configuration_adv.h as well? 

Thank you!

Cal-LF, 2021-09-05:
Only in the config.h