Jyers/Marlin #1339: Baudrate

eblieb, 2021-08-24:
Was wondering if there is any reason the firmware doesnt set the baud rate to 250000,

tititopher68-dev, 2021-08-25:
@eblieb
By default, Creality set the baudrate to 115200 for compatibility with their Creality Cloud Box.

In my own build, I integrated a new feature allowing to modify the BaudRate from the interface.  By default, it is set to 250,000.

Note also that by activating the possibility of using the M575 command in configuration.h and by configuring an action command with the latest build of Jyers, it is possible to have this same type of functionality ...

eblieb, 2021-08-31:
@tititopher68-dev is there any chance that new feature can be merged into the main Jyers branch?

tititopher68-dev, 2021-08-31:
@eblieb 
It's all up to @Jyers....