Jyers/Marlin #1340: Octoprint

mccolister1017, 2021-08-24:
the mesh is clean on the printer all green but on octoprint i got this and i thik this is the real mesh because when i print with my printer  its all bad ...

[image]

mccolister1017, 2021-08-24:
how can if fix this or what did i do wrong ?

mccolister1017, 2021-08-25:
[image]

Chronoseifer, 2021-09-03:
Did you press 'update mesh now'? Usually takes a couple of minutes.

otherwise, do an auto bed level on the ender 3 V2, and 'save' the results. then you could try it again.

DasSmeagle, 2021-09-07:
My advice would be to read, understand and use this:

https://marlinfw.org/docs/features/unified_bed_leveling.html