Jyers/Marlin #1345: Big Tree Tech filament sensor

jwkopp, 2021-08-25:
With filament sensor enabled the BTT sensor 8/10 times throws a filament runout prompt while doing the initial purge line on the side of the bed.  I'm using the 4.2.7 10x10 bltouch hs firmware.

tititopher68-dev, 2021-08-25:
@jwkopp 
Again, a little read on the discussion section would have given you the answer.
 A BTT Smartsensor end of filament detector is a so-called encoder wheel detector.  The detector therefore regularly sends a signal 0v 5v 0v 5v etc ...
 Also the firmware used is not, as it is, configured for this type of detector.  You must therefore compile your own firmware, by modifying the configuration.h file.
 It is necessary how out : 
 #define FILAMENT_MOTION_SENSOR
 #define FILAMENT_RUNOUT_DISTANCE_MM 15
 (this parameter can be modified in the interface).
 And make sure you have: #define FIL_RUNOUT_STATE LOW

BlueWings172, 2021-08-29:
I used the auto-compile features and enabled the 3 lines above. (I'm using High state as low didn't work for me)

After many tries, now the sensor triggers after the initial purge line just as @jwkopp described.

It doesn't seem to me that this sensor is compatible with this firmware.

jwkopp, 2021-08-29:
> I used the auto-compile features and enabled the 3 lines above. (I'm using High state as low didn't work for me)
> 
> 
> 
> After many tries, now the sensor triggers after the initial purge line just as @jwkopp described.
> 
> 
> 
> It doesn't seem to me that this sensor is compatible with this firmware.
> 
> 
If you scroll down further you will see filament runout distance.  Make sure to uncomment that and put 7 mm also right below there is another setting that needs to be uncommented.  Mine works fine in low state as well.  Let me know if you need any more help.

BlueWings172, 2021-08-29:
I have already set my runout distance to 10 mm. There is an issue either with these BTT sensors, or the firmware.

jwkopp, 2021-08-29:
> I have already set my runout distance to 10 mm. There is an issue either with these BTT sensors, or the firmware.

Shoot me an email usankopp@aol.com. I'll send you mine to try out.  Do you have the 4.2.2 board or the 4.2.7?

BlueWings172, 2021-08-29:
I have the 4.2.2 board. I sent you the email. I also attached the config file showing the last settings that I used for the auto-compile. I used low state before that but it did not work.

Thanks for your help.

jwkopp, 2021-08-29:
Cool when I get home I’ll take a look.  Do you have bltouch as well?

BlueWings172, 2021-08-29:
I do have BLtouch and I use the UBL-BLTouch-15x15 firmware.

jwkopp, 2021-08-29:
Try this file out.  It’s a 9x9 pobe grid that I like to use because its very accurate and doesn’t take long to do.  I like to probe before every print.  Try it and then let me know if you still want a more probing.  Also don’t forget to go in and set your probe margins for x and y.  the ones here are for my set up and prob wont work for you.  

From: BlueWings172
Sent: Sunday, August 29, 2021 10:31 AM
To: Jyers/Marlin
Cc: jwkopp; Mention
Subject: Re: [Jyers/Marlin] Big Tree Tech filament sensor (#1345)

I do have BLtouch and I use the UBL-BLTouch-15x15 firmware.

BlueWings172, 2021-08-31:
I have tried the firmware but i faced an issue with probing. The printer
starts probing very close to the edge of the bed and the probe would dive
into the metallic clips that hold the glass bed, if I don't stop it. I
tried to play with the x and y home offset and I've set the probe offsets
to the numbers I used before (-44 and -7).

Is there another offset setting that I need to change to ensure that the
probe is at a safe distance from the clips when leveling ?

jwkopp, 2021-08-31:
Would seem as thought the y probe offset is incorrect.  I would adjust it out to -15 and keep x at-44 and try again.

BlueWings172, 2021-09-02:
@jwkopp, I changed Y probe offset to -15 as you suggested and it worked. Thanks for the tip.

Unfortunately, the firmware you sent behaved exactly like mine and the sensor triggers immediately after the purge line. That said, the Filament Sensor check box remained checked unlike what happens with my firmware.

I hope future updates will solve this issue as it is not happening only with my printer.

thanks so much for your help though.

jwkopp, 2021-09-02:
That is weird I haven’t had any issues.  I’ll look at it again and see if I missed something.