Jyers/Marlin #1351: Screen glitch issue still occurring - commit included in details

melvinmicallef, 2021-08-28:
### Description

I have manually compiled the latest commit (fac4bf44e1033bbf6a53a2e79331685492054c90) and when i start printing, the screen start showing random pixels. I am using octoprint and have everything configured correctly. Previously, i was on the pre-sync with marlin version and everything worked fine. Any ideas please? Should i go on the v1.3.5b commit?

### Steps to Reproduce

1. Print and look at the screen :D

**Expected behavior:** Screen renders properly

**Actual behavior:** Screen does not render properly

#### Additional Information
[image]

tititopher68-dev, 2021-08-28:
@melvinmicallef 
This issue exist only with octoprint.
But I don't have this issue (never had !) with baudrate setting at 250000. Modify configuration.h with this value and compile again...

tititopher68-dev, 2021-08-28:
[image]

melvinmicallef, 2021-08-29:
I already have the 5v on my usb cable covered. Changing the baud rate to 250000 seems to fix the issue 👍 Thank you @tititopher68-dev

closed 2021-08-29 by melvinmicallef