Jyers/Marlin #1356: Feature request: Uncompiled code for all the preconfigured binaries?

basspaul, 2021-08-28:
Is there a source where we can get the uncompiled code that what used to create the binary files in the release folder? 

It would be much easier to take one of the preconfigured ones and slightly tweak it than starting from the "generic" Jyers code and re-enabling all the features that have been enable in the pre-built files.

I'm new to git, so I may be just blind to where the files are.

thanks

Cal-LF, 2021-08-29:
Ender_3_V2_JyersUI/Configuration Files/E3V2 Templates

basspaul, 2021-08-29:
Thanks. The 2 months ago tag on the files in there threw me off as they didn't look like current.

(non) Issue closed.

closed 2021-08-29 by basspaul

Cal-LF, 2021-08-29:
@basspaul The date for the config files is the last time they were updated. The last couple updates to the firmware did not change the config files but they are still current. Most changes occur in the menus and from the base marlin updates.