Jyers/Marlin #1361: Feature Request: Ability to Restart Last Print

9ae8sdf76, 2021-08-30:
### Description
Preface: I use Octoprint to manage my Ender 3 Pro v2's prints via a Raspberry Pi that's connected to the printer via the USB, and accessed via wireless over my intranet. My desktop and printer are physically several feet away from each other. I do use a CRTouch and the 15x15 UBL version of your fork for bed leveling. Yet, when printing I find that the z-offset needs minor tweaking to get the right depth for adhesion and flow rate.

Because of the physical distance between my printer and desktop, having to restart a print can become a tedious task. Adding a third button to the Stop confirmation - one that allowed restarting the print (after allowing the user to clear the bed) - would greatly enhance the tedious task of tweaking settings on the fly. I could also see a use case where something happens mid-print that necessitates restarting (the print bed getting knocked out of alignment, spaghetti prints, warping, etc). 

### Steps to Reproduce

N/A

**Expected behavior:** Selecting Stop during a print presents the normal confirmation, with an addition **Restart** button that would stop the current print, disable the steppers, and ask the user to confirm before restarting the same print job it just stopped. 

**Actual behavior:** N/A

#### Additional Information

The confirmation for restarting the print could even be split into two additional buttons: 

1. Restart with Settings: restart the print using whatever settings the printer currently knows about - including any on-the-fly changes made using the Tune menu
2. Restart: restart the print using the gcode settings and ignoring any on-the-fly changes made from the Tune menu

Thank you.

Cal-LF, 2021-08-30:
I like your idea. I am restarting a failed-for-what-ever-reason print.

IrishBrewer, 2021-09-08:
As far as this goes, it would also be nice to repeat last print after a successful print.  Better yet, be able to select from a queue of last 5-10 prints, if possible.

Jyers, 2021-09-09:
This will actually be achievable with the newest/upcoming octoprint integrations. Such as host action commands and print starting. Although it won't be in the stop dialog, it will allow you to easily restart a print from the screen

closed 2021-09-09 by Jyers

9ae8sdf76, 2021-09-09:
@Jyers does that mean we could restart prints from the printer itself? or would we still need to physically move between octoprint and the printer?

9ae8sdf76, 2021-09-09:
@Jyers Thank you! I understand now.