Jyers/Marlin #1362: "baby stepping always" is not working

tfaz099, 2021-08-30:
### Description

Normal operation of BABYSTEP_ALWAYS_AVAILABLE is that anytime you go to the z-offset menu and click once, you can move z-offset up and down and the z-axis will always move with it regardless of the state of the printer.

### Steps to Reproduce

1. Home all axis (not necessary but good practice)
2. Go to prepare menu
3. Scroll to z-offset and select
4. move dial left or right

**Expected behavior:** 
z-axis should move when dial is being turned all the time regardless of printing state

**Actual behavior:**
z-axis does not move until the button is clicked (good way to crash hot end into the bed)

#### Additional Information

*a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files is attached.
[My Config Files.zip](https://github.com/Jyers/Marlin/files/7080029/My.Config.Files.zip)

tititopher68-dev, 2021-08-31:
@tfaz099 
BABYSTEP_ALWAYS_AVAILABLE is not used in this firmware in connection with the principle of the screen interface and what our screen allows and it is a desired choice .... and preferable besides.

To keep it simple, we directly modify the ultimately really necessary variable defined in Marlin without going through intermediaries ... By activating the 'Live Adjustment', you have the equivalent of this function.  Coded in this way, it provides additional security in particular so as not to destroy the Bed if you are not careful ...

This subject has already been mentioned, at least 4 months ago.

tfaz099, 2021-08-31:
WHAT ?!?!? 
First clicking the checkbox does not enable it the way it used to work in any way shape or form.
Second, it was designed to allow you to move the print head up and down during that firtst layer so you get the right squish... as far as security of not crashing into the bed... well that is the chance you take using baby stepping.

Without this function almost everything else is useless.... anyone claiming it is just ok to let it be pulled out of how it is SUPPOSED to work is just not right in their understanding of the entire spirit of that code.

PLEASE PLEASE PLEASE put in a #DEFINE to put that function back the way it worked in SMITH3D.... no checkbox and it is live all the time when you open up z-Offset.

thank you

P.S. on the fact that it was discussed months ago... well i am guessing a lot of people had no clue how important that function is the way it was designed.

tfaz099, 2021-08-31:
Also... if the "checkbox" for live babystepping was the only difference (i.e. no homing) and it made it work like it used to that would be fine.  there is absolutely no reason to home because it does not matter where "Z" is.  that stepping function moves z up or down in the baby steps defined and click to save when done... and store config to write to eprom.

*THIS* is part and parcel of my skepticism of changing to someone elses firmware when changes are made to very core functions that have been using them flawlessly, and now need to revert back to Smith3D... this is not what his intentions were when he threw in the towel i am almost certain.

tititopher68-dev, 2021-08-31:
@tfaz099 
Why so much aggression?
What is proposed here works very well ... So good that the Project Marlin manager will officially adopt it for the Ender3V2 (https://github.com/MarlinFirmware/Marlin/pull/22422)....Smith3D has given up development because the basic Creality source code (on which it was designed) no longer followed the considerations and the latest developments wanted by Marlin .... And the creation of new functionalities became a real gas factory. Jyers facilitated the development of the code and therefore the possibility of bringing new elements. Now you are free to watch the videos on Youtube about this firmware (or on my version which is based on that of Jyers with additional functions). All the test videos highlight the quality of the firmware including Bed Leveling and the Z-Offset features offered.

So there is only one more thing to say. If you prefer the older version of Smith3D, continue to use it if it's convenient for you. You are completely free and no one is forcing you to adopt Jyers' firmware.

tfaz099, 2021-08-31:
There’s no aggression just a great amount of disappointment. 

I have been a developer since the 90s and would never remove the functionality that has become a core part of many people that use this software. Smith3D was one of the big developers that got all of these functions working and strongly endorses this replacement set a firmware. I was convinced to switch to it because it was continuing that legacy and bringing with it new functionality and new trains of thought. Everything else I see here is very good but turning off functionality that has been one of the best parts of previous versions of this firmware is just a tragedy.

“Don’t adapt to the energy in the room, influence it” 

What is proposed here works very well ... So good that the Project Marlin manager will officially adopt it for the Ender3V2 (MarlinFirmware#22422 has given up development because the basic Creality source code (on which it was designed) no longer followed the considerations and the latest developments wanted by Marlin .... And the creation of new functionalities became a real gas factory. Jyers facilitated the development of the code and therefore the possibility of bringing new elements. Now you are free to watch the videos on Youtube about this firmware (or on my version which is based on that of Jyers with additional functions). All the test videos highlight the quality of the firmware including Bed Leveling and the Z-Offset features offered.

tititopher68-dev, 2021-08-31:
@tfaz099 
Frankly, I don't understand you, sorry!
But you write: z-axis does not move until the button is clicked (good way to crash hot end into the bed)
It's wrong !
During printing, via the TUNE menu, you can modify the Z-Offset directly live, either by selecting the Z-Offset item and turning the dial, or by clicking (and even holding down!) On MicroStep. UP and Down (by 0.01 mm)!
This is exactly what I answered you in the discussion section (# 1360) and there, you act as if you do not understand!

Note: having used Smith3D firmware for a long time, I can also tell you that modifying a parameter also caused malfunctions linked to the design of this firmware.
But now if you're a developer, just do like me (and I'm not that young at all anymore), add your own functions as you like....

tfaz099, 2021-08-31:
I am not seeing what is hard to understand. 

Go to prepare 
Select z-offset
Click it
Turn dial right or left and z axis moves up or down in babysteps no matter if the printer is idle or printing. It just works without jumping through hoops

And i no longer code….i like my sleep lol

“Don’t adapt to the energy in the room, influence it”

tfaz099, 2021-08-31:
Part of the reason Marlin so easily accepts your stuff is that it is excellent.  If it were not for developers like yourselves we would be left with whatever Marlin spits out.  So please do not take what i said personally, but also Marlin does not have the group of people contributing like you do, nor the willingness to accept it.  So it is no surprise they will snatch it right up.  I have tried the babystepping with tune during printing... why is my z-offset 2.88 on Smith and needs to be 2.99 on Jyers? and nothing laying down correctly on jyers.

On the babystepping during print, using tune menu, it is not moving live....it seems to move the very first time i click babystep down, and thats it.  i am sure for most this flavor is fine, but does not seem to work right so far... i have only tried printing a bed level grid and that is horrible so far on jyers a 3 and an 8 or 9 on smith.  

is there an e-step think i am missing that is somehow different on jyers?

I refer to smith for one reason, it was very good... not too many bells and whistles, but since he has dropped it because jyers took over... we are all left with you or nothing...if we want to move forward.  It is not a matter of just revert to something else, Marlin will have updates and Smith will no longer move up with them.  TH3D has totally bastardized their version and no one i talked to can get theirs to print right that has tried that firmware.  

So, what is the issue with leaving an option to leave babystepping always anytime available... it is the only thing so far that is not good.  Aside from my now digging into why a known print wont print right under this firmware.

Hope you can see my point of view... i am very user oriented and have helped people from all over deal with their printer issues and compile or share my working binaries with them with great success... 

Thank you for hearing me out.

tititopher68-dev, 2021-09-01:
@tfaz099 
I reread the Smith3D code last night.  It uses an M851 Z'value 'command to modify the Z-Offset and M290 Z' +/- 0.01 'for the babystep

 It is better to avoid injecting gcodes via the serial bus as much as possible.
 In the firmware, the variable corresponding to the Z-Offset is directly modified and for the babystep, the same M290 command is used.

I didn't understand your babystep homing thing ... and for good reason.  Indeed in my build, I activated BAYBYSTEP_WITHOUT_HOMING, which Jyers did not choose.

Now in the code Marlin, BABYSTEP _..._ ALWAYS would correspond in the firmware in the Prepare Menu to : Live Adjustment enabled + BAYBYSTEP_WITHOUT_HOMING enabled.
And in the Tune Menu, just BAYBYSTEP_WITHOUT_HOMING activated

Also first, enable #define BAYBYSTEP_WITHOUT_HOMING in configuration_adv.h
And then I will submit to Jyers a very small modification by adding a simple condition like #ifdef BABYSTEP...ALWAYS, disabling the Live Adjustment item and forcing the corresponding Boolean variable.  Thus the 2 "options" will be possible ....

tfaz099, 2021-09-01:
Awesome 😊

Thank you for entertaining that

“Once you choose hope, anything’s possible”

tititopher68-dev, 2021-09-01:
@tfaz099 
The code : 
In creality_dwin.h (Marlin/src/lcd/e3v2/creality/)

add in line 33 : 
#ifndef BABYSTEP_WITHOUT_HOMING
  #define BABYSTEP_WITHOUT_HOMING
#endif

In creality_dwin.cpp (Marlin/src/lcd/e3v2/creality/)
line 1684, modify :
 
#define ZOFFSET_MODE (ZOFFSET_HOME + 1) 

by :

#define ZOFFSET_MODE (ZOFFSET_HOME + DISABLED(BABYSTEP_ALWAYS_AVAILABLE))

line 1268, modify :

Draw_Menu(ZOffset);

by :

#ifdef BABYSTEP_ALWAYS_AVAILABLE
  if (!liveadjust) {
	if (axes_should_home()) {
	 Popup_Handler(Home);
	 gcode.home_all_axes(true);
	}
	Popup_Handler(MoveWait);
	#if ENABLED(Z_SAFE_HOMING)
	  planner.synchronize();
	  sprintf_P(cmd, PSTR("G0 F4000 X%s Y%s"), dtostrf(Z_SAFE_HOMING_X_POINT, 1, 3, str_1), dtostrf(Z_SAFE_HOMING_Y_POINT, 1, 3, str_2));
	  gcode.process_subcommands_now_P(cmd);
    #else
	  gcode.process_subcommands_now_P(PSTR("G0 F4000 X117.5 Y117.5"));
    #endif
	gcode.process_subcommands_now_P(PSTR("G0 F300 Z0"));
	planner.synchronize();
   }
  Liveadjust = true;
#endif
Draw_Menu(ZOffset);

	
line 1704, modify :

case ZOFFSET_HOME:

by :

#ifndef BABYSTEP_ALWAYS_AVAILABLE
	case ZOFFSET_HOME:
	  if (draw) {
        Draw_Menu_Item(row, ICON_Zoffset, "Live Adjustment");
        Draw_Checkbox(row, liveadjust);
       }
       else {
         if (!liveadjust) {
           if (axes_should_home()) {
             Popup_Handler(Home);
             gcode.home_all_axes(true);
            }
            Popup_Handler(MoveWait);
            #if ENABLED(Z_SAFE_HOMING)
               planner.synchronize();
               sprintf_P(cmd, PSTR("G0 F4000 X%s Y%s"), dtostrf(Z_SAFE_HOMING_X_POINT, 1, 3, str_1), dtostrf(Z_SAFE_HOMING_Y_POINT, 1, 3, str_2));
               gcode.process_subcommands_now_P(cmd);
             #else
               gcode.process_subcommands_now_P(PSTR("G0 F4000 X117.5 Y117.5"));
             #endif
             gcode.process_subcommands_now_P(PSTR("G0 F300 Z0"));
             planner.synchronize();
             Redraw_Menu();
            }
           Liveadjust = !liveadjust;
         Draw_Checkbox(row, liveadjust);
       }
      break;
#endif

So, if you set #define BABYSTEP_ALWAYS_AVAILABLE in configuration_adv.h, you will find Smith3D mode again, if you don't define it, it will be Jyers mode ....
On the other hand I made sure to always have BABYSTEP_WITHOUT_HOMING activated .... even if we forget it.

tfaz099, 2021-09-01:
Thank you. I will try this tonight 

“Don’t adapt to the energy in the room, influence it”

tititopher68-dev, 2021-09-01:
@tfaz099 
I haven't tested the code (I would do it tomorrow morning), but it should be fine ...

tfaz099, 2021-09-01:
IM not in a rush. So, l'l wait for you

Sent from my Apple Watch

tfaz099, 2021-09-01:
I actually went ahead and tried it… didn’t work out as “smith babystepping”
Same as jyers before but there is no “live” option but it does not move up and down live… *and* funny… it autohomed on its own also when i click z-offest.

I also attached my .BIN from the smith I run if you want to see how it works.

I really like what you guys have done, especially because there are other options you give us that we can use or not if we want…. But this one is important

Thank you
And please let me know where I can buy ya a cup of coffee or a beer… I don’t even know where you are from.

tony

creality_dwin.h
creality_dwin.cpp

“Once you choose hope, anything’s possible”

tititopher68-dev, 2021-09-02:
@tfaz099
For homing at the start, this is only done if the position of the axes is no longer known (the steppers are deactivated).
In Smith3D, it's also like that. Unless of course, just before doing the Z-Offset, you did an AutoHome in the 120s!
So it is advisable to check the position of the axes!

What I haven't checked (and what I assumed), the (BABYSTEP_ALWAYS_AVAILABLE) feature is no longer used on Cartesian printers, only for Core XY and Delta! Which explains why the function does nothing ...

It's Marlin's choice! Sorry. cf babystep.cpp in Marlin / src / feature ...

I didn't understand the notion of Live .... That I took for Live Adjustment .... Idiot that I am ... or too Tired of doing several things at the same time .... Now I think I understood and I finally think to be able to answer your problem ... Again I do not know if Jyers will follow me ..... (@ Jyers I just modify the code to be like Live movement ... in summary, I move the condition :
'if (active_menu == ZOffset && liveadjust) {.....'

tfaz099, 2021-09-02:
Well, i thank you for looking into it. 
Somehow i do not think Marlin has abandoned that feature. Its been around a bit. TH3D couldnt get it working correctly either. Smith was the first to have it working. And BTT I believe has it working as my son uses it all the time also. 

If you want my smith folder which was last released in may, with only that and an offset change in the config it is there.  Also i notice there is one setting that smith has as 40 and in jyers its changed to a 1 with relation to stepping.  It was discussed in jyers github but i left the 1….i should try the 40. I wish I remembered what it was noted being used for. But at work so cannot look. 

The basic difference is with smith i select z offset and turn knob and z moves. And no it does not auto home ever even if i just turn printer on and go to z offset. 

With even your latest modification (which does not work as stated above). I have to select z offset. Then turn dial and click. Turn dial and click for each adjustment so cant do a fluid adjustment.  When you have a bed leveled to within .001 almost everywhere that fluid z offset baby stepping “any time” is the best.

Where you from?

Tony

“Don’t adapt to the energy in the room, influence it”

tititopher68-dev, 2021-09-02:
@tfaz099 
on efface tout le code modifié et on va faire cela : 

In creality_dwin.h (Marlin/src/lcd/e3v2/creality/)

add in line 33 :
#ifndef BABYSTEP_WITHOUT_HOMING
#define BABYSTEP_WITHOUT_HOMING
#endif

In creality_dwin.cpp (Marlin/src/lcd/e3v2/creality/) : 

Move (line 5008 to 5018 inclus) just after line 5063and before '}' line 5064 : 

if (active_menu == ZOffset && liveadjust) {
      planner.synchronize();
      current_position.z += (tempvalue/valueunit - zoffsetvalue);
      planner.buffer_line(current_position, homing_feedrate(Z_AXIS), active_extruder);
      current_position.z = 0;
      sync_plan_position();
    }
    else if (active_menu == Tune && selection == TUNE_ZOFFSET) {
      sprintf_P(cmd, PSTR("M290 Z%s"), dtostrf((tempvalue/valueunit - zoffsetvalue), 1, 3, str_1));
      gcode.process_subcommands_now_P(cmd);
    }

Next, add just after 'if (active_menu == ZOffset && liveadjust) {' : 

TERN_(BABYSTEP_ALWAYS_AVAILABLE, gcode.reset_stepper_timeout());
*(float*)valuepointer = tempvalue/valueunit;

Next modify line 4996 : 

if (encoder_diffState == ENCODER_DIFF_NO) return;

by : 

if (encoder_diffState == ENCODER_DIFF_NO) {
  if (active_menu == ZOffset && liveadjust)     (TERN_(BABYSTEP_ALWAYS_AVAILABLE, gcode.reset_stepper_timeout());
  return;
}

To have the "Live" function, you must check Live Adjustment !
If you do an Autohome within 120s before going to the Z-Offset submenu, enabling Live Adjustment and enabling BABYSTEP_ALWAYS_AVAILABLE in configuration_adv.h, there will be no new Autohome. Otherwise there will only be one Autohome during the Z-Offset adjustment process !

tititopher68-dev, 2021-09-02:
@tfaz099 
Exactly, you are correct for #define BABYSTEP_MULTIPLICATOR_Z  1 -> #define BABYSTEP_MULTIPLICATOR_Z  40 !

tititopher68-dev, 2021-09-02:
@tfaz099 
Where you from? France !

tfaz099, 2021-09-02:
I am in upstate NY US

“Don’t adapt to the energy in the room, influence it”

tfaz099, 2021-09-02:
So i need to add this in addition to what i did yesterday?

Did you drop a new set of files out there i could grab?  I should put jyers back to normal for now and make a copy to try your mod

“Don’t adapt to the energy in the room, influence it”

tititopher68-dev, 2021-09-02:
> So i need to add this in addition to what i did yesterday? 
No, we are starting again from the last build of Jyers... We therefore forget the first version ...

tfaz099, 2021-09-02:
The first set you sent me was very clear 😊 of course line numbers changed once I added/moved lines.

Do you have the two files you can attach ? then I’ll juct change the #define for always use babystepping and that value from 1 to 40 and compile.

Thank you.

Tony

Why I am asking for the files…:

I must be missing something after I moved 5008 through 5018 to just after 5063 before the }

What do I do with that whole next section ?

if (active_menu == ZOffset && liveadjust) {
planner.synchronize();
current_position.z += (tempvalue/valueunit - zoffsetvalue);
planner.buffer_line(current_position, homing_feedrate(Z_AXIS), active_extruder);
current_position.z = 0;
sync_plan_position();
}
else if (active_menu == Tune && selection == TUNE_ZOFFSET) {
sprintf_P(cmd, PSTR("M290 Z%s"), dtostrf((tempvalue/valueunit - zoffsetvalue), 1, 3, str_1));
gcode.process_subcommands_now_P(cmd);
}

“Once you choose hope, anything’s possible”

tititopher68-dev, 2021-09-03:
@tfaz099 
The mod files : 
[creality_dwin.zip](https://github.com/Jyers/Marlin/files/7106123/creality_dwin.zip)

tfaz099, 2021-09-03:
Awesome my friend 

Thank you

“Don’t adapt to the energy in the room, influence it”

tititopher68-dev, 2021-09-03:
@tfaz099 
I modified 2 things....
[creality_dwin.zip](https://github.com/Jyers/Marlin/files/7107913/creality_dwin.zip)

tfaz099, 2021-09-03:
Cool. Ill try this when i get home from work this evening 

“Don’t adapt to the energy in the room, influence it”

tfaz099, 2021-09-04:
Hiya,
Copied in your sources
Modified for baby stepping always
(should I leave babystepping without homing commented out also?)

Anyway it failed… I attached the terminal output from the compiler.

Hope you have a good weekend

tony

“Once you choose hope, anything’s possible”

tititopher68-dev, 2021-09-04:
@tfaz099 
Sorry but no attached terminal output...
Have you choose 'creality_dwin2.zip' ?

tititopher68-dev, 2021-09-04:
@tfaz099
Sorry but no attached terminal output...
Ah ouppps ! a '(' too much....Fixed and compilation is now OK....Don't forget for you to #define BABYSTEP_ALWAYS_AVAILABLE in your configuration_adv.h
[creality_dwin3.zip](https://github.com/Jyers/Marlin/files/7109649/creality_dwin3.zip)

tititopher68-dev, 2021-09-04:
@Jyers 
Could you look at the code? I think it would be interesting to integrate it in the future release.
I open another parenthesis: it would also be desirable to define 2 digits after the comma for the Steps menu ... This makes the printer calibration more precise, especially for the extruder (if you have a BMG or a Titan ) .... and then it only asks to change the units from 10 to 100 ...

tfaz099, 2021-09-04:
Thank you I will try this morning

“Once you choose hope, anything’s possible”

tfaz099, 2021-09-04:
Good evening
I started fresh so ever file was pristine
Unzipped Jyers
Copied Jyers config into marlin
Unzipped DWIN3 Zip from you into folder and over wrote the default marlin files

This is the terminal

n file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:49,
                 from Marlin/src/HAL/STM32/HAL.cpp:30:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:1207:33: error: static assertion failed: It is strongly recommended to set DEFAULT_EJERK >= 10 when using LIN_ADVANCE.
1207 |     static_assert(DEFAULT_EJERK >= 10, "It is strongly recommended to set DEFAULT_EJERK >= 10 when using LIN_ADVANCE.");
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:49,
                 from Marlin/src/HAL/STM32/HAL_SPI.cpp:27:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:1207:33: error: static assertion failed: It is strongly recommended to set DEFAULT_EJERK >= 10 when using LIN_ADVANCE.
1207 |     static_assert(DEFAULT_EJERK >= 10, "It is strongly recommended to set DEFAULT_EJERK >= 10 when using LIN_ADVANCE.");
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:49,
                 from Marlin/src/HAL/STM32/Sd2Card_sdio_stm32duino.cpp:26:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:1207:33: error: static assertion failed: It is strongly recommended to set DEFAULT_EJERK >= 10 when using LIN_ADVANCE.
1207 |     static_assert(DEFAULT_EJERK >= 10, "It is strongly recommended to set DEFAULT_EJERK >= 10 when using LIN_ADVANCE.");
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:49,
                 from Marlin/src/HAL/STM32/Servo.cpp:27:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:1207:33: error: static assertion failed: It is strongly recommended to set DEFAULT_EJERK >= 10 when using LIN_ADVANCE.
1207 |     static_assert(DEFAULT_EJERK >= 10, "It is strongly recommended to set DEFAULT_EJERK >= 10 when using LIN_ADVANCE.");
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:49,
                 from Marlin/src/HAL/STM32/MarlinSerial.cpp:23:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:1207:33: error: static assertion failed: It is strongly recommended to set DEFAULT_EJERK >= 10 when using LIN_ADVANCE.
1207 |     static_assert(DEFAULT_EJERK >= 10, "It is strongly recommended to set DEFAULT_EJERK >= 10 when using LIN_ADVANCE.");
*** [.pio/build/STM32F103RET6_creality/src/src/HAL/STM32/HAL.cpp.o] Error 1
*** [.pio/build/STM32F103RET6_creality/src/src/HAL/STM32/Sd2Card_sdio_stm32duino.cpp.o] Error 1
*** [.pio/build/STM32F103RET6_creality/src/src/HAL/STM32/Servo.cpp.o] Error 1
*** [.pio/build/STM32F103RET6_creality/src/src/HAL/STM32/HAL_SPI.cpp.o] Error 1
*** [.pio/build/STM32F103RET6_creality/src/src/HAL/STM32/MarlinSerial.cpp.o] Error 1
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:49,
                 from Marlin/src/HAL/STM32/eeprom_bl24cxx.cpp:31:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:1207:33: error: static assertion failed: It is strongly recommended to set DEFAULT_EJERK >= 10 when using LIN_ADVANCE.
1207 |     static_assert(DEFAULT_EJERK >= 10, "It is strongly recommended to set DEFAULT_EJERK >= 10 when using LIN_ADVANCE.");
*** [.pio/build/STM32F103RET6_creality/src/src/HAL/STM32/eeprom_bl24cxx.cpp.o] Error 1
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:49,
                 from Marlin/src/HAL/STM32/eeprom_flash.cpp:27:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:1207:33: error: static assertion failed: It is strongly recommended to set DEFAULT_EJERK >= 10 when using LIN_ADVANCE.
1207 |     static_assert(DEFAULT_EJERK >= 10, "It is strongly recommended to set DEFAULT_EJERK >= 10 when using LIN_ADVANCE.");
*** [.pio/build/STM32F103RET6_creality/src/src/HAL/STM32/eeprom_flash.cpp.o] Error 1
=========================================================== [FAILED] Took 13.17 seconds ===========================================================

Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  FAILED    00:00:13.168
====================================================== 1 failed, 0 succeeded in 00:00:13.168 ======================================================
The terminal process "platformio 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

“Once you choose hope, anything’s possible”

tfaz099, 2021-09-04:
When I change from 4 to 10 it compiles without failure
#define DEFAULT_EJERK    10.0  // May be used by Linear Advance

“Once you choose hope, anything’s possible”

tfaz099, 2021-09-04:
Hello again,
So I tried loading in printer.

Result is that babystepping always still does not work

Also, when “live” is not checked and I check it… it drops nozzle to bed like I was going to do a fine adjustment… this is not how babystepping is designed to work

It should work from wherever the position of the z axis is.  It does *not* need to move to the bed as it defeats the purpose of adjusting the z-height wherever you are at the moment.

It was worth trying tho :/.

Thank you
tony

“Once you choose hope, anything’s possible”

tititopher68-dev, 2021-09-04:
@tfaz099 
I don't think we understand each other.  It is a question of adjusting the Z-offset  (the distance between the nozzle and the pin of the BLtouch trigged....taking as a reference the position 0 of the Bed at the thickness near a sheet of paper)...

And why do you activate the incompatible Linear Advance unless you physically modify the motherboard ....

tfaz099, 2021-09-04:
I did not activate linear advance, That is how it is set when I unzip the Jyers from github.

Yes I know the z-offset is the offset from where the probe gets us to the bed.

Odd, I know…. But I changed nothing from the default Jyers config except to uncomment babystepping always available… that offset from 1 to 40, and dropped in your configuration files.

Upon compile it fails that the setting of 4 for EJERK needs to be at least 10… so I changed to 10 to see if it would compile, and it did.

Other than that I touched nothing.

But that babystepping did not work either way ☹

How was your night?

“Once you choose hope, anything’s possible”

tititopher68-dev, 2021-09-04:
@tfaz099 
Smith3D's approach (via M851) is completely different on the notion of Z-Offset and how to set Z-Offset ....
I found the why of this choice by Jyers : #979 -> "this is the intended behavior. The step up and step down buttons are meant to be a z offset adjustment, not just babystepping. This is also the case because if I were to make such behavior only work when babystep z-offset is enabled, it would break all other firmware functions that rely on babystepping. Hypothetically I could add an on screen setting to enable or disable this functionality, but I can't really see why anyone would need to babystep during printing without adjusting their offset."

Note: it is 4 p.m. in France ;-)

tfaz099, 2021-09-04:
Innoway that is exactly correct when you were adjusting your Z offset during printer that is when baby stepping is working so effectively when you are baby step and you are changing your Z offset and then when you click after you’ve moved up or down it saves but the Z axis is always moving up and down when you turn that knob when you’re in the Z offset. Have you ever tried the Smith I did send you a binary 

“Don’t adapt to the energy in the room, influence it”

Jyers, 2021-09-09:
@tfaz099 @tititopher68-dev 

I will admit, I didn't fully read through all of that, so let me check if I fully understand the issue here...

First what the firmware should do as I intended ( to avoid any mis-understanding ):
In the Z-Offset menu, you can select "Live-Adjust", this options only affects the Z-Offset menu and not the tune menu or any other operations. When selected in the Z offset menu, it homes the printer and brings the Z axis to what it thinks is 0. From there you can use the Z offset adjustments to move the printhead until you're nozzle gap is correct. This allows you to easily and quickly set the Z offset without the need to re-home after every change.
In the tune menu there are also Z offset adjustments. These adjustments are always live. Whenever you change the Z offset in the tune menu it will move the print head accordingly. This allows for easy fine-tuning of the printer when printing to make sure you have the offset just right.

From what I understand, the behavior that you might be looking for is the movement of the axis -while- you are turning the encoder. That is simply not the implementation I went with as far as the adjustment. The new offset is applied after you "select" it by clicking, if you are looking to dial in the offset slightly, I would highly recommend using the step up/down buttons rather than the knob.

If there is still an issue you see with this implementation, feel free to let me know. I am well aware people have different preferences for how operations like this should function, this is just what I found to be most effective.

lnevo, 2021-09-09:
“In the tune menu there are also Z offset adjustments. These adjustments are always live. Whenever you change the Z offset in the tune menu it will move the print head accordingly. This allows for easy fine-tuning of the printer when printing to make sure you have the offset just right.”

i do not get this behavior. It may move once in the Tune menu, if that, but I do not get any adjustment during a print for me to think the tune menu works at all for z offset. I can use tune for other things, but z offset I have to stop/start printing every time.  Using 1.3.5b for the record.

tfaz099, 2021-09-09:
Im sorry… it just does not work.
The intent of babystepping always is that we can adjust it anytime with the encoder…. Too much work to do a simple adjustment this way.
You have done a lot of work with the firmware… maybe too much, but I get that some that need the handholding like it.
I will stick with babystepping that works always without homing

Thank you,

tony

“Once you choose hope, anything’s possible”

tititopher68-dev, 2021-09-09:
@tfaz099 @lnevo 
I totally agree with @Jyers .... Sorry to contradict you but it works ...
When you print and activate the Tune menu and use the Microstep Up or Down, the Z axis goes up and down immediately ... I'll show it to you on a video ...
I don't know what your configurations are, but you have to activate things that block ... The way of approaching Z-offset (Smith3D type) comes from the old Marlin firmware 1.0.x. I'm not saying it doesn't work, I'm just saying the method is different. I also had fluctuating Z-Offsets with Smith3D. I haven't had them since with the Jyers database .... And as I always go to the end of my tests, I tested the gap by actually measuring my Z-Offset using precision calibrated shims and the Z-offset I get is right!

The configuration files are a base of parameters for 3D printers but not all parameters can be used for all 3D printers and doing a multitude of #if or #else tests to know if we can or not use a particular parameter would become a Gaz factory. Also some parameters relating to the notion of babystep cannot be used here, it is a perfectly valid choice, otherwise Marlin would not have validated it in the latest Bugfixes ...
I believe that the real problem is the change of practice, of way of doing .... There may be things to adjust like being able to actually move the Z axis when you turn the knob in the Prepare menu - Z-Offset and more after validation ... A Jyers to decide if this is relevant or not ... but for me, it does not disturb ...

lnevo, 2021-09-09:
> @tfaz099 @lnevo @Jyers
> I totally agree with @Jyers .... Sorry to contradict you but it works ...
> When you print and activate the Tune menu and use the Microstep Up or Down, the Z axis goes up and down immediately ... I'll show it to you on a video ...

I will put a mark on the z axis to watch for movements, but nothing I do in Tune to change z offset click the up/down, change it in the menu and click. Neither has affect. Maybe I don’t see it so will make a mark and watch for movement. Everyone says this is how they tune zoffset during the skirt print. I have yet to see this work..,

tititopher68-dev, 2021-09-09:
Video Microstep UP (Printing file) - only first layer....
https://user-images.githubusercontent.com/84127160/132682994-4e8282ce-8dc4-4fef-869e-ad35b38fcab5.mp4

tfaz099, 2021-09-09:
Regardless of what you find acceptable the bottom line is that when you take over maintaining and improving software you *do not* cripple what already has proven to work.  Improvements are great, but in this case something that was not broken and in fact worked very well, was bastardized in the process.

I have read all the comments about what “worked and didn’t work” and honestly this firmware is cumbersome and cannot recommend it to anyone.

I understand that Smith obviously was tired of what he had taken on, but bottom line, it is still superior.

I will sign out of this thread as it is not going anywhere.

Good luck to you all

tony

“Once you choose hope, anything’s possible”

tfaz099, 2021-09-09:
Can you babystep up and down anytime… even right after turning printer on with no homing ?

“Once you choose hope, anything’s possible”

https://user-images.githubusercontent.com/84127160/132682994-4e8282ce-8dc4-4fef-869e-ad35b38fcab5.mp4

tititopher68-dev, 2021-09-09:
@tfaz099 
Before doing a Z-offset, a printer needs to know its current position, so yes, an autohome will be done if necessary, if the position of the axes cannot be known with certainty and this is a Marlin function.
As I told you, this firmware works very well, has been tested by many Youtubers, like CHEP, JSTECH, etc ... who are serious people.

There are thousands of us who use this firmware with satisfaction.

I spent a lot of time analyzing the differences and trying to find a connection on certain points.

I do not understand your blocking on this method. You do not wish to change it, well, perfect, it is your right but that does not allow you to insult the work of Jyers and the other contributors ("was bastardized in the process") also rightly recognized !! ! ... There are other firmwares for the Ender3V2 ... Like Mriscoc for example and good luck to have the real sources of these releases for your request, catr he does not share anything ....

tititopher68-dev, 2021-09-09:
@tfaz099 
you wrote : "I will sign out of this thread as it is not going anywhere."
Certainly a good idea...

tititopher68-dev, 2021-09-09:
@lnevo 
Sorry for my moodiness towards tfaz099, but hey ....

Can you give more details on your firmware (Using 1.3.5b for the record. -> too imprecise ;-) ), your configuration files ... Last build compiled (commit fac4bf4)?

tfaz099, 2021-09-09:
Lmfao

Funny. But you are obviously only programmers and have no idea how to “develop” for end users. 

Sorry for my moodiness also but fact is fact

“Don’t adapt to the energy in the room, influence it”

tfaz099, 2021-09-09:
It will never be fixed

closed 2021-09-09 by tfaz099

tititopher68-dev, 2021-09-09:
@tfaz099 
Pourquoi cloturé le fil alors que vous continuez....Les utilisateurs finaux que nous sommes ici (et je pense pouvoir parlé ainsi dans ce cas précis) apprécions à sa juste valeur, le firmware remarquable de @Jyers....Vous êtes ici le seul à insulter le travail fait et vous êtes le seul....

tititopher68-dev, 2021-09-09:
@lnevo 
I invite you to open a new thread with your issue ... ;-)

lnevo, 2021-09-09:
Of course, on testing, the function performed the way it was supposed to and I was able to influence the z-offset during print. Sorry to add noise.   

I'm using the 4.2.2 BLTouch High Speed 15x15 UBL precompiled release for the record. 

Thanks to all the developers and users providing support.

ledmirage, 2021-09-09:
i have been reading this thread on and off, without saying who is right and who is wrong:

1) this is open source software, no one owes anyone anything.

2) People fork software for different reasons, mostly because the original software didn't give what they need hence people fork it and implement it themselves, this is exactly what open source software meant to be.

3) If you disagree with what this s/w feature, go back to the basic and see point 2, fork it yourself.

Linus the creator of the Linux kernel, maintain his own (official) source tree, and rejected many commits he disagrees with, but he doesn't stop you from forking your own branch and do whatever you want there.

These developers put in a lot of effort to write the s/w, have fun with it, and share with other people for free, and from what i see most end users including myself appreciate it very much. I personally think this is superb compare to stock Ender firmware. really can't ask for more. You can argue until the cow come home saying this is wrong, but sorry, this is Jyers' repo, he has the last call, but you are free to open another repo and write your own amazing s/w, and if its good, people will thank you for that too, sounds like a win-win to me, lol

tititopher68-dev, 2021-09-09:
@lnevo 
Happy for you :)

tititopher68-dev, 2021-09-09:
@ledmirage 
Well said !
We can fork the firmware and add your own modifications or additions (like others or I have done lol) and submit them to @Jyers who will do what he wants and even who will help you, sometimes advise on the code.  ..True sharing ...