Jyers/Marlin #1367: Stuck on print in progress screen after cancellation

urmonator, 2021-09-01:
### Description

I am using the E3V2-BLTouch-5x5-v4.2.2-v1.3.5.b release of Jyers, with Octoprint attached as well. I have made no changes to the firmware, and am using Octoprint to monitor the prints.

Nearly every time I cancel a print, the printer stops, presents the print to me on the build plate, but then the screen goes into a "waiting for input" screen. I click the button, and it just goes back to the screen showing the current print in progress instead of back to the main menu. No matter what I press, it is stuck on that screen until a reboot. It also sometimes asks if I want to purge more filament. Clicking yes on that does nothing.

### Steps to Reproduce

1. Run print.
2. Cancel print in progress.
3. Press the button on the "waiting for input screen."

**Expected behavior:** Should return to the home screen.

**Actual behavior:** Remains stuck on print in progress screen.

Some pictures of the screens: 

[image]
[image]
[image]
[image]
[image]
[image]
[image]

tititopher68-dev, 2021-09-01:
@urmonator 
Looks like a communication problem with Octoprint ... Octoprint does not seem to take into account stopping printing and always wants to continue printing hence the non logical popups.

Compile the latest build and change the initial baudrate in configuration.h from 115200 to 250000.

eblieb, 2021-09-06:
> @urmonator
> Looks like a communication problem with Octoprint ... Octoprint does not seem to take into account stopping printing and always wants to continue printing hence the non logical popups.
> 
> Compile the latest build and change the initial baudrate in configuration.h from 115200 to 250000.

Haven't ever compiled the firmware myself so not sure where to start. Any suggestions?

kineticscreen, 2021-09-07:
Also having this issue, and happens whether the print is cancelled on the screen, or via the OctoPrint interface.

BTW this is when printing from the SD card via the printer's UI - only using OctoPrint for monitoring from an old Raspberry Pi B+ (not going to attempt printing from the Pi on that old thing!)

It almost seems like the printer is getting a 'change filament' code instead of the 'cancel print' when octoprint is connected.
Works normally when it is unplugged.

Jyers, 2021-09-09:
Very strange. If I have time I'll definitely give this a try and see if I can find any reasoning for this response

cnorton-webdev, 2021-09-30:
I've noticed the same issue as well, and I've noticed that if I don't have anything in the after cancel gcode in Octoprint it gets stuck on the printing screen, but if I have something in the Octoprint cancel gcode script it will at least let me back to use the main menu. It's a very strange bug.

kineticscreen, 2021-10-03:
Not sure if something has happened on Octoprint's side, but I just cancelled a print via the Ender 3's screen, and it all behaved itself normally.