Jyers/Marlin #1370: Octorprint Disconnecting - Past Month Issue

HermanGaviriaMarin, 2021-09-03:
### Description

Octoprint continues to disconnect randomly from my printer, right as I'm about to check the octoprint webpage it says that it has been disconnected. Its been happening several times a day now. Only thing I have seen slight correlation with or maybe its just been a coincidence is hitting the preheat button and then going to the computer just to see octoprint disconnected. 

### Steps to Reproduce

1. Have latest version of octoprint installed in raspberry pi 4B 4GB on an 128GB SD card (samsung)
2. Have an ender 3 v2 running latest version of Jyers firmware (mine is custom and manually compiled though I don't mess with anything regarding octoprint, or so I think)
3. Preheat (or so I think) then go print something, only to find out octoprint has disconnected and you have to reconnect again. Another correlation I think I've see is leaving the printer Idle for some minutes (10-20)
4. Only firmware changes I make are regarding the filament sensor (motion BTT), thermistor, max temperature, and linear advance (yes I enabled UART using a soldering mod, which I can confirm was done right since I was able to calibrate K valuie). I currently have my Linear advance turned off (K value at zero and also switched the stepper back to stealth chop).

**Expected behavior:** 
Octoprint shouldn't disconnect randomly unless I disconnect the micro USB port from the printer or the Pi. 

**Actual behavior:** 
Its disconnecting when its not supposed to. 

#### Additional Information

[JYERS. Herman CONFIG (2).zip](https://github.com/Jyers/Marlin/files/7108159/JYERS.Herman.CONFIG.2.zip)

tititopher68-dev, 2021-09-03:
@HermanGaviriaMarin 
It's not a firmware issue but like the octoprint community said, a octoprint issue. 
Check the power supply of your Raspberry ... USB camera? And above all check that you have masked the 5v pin of the USB socket of your cable!  Power LOW source for the RPi ...

Cal-LF, 2021-09-03:
@tititopher68-dev Possible the baud rate again?

tititopher68-dev, 2021-09-03:
@Cal-LF 
I don't think so, although setting the baurate at 250,000 is better.
 There are also problems between Octoprint and the Rpi4 ... an Rpi 3b + is more reliable.
 The cable problem is often pointed out in cases of disconnection, especially with the ender3 and other printers .... because the Rpi also wants to supply the printer with 5v ... It is therefore necessary to hide the 5v pin of the USB socket  .

melvinmicallef, 2021-09-04:
Did you try another cable? I had this issue but switching to a better usb data cable solved this

Jyers, 2021-09-09:
Going to close this as it is most likely a hardware, octoprint, or upstream issue. Feel free to keep troubleshooting as I'm sure other community members would be happy to help you get it sorted out.

closed 2021-09-09 by Jyers