Jyers/Marlin #1376: how to confirm Manual Mesh works during printing?

liuxyc, 2021-09-05:
### Description
I did manual mesh successfuly on ender3v2 with E3V2-ManualMesh-3x3-v4.2.2-v1.3.5.b.bin firmware, i can view the mesh data from the lcd display.
Also,  after  G28 ,  i applied M420 S1 command and get  Bed Level On response.
but i dont see any Z offset change during the print when the nozzle moving from one area to another.
does the manual mesh works on my printing?  any data to confirm ?

### Steps to Reproduce

1. do 3x3 manual mesh from  LCD controller
2. write  M420 S1 after  G28 in  gcode file
3. do print

**Expected behavior:** [What you expect to happen]
Z offset changes to mesh data, when nozzle move to correspond area.

**Actual behavior:** [What actually happens]
Z offset no changes

PapsDizzy, 2021-09-05:
make a mark on your z clutch and start printig. take a look at the mark you made on the clutch, you will see the mark is moving.
on the display will the z correction off the mesh not shown. sorry for my bad language.

Jyers, 2021-09-09:
@PapsDizzy Is correct. The best way is to watch the Z coupler for movement, as there is no direct lcd response that will be able to indicate mesh movement.

closed 2021-09-09 by Jyers