Jyers/Marlin #1377: How is DWIN overide done ?

spacecabbie, 2021-09-05:
### Description

Where do you override the LCD display I am adapting this to work with SKR Mini e3 V2.0 board.
Since the only thing I changed is environment and configuration.h and configuration_adv.h
Now I get the  #error "LCD_BED_LEVELING requires a programmable LCD controller." 

I know @Jyers overrides it somewhere was wondering how, so I can get manual bed level menu back again.

Ps. out of curiosity why on earth is marlin official still yelling every where is not possible while it clearly is.

tititopher68-dev, 2021-09-06:
@spacecabbie 
Often the answers are either in the discussion section or the closed issues...
Please read this #1323...

Jyers, 2021-09-09:
To answer this a little more simply, LCD_BED_LEVELING and such are not currently used by the firmware, it is simply enabled by default without the need for that define, this may be fixed upstream, but for now that is how the "override" is handled

closed 2021-09-09 by Jyers