Jyers/Marlin #1385: Printing with Octoprint - Printer keeps requesting line ### again and again, communication stuck

melvinmicallef, 2021-09-10:
### Description

Whilst printing using octoprint, after some time i start getting errors of this type 'INFO - Ignoring resend request for line 450353, that still originates from lines we sent before we got the first resend request'

This happens multiple time but probably the print does not stop since the printer catches up. After multiple occasions, the printer disconnects with an error 'Printer keeps requesting line 450356 again and again, communication stuck'.

This happened to me multiple times with prints that exceed 6 hours and i cannot seems to print past that. Is this a firmware issue or should i open a bug in the octoprint repo instead? I have attached the octoprint log file if you want to have a look for yourself. You can search in the file by 'Ignoring resend request for line' and 'again and again, communication stuck'.

 Of course, using the sd is an option but printing through octoprint is much more comfortable especially for monitoring.

### Steps to Reproduce

1.Slice
2.Upload on Octoprint
3.Weld using ArcWelder (I am forced to use this due to stuttering/buffer issues i was having)
4.Print using Meatpack 
5.Print stops after a few hours

**Expected behavior:** [What you expect to happen]
Print does not stop :P

**Actual behavior:** [What actually happens]
Print stops :D

#### Additional Information
I have attached the octoprint log file, two gcode files (the original and a manually changed one trying to resume the print) and my firmware configs. Please let me know if i can provide or try anything else. Sorry but i couldnt upload the zip file here so had to share on drive https://drive.google.com/file/d/1VV4MEJPOCbDCWqArIBG6cCfwi85wOSXM/view?usp=sharing

Cal-LF, 2021-09-10:
The discussions area would be a better place to post Octoprint issues. This is not a Jyers firmware issue.

closed 2021-09-11 by melvinmicallef

jonfairbanks, 2021-09-28:
Also seeing this. Created a discussion here: https://github.com/Jyers/Marlin/discussions/1439