Jyers/Marlin #1386: Buzzing sound wrong firmware ender 3 v2

Whitewater92, 2021-09-11:
Just got an ender 3V2 in this week (first 3d printer). Did a firmware upgrade but thought I had latest version of mainboard 4.2.7. 
I have however a 4.2.2 resulting in faulty upload and buzzing screen and no display. (even with power off it continues buzzing). 
donwloaded the 1.3.5b version and installed new screen firmware (Gotcha) found in sourcecode.zip (still buzzing after update)
Then I tried installing v 1.3.5 o,n the printer. Keeps on buzzing, nothing happens, stopped after 5 minutes of buzz... any ideas?

PapsDizzy, 2021-09-11:
rename the firmware for the 4.2.2 board to something else like new.bin, put it on the sd card, only the firmware file you renamed is on the sd card. turn on printer and it should be done in a few seconds.

Whitewater92, 2021-09-11:
thx, Just did that and same story, buzzing and nothing else, nothing on the display neither...

PapsDizzy, 2021-09-11:
i hope you did not kill your board with the wrong firmware. but in my opinion it sounds like you grilled your board.

Whitewater92, 2021-09-11:
interesting, I didn't even manage to print something as I wanted to have the firmware update with the level feature to solve my initial problem... Can i check the mainboard in a way to see if it's RIP ?

Whitewater92, 2021-09-11:
the Blue LED on the PCB is still on...

Whitewater92, 2021-09-11:
the LED seems to be on when a USB cable is connected

PapsDizzy, 2021-09-11:
unfortunately I can't help you because I don't have the knowledge for that

kted, 2021-09-11:
Disconnect the USB cable and try again. The mainboard is beeing back-powered.

tititopher68-dev, 2021-09-11:
@Whitewater92 
Which firmware precisely?  A BLTouch?  If so, hope you haven't used a pin27 adapter.  Indeed, the use of this adapter causes this phenomenon ... Otherwise maybe a power problem.  Check the connections (green terminals) and those on the power supply side ...

Whitewater92, 2021-09-12:
Default 1.3.5.b version. Disconnected USB cable and it updated the mainboard !! Problem solved but very strange cable is not allowed to be connected when upgrading with SD card... anyway, problem solved ! Thx

weav2k4, 2021-09-14:
If USB is connected to the mainboard it never allows the board to power down as USB is supplying juice.  Bad design by creality to have USB power go to the board.  If it never powers down then it will never see a power cycle to flash.  Is why many people will cut the power wire on the usb cable itself to keep it from powering the mainboard.  Also needed if using a pi with octoprint as the pi will keep power to the printer mainboard via it's usb connection and it shouldn't. fwiw.

Whitewater92, 2021-09-14:
yep, not very professional. I would at least expect a power saver feature cutting down the power or bringing it to a level not high enough to power up the display.