Jyers/Marlin #1397: Screen glitch on 1.3.5b

patke92, 2021-09-17:
Hello,
I recently upgraded to a 4.2.7 board and along with that installed the latest release - 1.3.5b.
Somewhere after 12 hours, the screen gets scrambled as mentioned in Issue #1351 - I don't have Octoprint though!

tititopher68-dev, 2021-09-18:
@patke92 
In your case, compile your own firmware with the latest build and commit here....and you can modify your configuration.h to fix the baudrate at 250000 bauds.

The release 1.3.5b. is older than the existing build (to be compiled) ... The latest build is based on the latest Marlin's bugfixes, correcting the communication management of STM32 CPUs equipping our motherboards v4.2.2 and v4.2.7 ...

melvinmicallef, 2021-09-19:
Unfortunately, i was having this issue but then i switched to 250000 bauds but recently, it started happening again. I am on latest commit not on release so i should have the actual latest. My board is a 4.2.2

tititopher68-dev, 2021-09-20:
@melvinmicallef 
This glitch already existed on many versions (including with Ender 3 V1 screens using firmware only from Marlin!) but sometimes did not appear on very long Print ....
The problem is serial communication between the screen and the motherboard.

The latest modifications made by Marlin on the management of serial ports and the CPU have as aggravated the problem with our Creality motherboards :-(

I have a feeling that cycle time also plays a role in this phenomenon. It is not multitasking, but a kind of loop that runs ... I also noticed that if a cable is connected to the USB Port (without the 5v Pin being hidden), there was disturbances (whether or not OctoPrint is used ...).

For my part, I made sure to reduce the firmware cycle time without affecting the functionalities, hid the Pin 5v of my USB cable (but you can unplug it, if you are not using OctoPrint or other software), carried the Rate Baud to 250000. And even forced the screen Baudrate to 115200 in configuration.h (This might sound silly, because normally there is a condition that forces this value, but .... Say the opposite...).

Also, since I have long adopted these principles, I have never had this issue.

melvinmicallef, 2021-09-20:
That explains why that this started happening only after the merge with the main marlin branch. I experience the same issues where during some prints the screen works fine, and during others it glitches. I am using octoprint with the 5v power pin covered, however i have my baud set to 115 since 250 causes other issues :/ Well to be fair the screen glitching out is not hurting me much cause most of the time it is off whilst printing since i control through octoprint.